espressif/esp-idf #14280: ESP32-C6 crashes in adc_oneshot_read() after bootloader_random_enable()

pelya, 2024-07-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### Espressif SoC revision.

ESP32-C6

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C6

### Power Supply used.

USB

### What is the expected behavior?

Not crash.

### What is the actual behavior?

The crash only happens on ESP32-C6. The same code works fine on ESP32-C3.

### Steps to reproduce.

Please see the attached code sample to reproduce the crash. Removind bootloader_random_enable() and bootloader_random_disable() will stop the crashing. ADC is freed before calling bootloader_random_enable() and re-initialized after calling bootloader_random_disable(), so there should be no issue with concurrent usage of ADC.

```
#include <freertos/FreeRTOS.h>
#include "esp_log.h"
#include "esp_adc/adc_oneshot.h"
#include "esp_random.h"
#include "bootloader_random.h"

const static char *TAG = "EXAMPLE";

#define EXAMPLE_ADC1_CHAN0          ADC_CHANNEL_0
#define EXAMPLE_ADC_ATTEN           ADC_ATTEN_DB_12

void app_main(void)
{

    adc_oneshot_unit_handle_t adc1_handle;
    adc_oneshot_unit_init_cfg_t init_config1 = {
        .unit_id = ADC_UNIT_1,
    };
    ESP_ERROR_CHECK(adc_oneshot_new_unit(&init_config1, &adc1_handle));

    adc_oneshot_chan_cfg_t config = {
        .bitwidth = ADC_BITWIDTH_DEFAULT,
        .atten = EXAMPLE_ADC_ATTEN,
    };
    ESP_ERROR_CHECK(adc_oneshot_config_channel(adc1_handle, EXAMPLE_ADC1_CHAN0, &config));

    while (1) {
        int adc_raw = 0;
        ESP_ERROR_CHECK(adc_oneshot_read(adc1_handle, EXAMPLE_ADC1_CHAN0, &adc_raw));
        ESP_LOGI(TAG, "ADC%d Channel[%d] Raw Data: %d", ADC_UNIT_1 + 1, EXAMPLE_ADC1_CHAN0, adc_raw);
        vTaskDelay(pdMS_TO_TICKS(1000));

        ESP_ERROR_CHECK(adc_oneshot_del_unit(adc1_handle));
        bootloader_random_enable();
        uint32_t value = esp_random();
        ESP_LOGI(TAG, "Generated true random value: 0x%08lx", value);
        bootloader_random_disable();
        ESP_ERROR_CHECK(adc_oneshot_new_unit(&init_config1, &adc1_handle));
        ESP_ERROR_CHECK(adc_oneshot_config_channel(adc1_handle, EXAMPLE_ADC1_CHAN0, &config));
    }
}
```

### Debug Logs.

```plain
I (321) main_task: Calling app_main()
I (321) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (331) EXAMPLE: calibration scheme version is Curve Fitting
I (341) EXAMPLE: Calibration Success
I (341) EXAMPLE: ADC1 Channel[0] Raw Data: 2465
I (351) EXAMPLE: ADC1 Channel[0] Cali Voltage: 2367 mV
I (1351) EXAMPLE: Generated true random value: 0xb467407f
I (1351) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
...
```

### More Information.

Reproducibility 100%

Bruce297, 2024-08-01:
`bootloader_random_disable` should not be called after ADC initialize, that's a known conflict.
You can see on https://github.com/espressif/esp-idf/blob/41dd1a351b4f8630a0b9dbdd87a6bf782094dd66/components/bootloader_support/include/bootloader_random.h#L39
 
We will consider to fix it if possible.

playfulFence, 2024-08-07:
I think this bug is already being worked on in a similar https://github.com/espressif/esp-idf/issues/14124 issue

closed 2024-08-14 by espressif-bot