espressif/esp-idf #14282: How to enable WAPI feature

travisbit, 2024-07-31:
### Is your feature request related to a problem?

I saw in the development kit(esp-idf-v5.2.2) that there is a CONFIG_WPA_WAPI_PSK macro definition, but I didn't see any corresponding code processing. Is it the matching version or do I need to download a special version

### Describe the solution you'd like.

Tell me the correct version, or how to enable the wapi feature, preferably with examples

### Additional context.

esp-idf version :v5.2.2

kapilkedawat, 2024-08-01:
@travisbit Please enable `[ESP_WIFI_WAPI_PSK](https://github.com/espressif/esp-idf/blob/41dd1a351b4f8630a0b9dbdd87a6bf782094dd66/components/esp_wifi/Kconfig#L533)` which will enable WAPI support. Handling is present inside wifi libs.

travisbit, 2024-08-01:
@kapilkedawat 1：WAPI has two authentication methods, certificate authentication and PSK authentication. Currently, does it only support PSK authentication? The certificate authentication option cannot be seen in sdkconfig.
2：In the wpa_validate_wpa_ie function, there is a selector local variable used to select encryption functions, etc. However, I did not see this variable being used and only assigned a value, which is logically meaningless. What I want to confirm is that WAPI has been fully tested and verified, right.

kapilkedawat, 2024-12-18:
Hi @travisbit, apologies for the delayed, I somehow missed the comment. Yes only PSK auth in WAPI is supported and yes its fully tested.

closed 2024-12-18 by Alvin1Zhang