espressif/esp-idf #14283: can not build littlefs example project on win11 with ESP-IDF V1.8.0 (VSC-1455)

masm31, 2024-07-31:
### OS

Windows

### Operating System version

11

### Visual Studio Code version

1.91.1

### ESP-IDF version

1.8.0

### Python version

3.12.2

### Doctor command output

Hi,
I want to experiment with littlefs  example, But It can not build the project, although I can build other projects like the blinky

Here is the output of the terminal for your reference

Executing task: C:\Users\Ali\.espressif\tools\ninja\1.11.1\ninja.exe  

[0/1] Re-running CMake...-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
Processing 2 dependencies:
[1/2] idf (5.2.2)
[2/2] joltwallet/littlefs (1.14.8)
-- Project sdkconfig file C:/Users/Ali/Desktop/espTest/lfs/littlefs/sdkconfig
Loading defaults file C:/Users/Ali/Desktop/espTest/lfs/littlefs/sdkconfig.defaults...
-- Compiler supported targets: xtensa-esp-elf
-- App "littlefs_example" version: 1
-- Adding linker script C:/Users/Ali/Desktop/espTest/lfs/littlefs/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/Users/Ali/Desktop/espTest/lfs/littlefs/build/esp-idf/esp_system/ld/sections.ld.in
-- Adding linker script C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script C:/Users/Ali/esp/v5.2.2/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_bootloader_format esp_coex esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 joltwallet__littlefs json log lwip main mbedtls mqtt newlib nvs_flash nvs_sec_provider openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Users/Ali/esp/v5.2.2/esp-idf/components/app_trace C:/Users/Ali/esp/v5.2.2/esp-idf/components/app_update C:/Users/Ali/esp/v5.2.2/esp-idf/components/bootloader C:/Users/Ali/esp/v5.2.2/esp-idf/components/bootloader_support C:/Users/Ali/esp/v5.2.2/esp-idf/components/bt C:/Users/Ali/esp/v5.2.2/esp-idf/components/cmock C:/Users/Ali/esp/v5.2.2/esp-idf/components/console C:/Users/Ali/esp/v5.2.2/esp-idf/components/cxx C:/Users/Ali/esp/v5.2.2/esp-idf/components/driver C:/Users/Ali/esp/v5.2.2/esp-idf/components/efuse C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp-tls C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_adc C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_app_format C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_bootloader_format C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_coex C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_common C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_eth C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_event C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_gdbstub C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_hid C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_http_client C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_http_server C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_https_ota C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_https_server C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_hw_support C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_lcd C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_local_ctrl C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_mm C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_netif C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_netif_stack C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_partition C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_phy C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_pm C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_psram C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_ringbuf C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_rom C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_system C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_timer C:/Users/Ali/esp/v5.2.2/esp-idf/components/esp_wifi C:/Users/Ali/esp/v5.2.2/esp-idf/components/espcoredump C:/Users/Ali/esp/v5.2.2/esp-idf/components/esptool_py C:/Users/Ali/esp/v5.2.2/esp-idf/components/fatfs C:/Users/Ali/esp/v5.2.2/esp-idf/components/freertos C:/Users/Ali/esp/v5.2.2/esp-idf/components/hal C:/Users/Ali/esp/v5.2.2/esp-idf/components/heap C:/Users/Ali/esp/v5.2.2/esp-idf/components/http_parser C:/Users/Ali/esp/v5.2.2/esp-idf/components/idf_test C:/Users/Ali/esp/v5.2.2/esp-idf/components/ieee802154 C:/Users/Ali/Desktop/espTest/lfs/littlefs/managed_components/joltwallet__littlefs C:/Users/Ali/esp/v5.2.2/esp-idf/components/json C:/Users/Ali/esp/v5.2.2/esp-idf/components/log C:/Users/Ali/esp/v5.2.2/esp-idf/components/lwip C:/Users/Ali/Desktop/espTest/lfs/littlefs/main C:/Users/Ali/esp/v5.2.2/esp-idf/components/mbedtls C:/Users/Ali/esp/v5.2.2/esp-idf/components/mqtt C:/Users/Ali/esp/v5.2.2/esp-idf/components/newlib C:/Users/Ali/esp/v5.2.2/esp-idf/components/nvs_flash C:/Users/Ali/esp/v5.2.2/esp-idf/components/nvs_sec_provider C:/Users/Ali/esp/v5.2.2/esp-idf/components/openthread C:/Users/Ali/esp/v5.2.2/esp-idf/components/partition_table C:/Users/Ali/esp/v5.2.2/esp-idf/components/perfmon C:/Users/Ali/esp/v5.2.2/esp-idf/components/protobuf-c C:/Users/Ali/esp/v5.2.2/esp-idf/components/protocomm C:/Users/Ali/esp/v5.2.2/esp-idf/components/pthread C:/Users/Ali/esp/v5.2.2/esp-idf/components/sdmmc C:/Users/Ali/esp/v5.2.2/esp-idf/components/soc C:/Users/Ali/esp/v5.2.2/esp-idf/components/spi_flash C:/Users/Ali/esp/v5.2.2/esp-idf/components/spiffs C:/Users/Ali/esp/v5.2.2/esp-idf/components/tcp_transport C:/Users/Ali/esp/v5.2.2/esp-idf/components/ulp C:/Users/Ali/esp/v5.2.2/esp-idf/components/unity C:/Users/Ali/esp/v5.2.2/esp-idf/components/usb C:/Users/Ali/esp/v5.2.2/esp-idf/components/vfs C:/Users/Ali/esp/v5.2.2/esp-idf/components/wear_levelling C:/Users/Ali/esp/v5.2.2/esp-idf/components/wifi_provisioning C:/Users/Ali/esp/v5.2.2/esp-idf/components/wpa_supplicant C:/Users/Ali/esp/v5.2.2/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/Ali/Desktop/espTest/lfs/littlefs/build

[570/877] cmd.exe /C "cd /D C:\Users\Ali\Desktop\espTest\lfs\littlefs\build\esp-idf\main &...ke/3.24.0/bin/cmake.exe -P C:/Users/Ali/esp/v5.2.2/esp-idf/tools/cmake/scripts/fail.cmake 
FAILED: esp-idf/main/CMakeFiles/littlefs C:/Users/Ali/Desktop/espTest/lfs/littlefs/build/esp-idf/main/CMakeFiles/littlefs
cmd.exe /C "cd /D C:\Users\Ali\Desktop\espTest\lfs\littlefs\build\esp-idf\main && C:\Users\Ali\.espressif\tools\cmake\3.24.0\bin\cmake.exe -E echo "Windows does not support LittleFS partition generation" && C:\Users\Ali\.espressif\tools\cmake\3.24.0\bin\cmake.exe -E remove C:/Users/Ali/Desktop/espTest/lfs/littlefs/build/esp-idf/main/xk2Ig.cmake && C:\Users\Ali\.espressif\tools\cmake\3.24.0\bin\cmake.exe -E env "FAIL_MESSAGE=Failing the build (see errors on lines above)" C:/Users/Ali/.espressif/tools/cmake/3.24.0/bin/cmake.exe -P C:/Users/Ali/esp/v5.2.2/esp-idf/tools/cmake/scripts/fail.cmake"
Windows does not support LittleFS partition generation
CMake Error at C:/Users/Ali/esp/v5.2.2/esp-idf/tools/cmake/scripts/fail.cmake:3 (message):
  Failing the build (see errors on lines above)

[590/877] Performing configure step for 'bootloader'
-- Found Git: C:/Users/Ali/.espressif/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1")
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/Ali/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe
[... cut ...]
-- Build files have been written to: C:/Users/Ali/Desktop/espTest/lfs/littlefs/build/bootloader
ninja: build stopped: subcommand failed.

 *  The terminal process "C:\Users\Ali\.espressif\tools\ninja\1.11.1\ninja.exe" terminated with exit code: 1. 

### Debug Message

```plain
I can not even build the project
```

### Other Steps to Reproduce

asd

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

igrr, 2024-07-31:
Thanks for the issue report. LittleFS supports filesystem generation on Windows since esp-littlefs 1.9.0, so we should remove this part in the example:

https://github.com/espressif/esp-idf/blob/41dd1a351b4f8630a0b9dbdd87a6bf782094dd66/examples/storage/littlefs/main/CMakeLists.txt#L10

masm31, 2024-07-31:
thanks , my problem solved

masm31, 2024-08-01:
I comment this line # fail_at_build_time(littlefs "Windows does not support LittleFS partition generation")
and my project built and I wrote in flash , but the external txt file ("/littlefs/example.txt") not open , and print "Failed to open file for reading".

igrr, 2024-08-02:
Have you only commented that one line? Please take a look at the two lines right above:

https://github.com/espressif/esp-idf/blob/41dd1a351b4f8630a0b9dbdd87a6bf782094dd66/examples/storage/littlefs/main/CMakeLists.txt#L7-L9

this means that the function to generate the filesystem likely wasn't called at all.