espressif/esp-idf #14290: ESP32S3 gpio_glitch_filter causing major current consumption

pad52, 2024-08-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am using a ESP32S3 with power saving enabled with `CONFIG_FREERTOS_USE_TICKLESS_IDLE `
and
`  esp_pm_config_t pm_config;
  pm_config.max_freq_mhz = 80;
  pm_config.min_freq_mhz = 40;
  pm_config.light_sleep_enable= true;
  
  ESP_ERROR_CHECK(esp_pm_configure(&pm_config));`
  
My custom board (with LEDs etc.) consumes about 20mA in this configuration.
As soon as I enable the gpio_glitch_filter on one pin using this code:
`gpio_glitch_filter_handle_t filter;
    gpio_pin_glitch_filter_config_t filter_config = {GLITCH_FILTER_CLK_SRC_DEFAULT, GPIO_555_i};
    
    ESP_ERROR_CHECK(gpio_new_pin_glitch_filter(&filter_config,&filter));
    ESP_ERROR_CHECK(gpio_glitch_filter_enable(filter));`
  
The board jumps immediately to 30mA. Maybe the gpio_glitch_filter is not working well with power saving mode?

suda-morris, 2024-08-02:
`gpio_glitch_filter_enable` this, will acquire a `ESP_PM_NO_LIGHT_SLEEP` pm_lock inside, which would prevent the system from going to sleep. This is because when the system sleeps, the clock source used by the glitch filter will change or disabled, thus making the glitch filter doesn't work correctly.

So, in you application, when you think the glitch filter doesn't need to work any more, just call `gpio_glitch_filter_disable`, it will release the pm_lock and your system can go to sleep afterwards.

esp-lis, 2024-08-26:
If the gpio_glitch_filter is not needed to operate throughout the full lifecycle of the application, we recommend disabling it when the gpio_glitch_filter function is not required.