espressif/esp-idf #14291: IDF v5.3 does not build with latest esp-cryptoauthlib's commit and ESP32-C6

dragomat, 2024-08-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

After setting esp32c6 target, build process should complete successfully.

### What is the actual behavior?

Compiler stops with following error:

[image]

### Steps to reproduce.

1. Download and update esp-cryptoauthlib component to the latest commit
2. Update esp-idf from any v5.x to v5.3
3. Try executing idf.py build using esp32c6 as target

### More Information.

Based on esp-cryptoauthlib's commit history, it seems that the solution initially provided, i.e., to use SOC_HP_I2C_NUM instead of SOC_I2C_NUM, did not work for versions of IDF older than 5.3...  And it was reverted back to SOC_I2C_NUM. 
However, ESP32-C6 now has SOC_I2C_NUM=2 since it adds HP_I2C with LP_I2C, but I2C_NUM_1 is still not defined, thus such inconsistency/error was created. 

Can you please confirm my findings?

mahavirj, 2024-08-08:
@dragomat 

ESP32-C6 target is not yet supported in esp-cryptoauthlib. We will consider adding the support soon.

dragomat, 2024-08-08:
> ESP32-C6 target is not yet supported in esp-cryptoauthlib. We will consider adding the support soon.

Thanks for the reply. Is it only a matter of defining the MACROS for ESP32-C6 or are you referring to some library's core functionalities as well? 

I am currently using esp-cryptoauthlib on ESP32-C6 and, besides the mentioned issue on the macro, it seems to work fine.

closed 2024-08-14 by espressif-bot

nileshkale123, 2024-08-14:
@dragomat 

Thank you for bringing this issue to our attention. We have implemented the necessary fix to address the problem you encountered. You can review the changes in detail [here](https://github.com/espressif/esp-cryptoauthlib/commit/7cafbc1d2820d0f7198c08bed37c16c139233464).

In this update, we have refined the macro usage to align with the IDF version and target specifications. Additionally, we have extended support for the ESP32C6 target within this component.

We appreciate your patience and your commitment.