espressif/esp-idf #14292: IDF 5.3 does not installed correct version of Python

readmodifywrite, 2024-08-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-256-gdc7fb34fca

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

It builds.

### What is the actual behavior?

Build fails, no Python.

### Steps to reproduce.

Build a project in 5.3.  It will fail.

### Build or installation Logs.

```plain
Detecting the Python interpreter
Checking "python3" ...
Python 3.10.12
"python3" has been detected
Checking Python compatibility
Checking other ESP-IDF version.
Adding ESP-IDF tools to PATH...
Not using an unsupported version of tool openocd-esp32 found in PATH: 0.11.0.
WARNING: The following issue occurred while accessing the ESP-IDF version file in the Python environment: [Errno 2] No such file or directory: '/home/jeremy/.espressif/python_env/idf5.3_py3.10_env/idf_version.txt'. (Diagnostic information. It can be ignored.)
Checking if Python packages are up to date...
ERROR: /home/jeremy/.espressif/python_env/idf5.3_py3.10_env/bin/python doesn't exist! Please run the install script or "idf_tools.py install-python-env" in order to create it
```

### More Information.

The .espressif folder has a idf5.1_py3.10_env which is clearly the wrong version since the 5.3 IDF is looking for 5.3.

readmodifywrite, 2024-08-01:
To get the build to work, you can rename the 5.1 to 5.3 in .espressif.
However, then esptool.py will fail, because the 5.3 esptool is looking for the 5.1 Python.

So the full workaround is as follows:
Install the IDF from version *5.1*
Then install 5.3.  You should have a 5.1 Python in .espressif/python_env.
In .espressif/python_env, copy the 5.1 Python and rename it to 5.3.  Leave both copies.

dobairoland, 2024-08-02:
Hi @readmodifywrite. 

- Are you running `install.sh` in a clean terminal where previous version of ESP-IDF wasn't activated yet?
- Could you check if you've set any ESP-IDF environment variable permanently which are set globally and for all terminal windows?

dobairoland, 2024-08-19:
I'm closing this for inactivity.

closed 2024-08-19 by dobairoland

readmodifywrite, 2024-09-16:
Hi,

I run install.sh in a clean terminal.

I do not have any env vars set.  My build scripts set up clean environments each time.

The IDF simply installs the wrong version.

dobairoland, 2024-09-17:
Could you share the output of `export` run in a clean terminal? I cannot think of anything else other than there must be something set which interferes. 

> v5.3-256-gdc7fb34fca

How did you get this? This is the output of `git describe` in the IDF_PATH directory?