espressif/esp-idf #14294: Broken build after [fix/component_manager_load_all_local_components_v5.2]

metarutaiga, 2024-08-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2

### Operating System used.

macOS

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Build succeed.

### What is the actual behavior?

fatal error: freertos/FreeRTOS.h: No such file or directory

### Steps to reproduce.

git reset 3e45883091ca80318c218727f9b1c1e0e093830c --hard

### Build or installation Logs.

```plain
[1/640] Building C object esp-idf/esp_wifi/CMakeFiles/__idf_esp_wifi.dir/src/smartconfig_ack.c.obj
FAILED: esp-idf/esp_wifi/CMakeFiles/__idf_esp_wifi.dir/src/smartconfig_ack.c.obj
/Users/metarutaiga/Documents/esp32/esp-idf/components/esp_wifi/src/smartconfig_ack.c:12:10: fatal error: freertos/FreeRTOS.h: No such file or directory
   12 | #include "freertos/FreeRTOS.h"
      |          ^~~~~~~~~~~~~~~~~~~~~
```

### More Information.

After [pip install idf-component-manager==2.0.0.dev1]

[86/349] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/aesni.c.obj
FAILED: esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/aesni.c.obj 
/Users/metarutaiga/Documents/esp32/esp-idf/components/mbedtls/port/include/aes/esp_aes.h:27:10: fatal error: hal/aes_types.h: No such file or directory
   27 | #include "hal/aes_types.h"
      |          ^~~~~~~~~~~~~~~~~

kumekay, 2024-08-05:
Hi @metarutaiga 
Thank you for opening the issue,
I tried to reproduce the issue following your steps, but it worked smoothly with `2.0.0.dev1`
Could you please share a bit more information about your project to help us reproduce? Like `CMakeLists.txt` of your project

Could you please also try with the latest version from the main branch? https://github.com/espressif/idf-component-manager?tab=readme-ov-file#installing-a-development-version-of-the-component-manager

cristianfunes79, 2024-08-18:
I also tried the same but in linux `Linux ubuntu-ThinkPad-T14s-Gen-1 6.8.0-40-generic #40~22.04.3-Ubuntu SMP PREEMPT_DYNAMIC Tue Jul 30 17:30:19 UTC 2 x86_64 x86_64 x86_64 GNU/Linux` and it doesn't allow me to run a build after running `pip install idf-component-manager==2.0.0.dev1`. When I try to build it shows this message:
```sh
The following Python requirements are not satisfied:
Requirement 'idf-component-manager~=1.2' was not met. Installed version: 2.0.0.dev1
To install the missing packages, please run "install.sh"
Diagnostic information:
    IDF_PYTHON_ENV_PATH: /home/ubuntu/.espressif/python_env/idf5.2_py3.11_env
    Python interpreter used: /home/ubuntu/.espressif/python_env/idf5.2_py3.11_env/bin/python
Constraint file: /home/ubuntu/.espressif/espidf.constraints.v5.2.txt
Requirement files:
 - /home/ubuntu/esp/esp-idf/tools/requirements/requirements.core.txt
Python being checked: /home/ubuntu/.espressif/python_env/idf5.2_py3.11_env/bin/python

ESP-IDF v5.2.2-143-g3e45883091

```

kumekay, 2024-08-19:
@cristianfunes79 

The final version of the component manager 2.0 was released last week for the master branch and we will update the constraint files today to make it work. 

The current work around is to disable the checks fist:

```
export IDF_PYTHON_CHECK_CONSTRAINTS=no
pip install -U idf-component-manager
```

The installed version should be 2.0.1

kumekay, 2024-09-02:
Hello @cristianfunes79, did you manage to solve the issue?

Alvin1Zhang, 2025-03-06:
@cristianfunes79 hi, can you please help share if any updates for the issue? Thanks.

cristianfunes79, 2025-03-06:
> [@cristianfunes79](https://github.com/cristianfunes79) hi, can you please help share if any updates for the issue? Thanks.

Hi I'm working on that, I'll update the results.