espressif/esp-idf #14295: ethernet examples: Wrong chip ID for IP101 on P4 EV board

ftab, 2024-08-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### Espressif SoC revision.

ESP32-P4 v0.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-P4 EV board

### Power Supply used.

USB

### What is the expected behavior?

Run ethernet examples

### What is the actual behavior?

Crashes with 'wrong chip ID'

### Steps to reproduce.

1. cd examples/ethernet/iperf
2. idf.py set-target esp32p4
3. idf.py build flash monitor

### Debug Logs.

```plain
E (507) ip101: ip101_init(197): wrong chip ID
E (507) esp_eth: esp_eth_driver_install(229): init phy failed
E (507) example_eth_init: eth_init_internal(100): Ethernet driver install failed
E (597) example_eth_init: example_eth_init(260): internal Ethernet init failed
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x4000ca1a
0x4000ca1a: register_ethernet at /home/build/esp/esp-adf/esp-idf/examples/ethernet/iperf/main/cmd_ethernet.c:213 (discriminator 1)

file: "/IDF/examples/ethernet/iperf/main/cmd_ethernet.c" line 213
func: register_ethernet
expression: example_eth_init(&s_eth_handles, &s_eth_port_cnt)

abort() was called at PC 0x4ff0749d on core 0
```

### More Information.

My understanding from https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32p4/esp32-p4-function-ev-board/user_guide.html#block-diagram is that the P4 EV board is using IP101 so that is the PHY selected in the menuconfig (which was already selected by default).

kostaond, 2024-08-08:
@ftab what is HW revision of your board?

ftab, 2024-08-08:
> @ftab what is HW revision of your board?

@kostaond EV board v1.4 with ESP32-P4 v0.1 chip

kostaond, 2024-08-08:
It seems the newer versions of the board use different GPIOs. Please try to configure MDIO to GPIO52 and PHY reset to GPIO51 in Menuconfig.

ftab, 2024-08-08:
That did the trick! Along with 514d8cb to silence the insufficient TX buffer size log, the iperf example runs great.

Thanks!

closed 2024-08-08 by ftab

kostaond, 2024-08-08:
I'll fix the default examples configuration. I can imagine it has potential to confuse people.