espressif/esp-idf #14296: ld fails with `The gap between .flash.rodata and .eh_frame_hdr must not exist to produce the final bin image.`

Equidamoid, 2024-08-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

The project compiles.

### What is the actual behavior?

When building the project for esp32-c3 the linker fails with:
```
/home/build/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: The gap between .flash.rodata and .eh_frame_hdr must not exist to produce the final bin image.
/home/build/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: warning: blink.elf has a LOAD segment with RWX permissions
```

The same project builds just fine for the esp32 and esp32-s2 targets.

### Steps to reproduce.

1. Having an existing project that uses cmake (`idf_build_process` / `idf_build_executable`)
2. run cmake with `-DCMAKE_TOOLCHAIN_FILE=/path-to-esp-idf-v5/tools/cmake/toolchain-esp32c3.cmake`
3. `ninja flash`

Unfortunately, I don't have a minimal shareable reproduction scenario.

### Build or installation Logs.

```plain
[5/8] Linking CXX executable blink.elf                                                                                                                         
FAILED: blink.elf                                                                                                                                              
: && /home/build/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/riscv32-esp-elf-g++ -march=rv32imc_zicsr_zifencei -nostartfiles -march=rv32imc_zicsr_zifencei --specs=nosys.specs -Wl,--eh-frame-hdr -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32c3.peripherals.ld -T esp32c3.rom.ld -T esp32c3.rom.api.ld -T esp32c3.rom.libgcc.ld -T esp32c3.rom.version.ld -T esp32c3.rom.eco3.ld -T esp32c3.rom.newlib.ld -T memory.ld -T sections.ld -T rom.api.ld 
<a bunch of .obj and .a files> 
-Wl,--undefined=FreeRTOS_openocd_params  -u app_main  -u start_app  -u __ubsan_include  -u esp_system_include_startup_funcs  -u __cxa_guard_dummy  -u __cxx_init_dummy  -lstdc++  esp-idf/pthread/libpthread.a  esp-idf/cxx/libcxx.a  -lc  -lm  -lgcc  esp-idf/newlib/libnewlib.a  -u newlib_include_heap_impl  -u newlib_include_syscalls_impl  -u newlib_include_pthread_impl  -u newlib_include_assert_impl  -u newlib_include_init_funcs  -u __assert_func  -u esp_app_desc  -u esp_efuse_startup_include_func  -u esp_timer_init_include_func  -u uart_vfs_include_dev_init  -u usb_serial_jtag_vfs_include_dev_init  -u usb_serial_jtag_connection_monitor_include  -u esp_system_include_coredump_init  -u vfs_include_syscalls_impl  -u include_esp_phy_override  -lphy  -lbtbb  esp-idf/esp_phy/libesp_phy.a  -lphy  -lbtbb  esp-idf/esp_phy/libesp_phy.a  -lphy  -lbtbb  -u pthread_include_pthread_impl  -u pthread_include_pthread_cond_var_impl  -u pthread_include_pthread_local_storage_impl  -u pthread_include_pthread_rwlock_impl  -u pthread_include_pthread_semaphore_impl  -u esp_vfs_include_console_register && :
/home/build/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: The gap between .flash.rodata and .eh_frame_hdr must not exist to produce the final bin image.
/home/build/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: warning: blink.elf has a LOAD segment with RWX permissions
collect2: error: ld returned 1 exit status
```
```

Lapshin, 2024-08-10:
@Equidamoid , I could not reproduce it with `blink` example and with your config. I also had to remove `CONFIG_PARTITION_TABLE_CUSTOM=y` to build it (this requires `partitions.csv` file you did not attach). Tested on commit 9a44ced6724707faf73d46fa0c10f46b44f84f3d

To know what exactly is wrong you may try two options:

- Enable `CONFIG_COMPILER_ORPHAN_SECTIONS_WARNING=y` configure option. This should warn you about stray sections you have for your project. Please add output here in case you have new linking warnings after enabling this.

- Remove [this check](https://github.com/espressif/esp-idf/blob/master/components/esp_system/ld/esp32c3/sections.ld.in#L350) and build your app. After the elf file is generated you can see what data was placed between the sections in `build/*.map` file. Please attach a map file you get if it is possible. I would like to take a look and provide the next steps to fix the issue.

Equidamoid, 2024-08-10:
@Lapshin, oh, it's not the blink example. The cmake boilerplate stuff was initially copypasted from the example and nobody bothered to rename the binary since it doesn't affect anything.

Here is the partition table:
```
nvs,      data, nvs,       0x9000,  0x4000,
otadata,  data, ota,       0xd000,  0x2000,
phy_init, data, phy,       0xf000,  0x1000,
ota_0,    app,  ota_0,  0x10000,  1984K,
ota_1,    app,  ota_1,   0x210000,  1984K,
```

I added `CONFIG_COMPILER_ORPHAN_SECTIONS_WARNING=y` to the config, and commented out the line you mentioned.
The build progressed further and failed with "the binary is bigger than the partition". With some size optimization flags set the build succeeded. Here is the resulting defconfig.
```
CONFIG_BOOTLOADER_WDT_TIME_MS=60000
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_COMPILER_OPTIMIZATION_SIZE=y
CONFIG_COMPILER_OPTIMIZATION_ASSERTIONS_SILENT=y
CONFIG_COMPILER_OPTIMIZATION_CHECKS_SILENT=y
CONFIG_COMPILER_CXX_EXCEPTIONS=y
CONFIG_COMPILER_CXX_RTTI=y
CONFIG_COMPILER_ORPHAN_SECTIONS_WARNING=y
CONFIG_ESP_HTTPS_OTA_ALLOW_HTTP=y
CONFIG_ESP_PHY_MAX_WIFI_TX_POWER=10
CONFIG_ESP_MAIN_TASK_STACK_SIZE=8192
CONFIG_ESP_INT_WDT_TIMEOUT_MS=3000
CONFIG_ESP_TASK_WDT_PANIC=y
# CONFIG_ESP_WIFI_NVS_ENABLED is not set
# CONFIG_ESP_WIFI_ENABLE_WPA3_SAE is not set
CONFIG_ESP_COREDUMP_ENABLE_TO_UART=y
# CONFIG_MQTT_TRANSPORT_SSL is not set
# CONFIG_MQTT_TRANSPORT_WEBSOCKET is not set
```

 - I don't see any new warnings from ld. Only `ld: warning: blink.elf has a LOAD segment with RWX permissions` which was already present before.
 - The `.map` file didn't appear anywhere in the build dir (at least after `ninja app`), do I need to make extra changes and/or use different build target?

Equidamoid, 2024-08-19:
@Lapshin 
Done! (renamed to "blink.txt" because github)
[blink.txt](https://github.com/user-attachments/files/16663204/blink.txt)

Lapshin, 2024-08-20:
@Equidamoid , is your project built with `-fpie` or `-fpic` compile option? In case you don't know please grep build logs for it

Equidamoid, 2024-08-20:
Indeed, a dependency (shared between esp and linux builds) has -fpic enabled.

Lapshin, 2024-08-20:
@Equidamoid , could you enable this only for linux builds? Seems it does not make sense to enable it for esp builds

Equidamoid, 2024-08-20:
Without -fPIC I don't get the error anymore (and no, I didn't forget to uncomment it in the ld script :P).

Is this a fix or a workaround?

Lapshin, 2024-08-20:
@Equidamoid , I think it's the solution for you. Because -fpic is usually used for linking with shared libraries OR if you want to embed your static library with position-independent code inside. In case you build a monolith application from sources to run on esp32 chip I don't see any advantages in using this flag

BTW we wanted to distribute a library compiled with PIC but riscv linker is not ready for that https://sourceware.org/bugzilla/show_bug.cgi?id=31969

I'm closing this issue. If you have any others, please open a new one

closed 2024-08-21 by Lapshin

Equidamoid, 2024-08-21:
Understood. It is a solution for me indeed. And now others may be able to google this bug to find it =)

Will it be worth documenting somewhere? I can imagine more people may `add_subdirectory` some sloppy-written code with `-fPIC` just hardcoded because of a mix of static and shared libs.

Lapshin, 2024-08-21:
@Equidamoid , thank you for your concern! I have already created a merge request in our internal repository to detect such configurations and provide a human-readable explanation to terminal if any are found.