espressif/esp-idf #14297: esp32-p4-function-ev-board 开发板运行i2s_es8311运行不正常

666wsp, 2024-08-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-beta2-281-g44f58ecb51

### Espressif SoC revision.

ESP32-P4 chip revision: v0.1

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32-p4-function-ev-board

### Power Supply used.

USB

### What is the expected behavior?

在运行I2S_ES8311实例时候使用功能没法正常运行，Example mode设置为music时喇叭也没发音，设置为echo时也没有捕获到外部音频在喇叭中播放

### What is the actual behavior?

运行程序时喇叭会有发音

### Steps to reproduce.

1、我使用自带的出厂固件运行测试喇叭能正常工作，证明ES8311和功放电路完全没有问题
2、在运行I2S_ES8311demo时候我使用逻辑分析I2C通讯正常
3、将功放芯片PA_CTRL引脚设置为高电平输出打开功放输出喇叭也没有正常发音

### Debug Logs.

```plain
ESP-ROM:esp32p4-eco1-20240205
Build:Feb  5 2024
rst:0x1 (POWERON),boot:0x30f (SPI_FAST_FLASH_BOOT)
SPI mode:DIO, clock div:1
load:0x4ff33ce0,len:0x17e4
load:0x4ff2abd0,len:0xc38
load:0x4ff2cbd0,len:0x2fdc
entry 0x4ff2abda
I (25) boot: ESP-IDF v5.3-beta2-281-g44f58ecb51 2nd stage bootloader
I (26) boot: compile time Aug  4 2024 10:11:05
I (26) boot: Multicore bootloader
I (32) boot: chip revision: v0.1
I (34) boot.esp32p4: SPI Speed      : 80MHz
I (39) boot.esp32p4: SPI Mode       : DIO
I (44) boot.esp32p4: SPI Flash Size : 16MB
I (48) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  2 factory          factory app      00 00 00010000 00100000
I (88) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=40020020 size=a7f88h (688008) map
I (249) esp_image: segment 1: paddr=000b7fb0 vaddr=30100000 size=0000ch (    12) load
I (251) esp_image: segment 2: paddr=000b7fc4 vaddr=3010000c size=00038h (    56) load
I (257) esp_image: segment 3: paddr=000b8004 vaddr=4ff00000 size=08014h ( 32788) load
I (273) esp_image: segment 4: paddr=000c0020 vaddr=40000020 size=1d400h (119808) map
I (301) esp_image: segment 5: paddr=000dd428 vaddr=4ff08014 size=070e0h ( 28896) load
I (310) esp_image: segment 6: paddr=000e4510 vaddr=4ff0f100 size=01a20h (  6688) load
I (316) boot: Loaded app from partition at offset 0x10000
I (317) boot: Disabling RNG early entropy source...
I (330) cpu_start: Multicore app
W (340) clk: esp_perip_clk_init() has not been implemented yet
I (347) cpu_start: Pro cpu start user code
I (347) cpu_start: cpu freq: 360000000 Hz
I (348) app_init: Application information:
I (350) app_init: Project name:     i2s-es8311-example
I (356) app_init: App version:      1
I (360) app_init: Compile time:     Aug  4 2024 10:18:09
I (366) app_init: ELF file SHA256:  346245f08...
I (372) app_init: ESP-IDF:          v5.3-beta2-281-g44f58ecb51
I (378) efuse_init: Min chip rev:     v0.1
I (383) efuse_init: Max chip rev:     v0.99 
I (388) efuse_init: Chip rev:         v0.1
I (393) heap_init: Initializing. RAM available for dynamic allocation:
I (400) heap_init: At 4FF12200 len 00028DC0 (163 KiB): RAM
I (406) heap_init: At 4FF3AFC0 len 00004BF0 (18 KiB): RAM
I (412) heap_init: At 4FF40000 len 00060000 (384 KiB): RAM
I (418) heap_init: At 50108080 len 00007F80 (31 KiB): RTCRAM
I (425) heap_init: At 30100044 len 00001FBC (7 KiB): TCM
I (431) spi_flash: detected chip: generic
I (435) spi_flash: flash io: dio
W (439) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (450) main_task: Started on CPU0
I (460) main_task: Calling app_main()
i2s es8311 codec example start
-----------------------------
I (470) i2s_es8311: i2s driver init success
I (490) ES8311: ES8311 in Slave mode and I2S format
I (500) i2s_es8311: es8311 codec init success
I (500) gpio: GPIO[53]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (500) main_task: Returned from app_main()
I (10420) i2s_es8311: [music] i2s music played, 634240 bytes are written.
I (21340) i2s_es8311: [music] i2s music played, 640000 bytes are written.
I (32240) i2s_es8311: [music] i2s music played, 640000 bytes are written.
I (43160) i2s_es8311: [music] i2s music played, 640000 bytes are written.
```

suda-morris, 2024-08-05:
> 我使用自带的出厂固件运行测试喇叭能正常工作，证明ES8311和功放电路完全没有问题

@666wsp have you checked this code? 

https://github.com/espressif/esp-dev-kits/blob/master/esp32-p4-function-ev-board/examples/common_components/esp32_p4_function_ev_board/src/esp32_p4_function_ev_board.c

Y1hsiaochunnn, 2024-08-05:
Hi, @suda-morris ,I think he was using this demo and I had the same problem using this routine 

https://github.com/espressif/esp-idf/tree/master/examples/peripherals/i2s/i2s_codec/i2s_es8311

666wsp, 2024-08-05:
@Y1hsiaochunnn Yes, you're right, this demo doesn't seem to be running correctly for esp32-p4 at the moment

L-KAYA, 2024-08-07:
Swap the I2S DI and DO GPIO can help, thanks for reporting!

Y1hsiaochunnn, 2024-08-09:
Hi,@L-KAYA ,I'm running the above I2S demo with ESP32-P4 Fuction_EV_Board V1.4 but it doesn't work, I checked the pin definition and it seems to be right

closed 2024-08-14 by espressif-bot

Ncouch64, 2024-12-17:
I just got in my ESP32-P4 board and I am having the same problem. The example builds and runs, but there is no audio coming from speaker in either of the two modes.

L-KAYA, 2024-12-18:
Please refer to [this comment](https://github.com/espressif/esp-idf/issues/15047#issuecomment-2550175991) in #15047 if you still fail to get the audio by **speaker**, but success by earphone.

The P4 boards disable the PA by default. Please try to pull up the `PA_CTRL` to enable PA first.