espressif/esp-idf #14298: HTTPS Example fails on site "speechgen.io" since ESP-IDF v.5.x

kaloprojects, 2024-08-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Using Arduino IDE, latest espressif/arduino-esp32 Release v3.0.4 (based on ESP-IDF v5.1.4+).

Since update to IDF 5.x the connect to https server "speechgen.io" fails (SpeechToText API service). _Using older IDF v.4 the same code still connecting successfully to server._ I am also connecting to other https server with same code snippet (e.g. "api.openai.com" ), they work also on IDF v5.x. So it seems an dedicate issue on "speechgen.io" ?

The orginal post is posted in https://github.com/espressif/arduino-esp32/issues/10071, the Admin folks recommended to post it here (IDF Https library), as we assume an IDF issue: _The server is asking for a cipher renegotiation, and that is failing. It works fine if TLS 1.3 is enabled in IDF 5.1, but I have no idea why it succeeds in IDF 4.4. This issue comes from upstream, so you can open an issue at https://github.com/espressif/esp-idf/issues and tell them you have a problem with the [https_request example](https://github.com/espressif/esp-idf/tree/master/examples/protocols/https_request) when you change to the URL/server you want. 

So even i am using Arduino IDE only .. hoping you find root cause why the Https example fails on "sppeechgen.io" ?

Thank you !

```cpp
// Https Connect to Speechgen server "speechgen.io" port 443 
// -> always fails with Arduino-esp32 v. 3.x library (but same code result success with earlier Arduino-esp32 v. 2.x )
// -> other server (e.g. "api.openai.com" are working (also with 3.x)

#include <WiFi.h>           
#include <WiFiClientSecure.h> 
/* .. */
WiFiClientSecure client;  
client.setInsecure();                          
/* .. */  
if ( client.connect("speechgen.io", 443) )   // <- it FAILS here always (with ESP 3.x,/IDF v.5.x, only working with 2.x/v.4.x)
{  Serial.println("> Succesfully connected to speeschgen.io port 443"); 
}
... 
```

AdityaHPatwardhan, 2024-08-07:
Hi @kaloprojects Thanks for the issue.
This is due to the fact that the `MBEDTLS_HARDWARE_AES` is turned on and the required cipher is not supported by the hardware.
This bug has already been fixed with https://github.com/espressif/esp-idf/commit/2c85399da0eb16dc7edc3afcf8be250a7b6bcc44

This has been kept disabled by default on older branches to avoid silently increasing the flash footprint for users.
Please enable the fix using `Component Config -> mbedTLS -> Enable support for non-AES ciphers in GCM operation`
After that the handshake should succeed.

kaloprojects, 2024-08-07:
Awesome, thank you for this great feedback, this helps ! . I personally do not know how to customize (as i am using arduino-esp32 library, but pretty sure the Admin folks there (https://github.com/espressif/arduino-esp32/issues/10071,) can fix it with your answer) :)

thank you @AdityaHPatwardhan ! (i will follow up if they need any details more)

AdityaHPatwardhan, 2024-08-12:
Hi @kaloprojects I will go ahead and close this issue,
Please feel free to re-open if required.
Thanks,
Aditya

closed 2024-08-12 by AdityaHPatwardhan

AxelLin, 2024-08-12:
> This has been kept disabled by default on older branches to avoid silently increasing the flash footprint for users. Please 

I'm wondering if this should be enabled by default.
Otherwise, people may not aware of this bug until got user report. (And the solution is to enable it.)

AdityaHPatwardhan, 2024-08-13:
@AxelLin I understand, so we thought it as a tradeoff with probably of issue occurrence and increasing the binary size.
We decided to keep it disabled based on following considerations
1) Based on the feedback that we received, this cipher is rarely used.
2) The increased code size is considerable for the application (roughly 3KB)