espressif/esp-idf #14299: Can't generate SEGGER SystemView heap log

Massimiliano-solutiontech, 2024-08-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I’m using `esp-idf v5.3` with an `ESP32-C6 WROOM devkit`, and I would like to track the heap using `SEGGER SystemView`. However, when I launch the `gdbinit` script, it never stops (even though my code includes `heap_trace_stop()`), and the generated `svdat` file is empty except for the header. On the other hand, if I don’t track the heap and generate the binary log via `esp sysview start`, the file is correctly generated. However, if the log becomes too large, SystemView hangs during analysis

### gdbinit

```gdb

set pagination off
target remote :3333

mon reset halt
maintenance flush register-cache

tb heap_trace_start
commands
mon esp sysview start file://heap_log.svdat
c
end

tb heap_trace_stop
commands
mon esp sysview stop
end

c

```

### heap_log.svdat

```svdat
;
; Version     SEGGER SystemViewer V2.42
; Author      Espressif Inc
;

```

erhankur, 2024-08-05:
@Massimiliano-solutiontech Thanks for reporting the issue. I will test it and get back to you.

erhankur, 2024-08-07:
@Massimiliano-solutiontech It worked for me with the IDF 5.3 release, OpenOCD `Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:28)` and `GNU gdb (esp-gdb) 14.2_20240403`

Can you enable OpenOCD verbose log and attach the tog.txt ? 

```
set pagination off
target remote :3333

mon debug_level 3
mon log_output log.txt

mon reset halt
maintenance flush register-cache

tb heap_trace_start
commands
mon esp sysview start file:///tmp/heap_log.svdat
c
end

tb heap_trace_stop
commands
mon esp sysview stop
end

c
```

erhankur, 2025-03-07:
Closed due to inactivity.

closed 2025-03-07 by erhankur