espressif/esp-idf #14300: Metadata magic word does not match

JasonYan324, 2024-08-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

IDF5.2.2(v202212.00-98-gee2901e) & IDF5.3

### Espressif SoC revision.

ESP32-C3 (QFN32) (revision v0.4)

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-C3-DevKit-RUST-1

### Power Supply used.

USB

### What is the expected behavior?

python managed_components/espressif__esp_secure_cert_mgr/tools/configure_esp_secure_cert.py -p "COM19" --keep_ds_data_on_host --ca-cert "./main/certs/aws-root-ca.pem" --device-cert "./main/certs/certificate.pem.crt" --private-key "./main/certs/private.pem.key" --target_chip "esp32c3" --secure_cert_type cust_flash --priv_key_algo "RSA" "2048"

### What is the actual behavior?

E (392) esp_secure_cert: Metadata magic word does not match
E (392) esp_secure_cert: Error in reading the metadata
E (402) main: Error in getting device certificate. Error: ESP_FAIL
E (412) main: Error in getting CA certificate. Error: ESP_FAIL

### Steps to reproduce.

1.Run the following command to create and flash the certificate partition.
python managed_components/espressif__esp_secure_cert_mgr/tools/configure_esp_secure_cert.py -p "COM10" --keep_ds_data_on_host --ca-cert "./main/certs/aws-root-ca.pem" --device-cert "./main/certs/certificate.pem.crt" --private-key "./main/certs/private.pem.key" --target_chip "esp32c3" --secure_cert_type cust_flash --priv_key_algo "RSA" "2048"
2.ouput:
D:\99ESP\T3\iot-reference-esp32c3>python managed_components/espressif__esp_secure_cert_mgr/tools/configure_esp_secure_cert.py -p "COM10" --keep_ds_data_on_host --ca-cert "./main/certs/aws-root-ca.pem" --device-cert "./main/certs/certificate.pem.crt" --private-key "./main/certs/private.pem.key" --target_chip "esp32c3" --secure_cert_type cust_flash --priv_key_algo "RSA" "2048"
Using the same hmac key burned in efuse BLOCK_KEY1
Flashing the esp_secure_cert partition at 0xD000 offset
Note: You can skip this step by providing --skip_flash argument
esptool.py v4.7.0
Serial port COM10
Connecting...
Chip is ESP32-C3 (QFN32) (revision v0.4)
Features: WiFi, BLE, Embedded Flash 4MB (XMC)
Crystal is 40MHz
MAC: 40:4c:ca:89:c1:84
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x0000d000 to 0x00012fff...
Compressed 24576 bytes to 3339...
Writing at 0x0000d000... (100 %)
Wrote 24576 bytes (3339 compressed) at 0x0000d000 in 0.3 seconds (effective 660.7 kbit/s)...
Hash of data verified.
Leaving...
Hard resetting via RTS pin...
3. then build and flash the demo project by running idf.py -p "COM10" flash monitor
4. then monitoring the demo, and error output:
E (392) esp_secure_cert: Metadata magic word does not match
E (392) esp_secure_cert: Error in reading the metadata
E (402) main: Error in getting device certificate. Error: ESP_FAIL
E (412) main: Error in getting CA certificate. Error: ESP_FAIL
5. then in the function "esp_secure_cert_read_metadata" block verification just like this:
image
//if (metadata->magic_word != ESP_SECURE_CERT_METADATA_MAGIC_WORD) {
// ESP_LOGE(TAG, "Metadata magic word does not match");
// return ESP_FAIL;
//}
6. and the running out change to this(Private Key's Length is wrong, and the value can not be read ):
-----END CERTIFICATE-----
W (622) mmap: paddr block is mapped already, vaddr_start: 0x3c120000, size: 0x10000
W (632) mmap: paddr block is mapped already, vaddr_start: 0x3c120000, size: 0x10000
I (642) main:
Private Key:
Length: 1200
��=�����Lc_m�t�� �m���O��5�p3պ0ؤ���1\��ζI#��״�7�n�h�2M��%�Z��Zt��>]K���2S���s$ս�N[�ŨM7�[�eG�7�(��>u ���1{g-4b�Ym��7�:���]^�65?=
��q!���>_-��jx1(��$�p<�+�y�6c��Cآ�3���w�
,4�D�J� _�
I (672) gpio: GPIO[8]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (672) main:
CS Cert:
Length: 0

I (672) temp_sub_pub_and_led_control_demo: Sending subscribe request to agent for topic filter: /filter/TempSubPubLED with id 1
I (682) ota_over_mqtt_demo: OTA over MQTT demo, Application version 0.0.0

....

I (4402) main_task: Returned from app_main()
I (4412) wifi:idx:1 (ifx:0, 3e:06:a7:1b:0a:35), tid:0, ssn:279, winSize:64
E (4702) esp-tls-mbedtls: mbedtls_pk_parse_keyfile returned -0x3D00
E (4702) esp-tls-mbedtls: Failed to set client pki context
E (4702) esp-tls-mbedtls: Failed to set client configurations, returned [0x8019] (ESP_ERR_MBEDTLS_PK_PARSE_KEY_FAILED)
E (4712) esp-tls: create_ssl_handle failed
E (4712) esp-tls: Failed to open new connection
I (4742) ota_over_mqtt_demo: Received: 0 Queued: 0 Processed: 0 Dropped: 0
I (5032) core_mqtt_agent_manager: Retry attempt 1.
E (5362) esp-tls-mbedtls: mbedtls_pk_parse_keyfile returned -0x3D00
E (5362) esp-tls-mbedtls: Failed to set client pki context
E (5372) esp-tls-mbedtls: Failed to set client configurations, returned [0x8019] (ESP_ERR_MBEDTLS_PK_PARSE_KEY_FAILED)
E (5382) esp-tls: create_ssl_handle failed
E (5382) esp-tls: Failed to open new connection

### Debug Logs.

```plain
E (392) esp_secure_cert: Metadata magic word does not match
E (392) esp_secure_cert: Error in reading the metadata
E (402) main: Error in getting device certificate. Error: ESP_FAIL
E (412) main: Error in getting CA certificate. Error: ESP_FAIL
```

### More Information.

1.Use any other new PC for the same test, got the same result.
2.IDF5.3 test got the same result.
3.IDF5.2 test, esp_secure_cert is ok, but can not run
4.template project: https://github.com/FreeRTOS/iot-reference-esp32c3

AdityaHPatwardhan, 2024-08-06:
Hi @JasonYan324,
In the command to generate the esp_secure_cert partition on host, can you please use `cust_flash_tlv` as the esp_secure_cert partition type. I see that it is set to `cust_flash` in your case. `cust_flash` is a legacy flash format that has been deprecated now.
I think the note in demo example may need an update.

The https://github.com/espressif/esp_secure_cert_mgr component uses `cust_flash_tlv` format by default. If you still want to use cust_flash format for legacy reasons, then please set `Component Config-> esp_secure_cert_mgr -> Enable support for legacy formats` in menuconfig to enable legacy partition formats.

JasonYan324, 2024-08-07:
please update the note in demo example， and let me know how to use the new cmd，thanks！

AdityaHPatwardhan, 2024-08-07:
Hi @JasonYan324, the note has been updated https://github.com/FreeRTOS/iot-reference-esp32c3/pull/105

JasonYan324, 2024-08-08:
> Hi @JasonYan324, the note has been updated [FreeRTOS/iot-reference-esp32c3#105](https://github.com/FreeRTOS/iot-reference-esp32c3/pull/105)

Thanks

AdityaHPatwardhan, 2024-08-08:
I will go ahead and close this issue, Please feel free to re-open if required.
Thanks,
Aditya

closed 2024-08-08 by AdityaHPatwardhan