espressif/esp-idf #14301: Clicking the vscode IDF plugin will not use the MSYS2 default terminal

JDingOneSelf, 2024-08-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

xxxx@LAPTOP-8EUDJHTE  /d/xxxxx/01Work/01Demo/01ESP32S3_Demo/blink
$ & set IDF_PATH='d:\Program\Espressif\IDF\v4.4.5\esp-idf'
bash: syntax error near unexpected token `&'

xxxx@LAPTOP-8EUDJHTE  /d/SaiSiYun/01Work/01Demo/01ESP32S3_Demo/blink
$ & 'd:\Program\Espressif\445\python_env\idf4.4_py3.8_env\Scripts\python.exe' 'd:\Program\Espressif\IDF\v4.4.5\esp-idf\tools\idf_monitor.py' -p COM4 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 'd:\xxxxx\01Work\01Demo\01ESP32S3_Demo\blink\build\blink.elf' 
bash: syntax error near unexpected token `&'

brianignacio5, 2024-08-06:
Are you talking about https://github.com/espressif/vscode-esp-idf-extension ?

Why are you using MSYS2 ? Why don't use Powershell or Command Prompt (cmd.exe) ?

JDingOneSelf, 2024-08-06:
Yes, I am used to using MSYS2, and PowerShell and cmd have garbled characters, is there a solution?It worked before

brianignacio5, 2025-03-03:
This has been fixed in the latest release of https://github.com/espressif/vscode-esp-idf-extension 

Please give it a try.

Alvin1Zhang, 2025-03-07:
Thanks for reporting, will close now, feel free to create a new ticket in https://github.com/espressif/vscode-esp-idf-extension, thanks.

closed 2025-03-07 by Alvin1Zhang