espressif/esp-idf #14302: Get status code to Esp OTA HTTP

LouisMaiaDev, 2024-08-05:
### Is your feature request related to a problem?

By performance OTA HTTP to Esp, sometimes the function "esp_https_ota_perform()" don't return the status code. If you available a image firmware in a host server and remove it, the function show log "File not found", but continue trying download image.

### Describe the solution you'd like.

The solution is simple: create the public funtion esp_https_ota_get_status_code(esp_https_ota_handle_t * http_ota_handle) to allow get the last status code logged. This will allow the developer to make precise interventions in their code.

JasonYan324, 2024-08-06:
focus on this issue, wating for the reslut

nileshkale123, 2024-08-21:
@LouisMaiaDev 

Thank you for reporting this issue. 

I attempted to reproduce the behavior on my setup by hosting a local Python-based server using [python based server](https://github.com/espressif/esp-idf/tree/master/examples/system/ota#python-based-server) but I was unable to replicate the problem. 

Could you please share more details about your setup or any specific steps you followed that might help in reproducing the issue? This information will be crucial for us to investigate further.

LouisMaiaDev, 2024-08-25:
Yes, I can!
It's quite simple. Host the Esp firmware image on Google Drive or AWS (which is the way I'm using it) and launch Esp FOTA. If you still cannot reproduce the error, post here and I will publish the image that proves the existence of this error. As a solution, I had to implement the solution mentioned above in Espressif's HAL and this way I was able to intervene and handle this error appropriately.

nileshkale123, 2024-08-29:
Hello @LouisMaiaDev ,

Could you please share the respective OTA image? I could not reproduce the issue using the image uploaded to Google Drive, which is approximately 960KB in size.

Additionally, it would be greatly appreciated if you could share the patch detailing the changes you made to the file, as mentioned above.

nileshkale123, 2024-09-05:
Hi @LouisMaiaDev 

I have created a patch to address the issue. Please review it and let me know if this meets your requirements or if any additional changes are needed. 

[api_to_get_last_status_logged.txt](https://github.com/user-attachments/files/16884753/api_to_get_last_status_logged.txt)

Thanks.

LouisMaiaDev, 2024-09-05:
Show, good! Very good @nileshkale123 !

closed 2024-09-19 by espressif-bot