espressif/esp-idf #14308: How to create 2 sdmmc patitions in one emmc driver？

hibfh, 2024-08-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

esp32 module and idf is v5.2.2
```
static esp_err_t partition_card(const esp_vfs_fat_mount_config_t *mount_config,
                                const char *drv, sdmmc_card_t *card, BYTE pdrv)
{
    FRESULT res = FR_OK;
    esp_err_t err;
    const size_t workbuf_size = 4096;
    void* workbuf = NULL;
    ESP_LOGW(TAG, "partitioning card");

    workbuf = ff_memalloc(workbuf_size);
    if (workbuf == NULL) {
        return ESP_ERR_NO_MEM;
    }
 
    LBA_t plist[] = {90, 10, 0, 0};
    res = f_fdisk(pdrv, plist, workbuf);
    if (res != FR_OK) {
        err = ESP_FAIL;
    ESP_LOGD(TAG, "f_fdisk failed (%d)", res);
    goto fail;
    
    }
    
    size_t alloc_unit_size = esp_vfs_fat_get_allocation_unit_size(
                card->csd.sector_size,
                mount_config->allocation_unit_size);
    ESP_LOGW(TAG, "formatting card, allocation unit size=%d", alloc_unit_size);
    const MKFS_PARM opt = {(BYTE)FM_ANY, 0, 0, 0, alloc_unit_size};
    res = f_mkfs(drv, &opt, workbuf, workbuf_size);
    if (res != FR_OK) {
        err = ESP_FAIL;
        ESP_LOGD(TAG, "f_mkfs failed (%d)", res);
        goto fail;
    }

    free(workbuf);
    return ESP_OK;
fail:
    free(workbuf);
    return err;
}
static esp_err_t s_f_mount(sdmmc_card_t *card, FATFS *fs, const char *drv, uint8_t pdrv, const esp_vfs_fat_mount_config_t *mount_config)
{
    esp_err_t err = ESP_OK;
    FRESULT res = f_mount(fs, drv, 1);
    if (res != FR_OK) {
        err = ESP_FAIL;
        ESP_LOGW(TAG, "failed to mount %s card (%d)",drv, res);

        bool need_mount_again = (res == FR_NO_FILESYSTEM || res == FR_INT_ERR) && mount_config->format_if_mount_failed;
        if (!need_mount_again) {
            return ESP_FAIL;
        }

        err = partition_card(mount_config, drv, card, pdrv);
        if (err != ESP_OK) {
            return err;
        }

        ESP_LOGW(TAG, "mounting again");
        res = f_mount(fs, drv, 0);
        if (res != FR_OK) {
            err = ESP_FAIL;
            ESP_LOGD(TAG, "f_mount failed after formatting (%d)", res);
            return err;
        }
    }

    return ESP_OK;
}
static esp_err_t init_sdmmc_host(int slot, const void *slot_config, int *out_slot)
{
    *out_slot = slot;
    return sdmmc_host_init_slot(slot, (const sdmmc_slot_config_t*) slot_config);
}

static esp_err_t mount_to_vfs_fat(const esp_vfs_fat_mount_config_t *mount_config, sdmmc_card_t *card, uint8_t pdrv,
                                  const char *base_path, FATFS **out_fs)
{
    FATFS *fs = NULL;
    esp_err_t err;
    ff_diskio_register_sdmmc(pdrv, card);
    ff_sdmmc_set_disk_status_check(pdrv, mount_config->disk_status_check_enable);
    ESP_LOGD(TAG, "using pdrv=%i", pdrv);
    char drv[3] = {(char)('0' + pdrv), ':', 0};

    // connect FATFS to VFS
    err = esp_vfs_fat_register(base_path, drv, mount_config->max_files, &fs);
    *out_fs = fs;
    if (err == ESP_ERR_INVALID_STATE) {
        // it's okay, already registered with VFS
    } else if (err != ESP_OK) {
        ESP_LOGD(TAG, "esp_vfs_fat_register failed 0x(%x)", err);
        goto fail;
    }

    // Try to mount partition
    err = s_f_mount(card, fs, drv, pdrv, mount_config);
    if (err != ESP_OK) {
        goto fail;
    }
    return ESP_OK;

fail:
    if (fs) {
        f_mount(NULL, drv, 0);
    }
    esp_vfs_fat_unregister_path(base_path);
    ff_diskio_unregister(pdrv);
    return err;
}
static void call_host_deinit(const sdmmc_host_t *host_config)
{
    if (host_config->flags & SDMMC_HOST_FLAG_DEINIT_ARG) {
        host_config->deinit_p(host_config->slot);
    } else {
        host_config->deinit();
    }
}

esp_err_t mount_emmc_2_partitions(
                                  const sdmmc_host_t* host_config,
                                  const void* slot_config,
                                  const esp_vfs_fat_mount_config_t* mount_config,
                                  sdmmc_card_t** out_card)
{
    esp_err_t err;
    FATFS *fs1 = NULL;
    FATFS *fs2 = NULL;
    int card_handle = -1;   //uninitialized
    sdmmc_card_t* card = NULL;
    BYTE pdrv = FF_DRV_NOT_USED;
    char* dup_path1 = NULL;
    char* dup_path2 = NULL;
    bool host_inited = false;
    dup_path1 = strdup(MOUNT_POINT_DATA);
    dup_path2 = strdup(MOUNT_POINT_CFG);
    if(!dup_path1){
        ESP_LOGD(TAG, "could not copy %s",MOUNT_POINT_DATA);
        err = ESP_ERR_NO_MEM;
        goto cleanup;
    }
    if(!dup_path2){
        ESP_LOGD(TAG, "could not copy %s",MOUNT_POINT_CFG);
        err = ESP_ERR_NO_MEM;
        goto cleanup;
    }
    // not using ff_memalloc here, as allocation in internal RAM is preferred
    card = (sdmmc_card_t*)malloc(sizeof(sdmmc_card_t));
    if (card == NULL) {
        ESP_LOGD(TAG, "could not locate new sdmmc_card_t");
        err = ESP_ERR_NO_MEM;
        goto cleanup;
    }

    err = (*host_config->init)();
    CHECK_EXECUTE_RESULT(err, "host init failed");
    //deinit() needs to be called to revert the init
    host_inited = true;
    //If this failed (indicated by card_handle != -1), slot deinit needs to called()
    //leave card_handle as is to indicate that (though slot deinit not implemented yet.
    err = init_sdmmc_host(host_config->slot, slot_config, &card_handle);
    CHECK_EXECUTE_RESULT(err, "slot init failed");

    // probe and initialize card
    err = sdmmc_card_init(host_config, card);
    CHECK_EXECUTE_RESULT(err, "sdmmc_card_init failed");
    if (ff_diskio_get_drive(&pdrv) != ESP_OK || pdrv == FF_DRV_NOT_USED) {
        ESP_LOGD(TAG, "the maximum count of volumes is already mounted");
        err = ESP_ERR_NO_MEM;
        goto cleanup;
    }
    
    err = mount_to_vfs_fat(mount_config, card, pdrv, dup_path1, &fs1);
    CHECK_EXECUTE_RESULT(err, "mount_to_vfs pdrv1 failed");
    // char drv[3] = {(char)('0' + pdrv), ':', 0};
    // err = partition_card(mount_config, drv, card, pdrv);
    //     if (err != ESP_OK) {
    //         return err;
    //     }

    //     ESP_LOGW(TAG, "mounting again");
    //     err = f_mount(fs1, drv, 0);
    //     if (err != FR_OK) {
    //         err = ESP_FAIL;
    //         ESP_LOGD(TAG, "f_mount failed after formatting (%d)", err);
    //         return err;
    //     }
    err = mount_to_vfs_fat(mount_config, card, pdrv, dup_path2, &fs2);
    CHECK_EXECUTE_RESULT(err, "mount_to_vfs pdrv2 failed");
    if (out_card != NULL) {
        *out_card = card;
    } 
    uint64_t total,frees;
    esp_vfs_fat_info(MOUNT_POINT_DATA,&total,&frees);
    printf("MOUNT_POINT_DATA Total size: %llu,free size: %llu\n", total,frees);  
    esp_vfs_fat_info(MOUNT_POINT_CFG,&total,&frees);
    printf("MOUNT_POINT_CFG Total size: %llu,free size: %llu\n", total,frees);   
 
    return ESP_OK;
cleanup:
    if (host_inited) {
        call_host_deinit(host_config);
    }
    free(card);
    free(dup_path1);
    free(dup_path2);
    return err;
}
void emmc_init()
{
    ESP_LOGI(TAG, "Initializing eMMC");
    host.slot=SDMMC_HOST_SLOT_1;
    host.flags = SDMMC_HOST_FLAG_4BIT | SDMMC_HOST_FLAG_DDR;
    host.max_freq_khz = SDMMC_FREQ_52M;

    // This initializes the slot without card detect (CD) and write protect (WP) signals.
    // Other fields will be initialized to zero
    slot_config.cd = SDMMC_SLOT_NO_CD;
    slot_config.wp = SDMMC_SLOT_NO_WP;

    slot_config.width = 4;

    slot_config.flags |= SDMMC_SLOT_FLAG_INTERNAL_PULLUP;

    // //rst pull up
    if(gpio_set_level(GPIO_NUM_27, 0) == ESP_OK){
        ESP_LOGI(TAG, "RST Pull Up Success");
    }
    else{
        ESP_LOGE(TAG, "RST Pull Up Fail");
    }
}
void emmc_mount(){
    ESP_LOGI(TAG, "Mounting filesystem to 2 partitions");
    esp_err_t ret = custom_mount_emmc_2_partitions(&host, &slot_config, &mount_config, &card);

    if (ret != ESP_OK) {
        if (ret == ESP_FAIL) {
            ESP_LOGE(TAG, "Failed to mount filesystem. "
                     "If you want the eMMC to be formatted, set the EXAMPLE_FORMAT_IF_MOUNT_FAILED menuconfig option.");
        } else {
            ESP_LOGE(TAG, "Failed to initialize the eMMC (%s). "
                     "Make sure eMMC lines have pull-up resistors in place.", esp_err_to_name(ret));
        }
        return;
    }
    ESP_LOGI(TAG, "Filesystem mounted");

    // Card has been initialized, print its properties
    sdmmc_card_print_info(stdout, card);
}
```
esp_vfs_fat_info output seems the patitions are the same one.
Need Someone help,Thanks a lot.

hibfh, 2024-08-06:
I'm greenhand about fatfs,and not sure how to create two different size partitions

hibfh, 2024-08-06:
solve the problem by change drv params.

closed 2024-08-06 by hibfh

fmadhurakkariyan, 2024-08-15:
@hibfh I am having the same problem. Could I have the corrected code if you are willing to share?

hibfh, 2024-08-21:
Fine, @fmadhurakkariyan  here is the code:
```
esp_err_t custom_mount_emmc_2_partitions(
                                  const sdmmc_host_t* host_config,
                                  const void* slot_config,
                                  const esp_vfs_fat_mount_config_t* mount_config,
                                  sdmmc_card_t** out_card)
{
    esp_err_t err;
    FATFS *fs1 = NULL;
    FATFS *fs2 = NULL;
    int card_handle = -1;   //uninitialized
    sdmmc_card_t* card = NULL;
    BYTE pdrv = FF_DRV_NOT_USED;
    char* dup_path1 = NULL;
    char* dup_path2 = NULL;
    bool host_inited = false;
    dup_path1 = strdup(OBD_MOUNT_POINT_DATA);
    dup_path2 = strdup(OBD_MOUNT_POINT_CFG);
    if(!dup_path1){
        ESP_LOGD(TAG, "could not copy %s",OBD_MOUNT_POINT_DATA);
        err = ESP_ERR_NO_MEM;
        goto cleanup;
    }
    if(!dup_path2){
        ESP_LOGD(TAG, "could not copy %s",OBD_MOUNT_POINT_CFG);
        err = ESP_ERR_NO_MEM;
        goto cleanup;
    }
    // not using ff_memalloc here, as allocation in internal RAM is preferred
    card = (sdmmc_card_t*)malloc(sizeof(sdmmc_card_t));
    if (card == NULL) {
        ESP_LOGD(TAG, "could not locate new sdmmc_card_t");
        err = ESP_ERR_NO_MEM;
        goto cleanup;
    }

    err = (*host_config->init)();
    CHECK_EXECUTE_RESULT(err, "host init failed");
    //deinit() needs to be called to revert the init
    host_inited = true;
    //If this failed (indicated by card_handle != -1), slot deinit needs to called()
    //leave card_handle as is to indicate that (though slot deinit not implemented yet.
    err = init_sdmmc_host(host_config->slot, slot_config, &card_handle);
    CHECK_EXECUTE_RESULT(err, "slot init failed");

    // probe and initialize card
    err = sdmmc_card_init(host_config, card);
    CHECK_EXECUTE_RESULT(err, "sdmmc_card_init failed");
    if (ff_diskio_get_drive(&pdrv) != ESP_OK || pdrv == FF_DRV_NOT_USED) {
        ESP_LOGD(TAG, "the maximum count of volumes is already mounted");
        err = ESP_ERR_NO_MEM;
        goto cleanup;
    }
    
    err = mount_to_vfs_fat(mount_config,0, card, pdrv, dup_path1, &fs1);
    CHECK_EXECUTE_RESULT(err, "mount_to_vfs pdrv1 failed");
    err = mount_to_vfs_fat(mount_config,1, card, pdrv, dup_path2, &fs2);
    CHECK_EXECUTE_RESULT(err, "mount_to_vfs pdrv2 failed");
    if (out_card != NULL) {
        *out_card = card;
    } 
    uint64_t total,frees;
    esp_vfs_fat_info(OBD_MOUNT_POINT_DATA,&total,&frees);
    printf("OBD_MOUNT_POINT_DATA Total size: %llu,free size: %llu\n", total,frees);  
    esp_vfs_fat_info(OBD_MOUNT_POINT_CFG,&total,&frees);
    printf("OBD_MOUNT_POINT_CFG Total size: %llu,free size: %llu\n", total,frees);   
 
    return ESP_OK;
cleanup:
    if (host_inited) {
        call_host_deinit(host_config);
    }
    free(card);
    free(dup_path1);
    free(dup_path2);
    return err;
}

static esp_err_t mount_to_vfs_fat(const esp_vfs_fat_mount_config_t *mount_config, uint8_t partitionId,sdmmc_card_t *card, uint8_t pdrv,
                                  const char *base_path, FATFS **out_fs)
{
    FATFS *fs = NULL;
    esp_err_t err;
    ff_diskio_register_sdmmc(pdrv, card);
    ff_sdmmc_set_disk_status_check(pdrv, mount_config->disk_status_check_enable);
    ESP_LOGD(TAG, "using pdrv=%i", pdrv);
    char drv[3] = {(char)('0' + partitionId), ':', 0};

    // connect FATFS to VFS
    err = esp_vfs_fat_register(base_path, drv, mount_config->max_files, &fs);
    *out_fs = fs;
    if (err == ESP_ERR_INVALID_STATE) {
        // it's okay, already registered with VFS
    } else if (err != ESP_OK) {
        ESP_LOGD(TAG, "esp_vfs_fat_register failed 0x(%x)", err);
        goto fail;
    }

    // Try to mount partition
    err = s_f_mount(card, fs, drv, pdrv, mount_config);
    if (err != ESP_OK) {
        goto fail;
    }
    return ESP_OK;

fail:
    if (fs) {
        f_mount(NULL, drv, 0);
    }
    esp_vfs_fat_unregister_path(base_path);
    ff_diskio_unregister(pdrv);
    return err;
}
```
rewrite esp-idf as follow:
PARTITION VolToPart[] = {
{0, 1}, /* Logical drive 0 ==> Physical drive 0,  1 part*/
{0, 2} /* Logical drive 1 ==> Physical drive 0, 2 part*/
{2, 0} /* Logical drive 2 ==> Physical drive 2, auto detection */
};