espressif/esp-idf #14311: ESP32-P4 APLL/MCLK/WS not working at TDM8 32-bit

ftab, 2024-08-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### Espressif SoC revision.

ESP32-P4 v0.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-P4 EV board

### Power Supply used.

USB

### What is the expected behavior?

Create I2S clock from APLL

### What is the actual behavior?

fails to initialize clock

### Steps to reproduce.

Using ESP-ADF i2s_stream in a simple audio pipeline to read /usb/test.mp3, split from 2 to 8 channels with our audio DSP code, then play to a PCM1681 (8ch DAC) over I2S TDM

Here is the snippet of code where the i2s stream is set up

```c
#define I2S_STREAM_TDM_CFG_44_1_8CH() {                                          \
    .type = AUDIO_STREAM_WRITER,                                                \
    .transmit_mode = I2S_COMM_MODE_TDM,                                         \
    .chan_cfg = I2S_CHANNEL_DEFAULT_CONFIG(I2S_NUM_0, I2S_ROLE_MASTER),         \
    .tdm_cfg = {                                                                \
        .clk_cfg  = { \
            .sample_rate_hz = 44100, \
            .clk_src = I2S_CLK_SRC_APLL, \
            .mclk_multiple = I2S_MCLK_MULTIPLE_768, \
            .bclk_div = 8, /* only takes effect for slave role */ \
        },                          \
        .slot_cfg = I2S_TDM_PHILIPS_SLOT_DEFAULT_CONFIG(I2S_DATA_BIT_WIDTH_32BIT, I2S_SLOT_MODE_STEREO, \
                                                        I2S_TDM_SLOT0 | I2S_TDM_SLOT1 | I2S_TDM_SLOT2 | I2S_TDM_SLOT3 | \
                                                        I2S_TDM_SLOT4 | I2S_TDM_SLOT5 | I2S_TDM_SLOT6 | I2S_TDM_SLOT7), \
        .gpio_cfg = {                                                           \
            .invert_flags = {                                                   \
                .mclk_inv = false,                                              \
                .bclk_inv = false,                                              \
                .ws_inv   = false,                                              \
            },                                                                  \
        },                                                                      \
[... 95 lines cut ...]
I (474) AUD: [2.5] Waiting for USB file system to be ready
W (473) UsbHostTask: in UsbHostTask's while loop
W (475) UsbHostTask: in HandleMscDevice
W (473) UsbHostTask: in handle_usb_events
W (475) UsbHostTask: in handle_usb_events's while loop
I (476) UsbHostTask: Waiting for USB stick to be connected
W (476) UsbHostTask: in wait_for_msc_device_condition
W (990) UsbHostTask: in msc_event_cb
I (990) UsbHostTask: MSC device connected
*** Device descriptor ***
...snip...
I (1242) AUD: [2.6] Playing file: /usb/test.mp3
I (1242) AUD: [5.0] Create i2s stream to write data to codec chip
E (1243) esp_clk_tree: esp_clk_tree_src_get_freq_hz(82): freq shouldn't be 0, calibration failed
E (1243) i2s_tdm: i2s_tdm_calculate_clock(64): sample rate is too large for the current clock source
E (1243) i2s_tdm: i2s_tdm_set_clock(76): clock calculate failed
E (1244) i2s_tdm: i2s_channel_init_tdm_mode(231): initialize channel failed while setting clock
E (1244) I2S_STREAM_IDF5.x: I2S stream init failed
```

### More Information.

Using my fork of ESP-ADF that hacks some stuff out to work on the P4 as ESP-ADF v2.7 is not released yet: https://github.com/radiosound-com/esp-adf/tree/radiosound-modded-v2.6-141-g218cf614-idf-v5.3-modded

When I change I2S_CLK_SRC_APLL to I2S_CLK_SRC_DEFAULT, i2s does initialize, but MCLK is running at 20mhz (expected: 44,100 x 768 = 33,868,800) and WS runs at 26khz instead of 44.1khz

ftab, 2024-08-06:
Just spotted in esp-idf/components/esp_hw_support/port/esp32p4/esp_clk_tree.c:
```c
// case SOC_MOD_CLK_APLL: TODO: IDF-8884
```
That explains the APLL not working. How can I stay apprised of the status of IDF-8884? Any word on when it's planned to be implemented?

L-KAYA, 2024-08-30:
I2S on P4 only has XTAL and APLL two internal clock sources (the PLL clock source is not supported). As APLL support has not merged yet, I2S can only use XTAL clock for now, which is up to 40MHz.

See the similar issue #14448. And the [APLL support patch](https://github.com/espressif/esp-idf/issues/14448#issuecomment-2311783923) is uploaded there for testing.

ftab, 2024-08-30:
> Hi @ftab @DatanoiseTV , thanks for your reporting!
> 
> I2S on P4 only has XTAL and APLL two internal clock sources (the PLL clock source is not supported). As APLL support has not merged yet, I2S can only use XTAL clock for now, which is up to 40MHz.
> 
> 
> 
> See the similar issue #14448. And the [APLL support patch](https://github.com/espressif/esp-idf/issues/14448#issuecomment-2311783923) is uploaded there for testing.

I can confirm the patch works. Thanks!

L-KAYA, 2024-09-07:
FYI, APLL has been supported on the master in [this commit](https://github.com/espressif/esp-idf/commit/3937e225ecdfb235858c6e1047d2ac546f0d78ed) 

You can rebase on the master to use APLL as well

ftab, 2024-10-02:
@L-KAYA we have been testing this on the P4 EV board and noticed we're only getting about 83 dB signal to noise ratio (A-weighted) with a PCM1681 DAC. We think it's due to jitter on MCLK. Is there any way to improve this?

L-KAYA, 2024-12-23:
Hi @ftab, sorry for the late reply, I didn't test how MCLK affect the SNR before. If you think it could be the jitter from MCLK, maybe you can try to adjust the MCLK frequency meanwhile keep the sample rate (like increase the `mclk_multiple`). It might can help the `PCM1681` working at a higher frequency and sample more precise.
(Don't forget to change the configuration of the `PCM1681` as well if you increase the `mclk_multiple`).

L-KAYA, 2024-12-23:
This issue is going to be closed since the original issue have been solved.

If there are any other issues need to be discussed, welcome to open a new issue~

closed 2024-12-23 by Alvin1Zhang