espressif/esp-idf #14319: USB Host MSC: USB thumb drive cannot be initialized

rikaiot, 2024-08-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V5.3

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-S3-DEVKITC-1-N32R8V

### Power Supply used.

USB

### What is the expected behavior?

When running examples\peripherals\usb\host\main\msc_example_main.c a the example is waiting for an USB thumb drive to be connected and after connecting some tests are performed on the thumb drive.

### What is the actual behavior?

Some USB thumb drives cannot be initialized and generate an error and force the example to reboot:
```
I (367) example: Waiting for USB flash drive to be connected
I (8247) example: MSC device connected
E (12087) USB_MSC: msc_host_install_device(503):
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x42009fb8
0x42009fb8: app_main at ../../../main/msc_example_main.c:278 (discriminator 1)
```

The example is only working, when the USB thumb drive is already connected at startup of the example or when pressing the Reset button while the USB thumb drive is already connected.

### Steps to reproduce.

1. Build and Flash the examples\peripherals\usb\host\msc\ project to an ESP32-S3-DEVKITC-1-N32R8V
2. Short the Diode D7 to provide 5V on the USB-Port and start the ESP32-S3-DEVKITC-1-N32R8V
3. Connect an specific USB thumb drive (some do work some do not) to the USB-Port via USB micro-B to USB Typ A adapter (OTG cable)
4. The drive is detected as connected but an error and reboot occurs:
```
I (8247) example: MSC device connected
E (12087) USB_MSC: msc_host_install_device(503):
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x42009fb8
0x42009fb8: app_main at ../../../main/msc_example_main.c:278 (discriminator 1)
```

### Debug Logs.

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
[... 137 lines cut ...]
        iInterface 0
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x81   EP 1 IN
                bmAttributes 0x2        BULK
                wMaxPacketSize 64
                bInterval 0
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x2    EP 2 OUT
                bmAttributes 0x2        BULK
                wMaxPacketSize 64
                bInterval 0
Device info:
         Capacity: 3823 MB
         Sector size: 512
         Sector count: 7831551
         PID: 0x4200
         VID: 0x13FE
         iProduct: USB DISK 2.0
         iManufacturer:
         iSerialNumber: 0701767BB12A0C22
```

roma-jam, 2024-08-07:
Hi @rikaiot, 

Is that possible to change the verbosity level to `Verbose` via menuconfig and share the log during the device connection?

[image]

rikaiot, 2024-08-07:
Hi @roma-jam,

here is the verbose log output, when starting the example and plugging in the USB thumb drive:

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fce2810,len:0x178c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xcb8
load:0x403cb700,len:0x2db0
entry 0x403c8914
I (33) boot: ESP-IDF v5.3-dirty 2nd stage bootloader
I (33) boot: compile time Aug  7 2024 09:59:19
I (33) boot: Multicore bootloader
I (37) boot: chip revision: v0.1
I (40) boot.esp32s3: Boot SPI Speed : 80MHz
I (45) boot.esp32s3: SPI Mode       : SLOW READ
I (50) boot.esp32s3: SPI Flash Size : 2MB
I (55) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (94) boot: End of partition table
I (98) esp_image: segment 0: paddr=00010020 vaddr=3c040020 size=0ed44h ( 60740) map
I (121) esp_image: segment 1: paddr=0001ed6c vaddr=3fc92d00 size=012ach (  4780) load
I (123) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=32a98h (207512) map
I (176) esp_image: segment 3: paddr=00052ac0 vaddr=3fc93fac size=019d8h (  6616) load
I (178) esp_image: segment 4: paddr=000544a0 vaddr=40374000 size=0ec90h ( 60560) load
I (205) boot: Loaded app from partition at offset 0x10000
I (205) boot: Disabling RNG early entropy source...
V (216) mmap: after coalescing, 1 regions are left
I (217) cpu_start: Multicore app
D (217) cpu_start: Pro cpu up
D (217) cpu_start: Starting app cpu, entry point is 0x403752fc
0x403752fc: call_start_cpu1 at C:/SysGCC/esp32/esp-idf/v5.3/components/esp_system/port/cpu_start.c:195

D (0) cpu_start: App cpu up
V CACHE_ERR: illegal error intr clr & ena mask is: 0x3f
2 CACHE_ERR: corD 12ac)ess error intr _Clr &alena mask ial: : x1210
V CACHE_ERR: illegal error intr clr & ena mask is: 0x3f
V CACHE_ERR: core 0 access error intr clr & ena mask is: 0x1f
D (249) cpu_start: calling init function: 0x4200229c on core: 0
0x4200229c: __esp_system_init_fn_init_efuse_check at C:/SysGCC/esp32/esp-idf/v5.3/components/efuse/src/esp_efuse_startup.c:31

D (254) cpu_start: calling init function: 0x42002528 on core: 0
0x42002528: __esp_system_init_fn_init_show_cpu_freq at C:/SysGCC/esp32/esp-idf/v5.3/components/esp_system/startup_funcs.c:58

I (260) cpu_start: Pro cpu start user code
I (264) cpu_start: cpu freq: 160000000 Hz
D (269) cpu_start: calling init function: 0x420021cc on core: 0
0x420021cc: __esp_system_init_fn_init_show_app_info at C:/SysGCC/esp32/esp-idf/v5.3/components/esp_app_format/esp_app_desc.c:104

I (275) app_init: Application information:
I (280) app_init: Project name:     usb-msc
I (285) app_init: App version:      1
I (289) app_init: Compile time:     Aug  7 2024 09:59:00
I (295) app_init: ELF file SHA256:  05db7289e...
I (300) app_init: ESP-IDF:          v5.3-dirty
D (305) cpu_start: calling init function: 0x420022b0 on core: 0
0x420022b0: __esp_system_init_fn_init_efuse_show_app_info at C:/SysGCC/esp32/esp-idf/v5.3/components/efuse/src/esp_efuse_startup.c:41

I (311) efuse_init: Min chip rev:     v0.0
[... 285 lines cut ...]
D (16478) USB_MSC_SCSI: sense_key: 0x06, code: 0x28, qualifier: 0x00
E (16488) USB_MSC: msc_host_install_device(503):
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x4200a500
0x4200a500: app_main at ../../../main/msc_example_main.c:278 (discriminator 1)

file: "../../../main/msc_example_main.c" line 278
func: app_main
expression: msc_host_install_device(msg.data.new_dev_address, &msc_device)

abort() was called at PC 0x4037a227 on core 0
0x4037a227: _esp_error_check_failed at C:/SysGCC/esp32/esp-idf/v5.3/components/esp_system/esp_err.c:49

Backtrace: 0x40375b0a:0x3fc98bc0 0x4037a231:0x3fc98be0 0x40380a89:0x3fc98c00 0x4037a227:0x3fc98c70 0x4200a500:0x3fc98ca0 0x420321cf:0x3fc98de0 0x4037acc5:0x3fc98e10
0x40375b0a: panic_abort at C:/SysGCC/esp32/esp-idf/v5.3/components/esp_system/panic.c:463
0x4037a231: esp_system_abort at C:/SysGCC/esp32/esp-idf/v5.3/components/esp_system/port/esp_system_chip.c:92
0x40380a89: abort at C:/SysGCC/esp32/esp-idf/v5.3/components/newlib/abort.c:38
0x4037a227: _esp_error_check_failed at C:/SysGCC/esp32/esp-idf/v5.3/components/esp_system/esp_err.c:49
0x4200a500: app_main at ../../../main/msc_example_main.c:278 (discriminator 1)
0x420321cf: main_task at C:/SysGCC/esp32/esp-idf/v5.3/components/freertos/app_startup.c:208
0x4037acc5: vPortTaskWrapper at C:/SysGCC/esp32/esp-idf/v5.3/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

ELF file SHA256: 05db7289e

Rebooting...
```

roma-jam, 2024-08-07:
@rikaiot thanks for the additional information. 

Which port do you use to power the dev kit with attachable USB drive? There might be not enough power to support both devices.
Could you try to use an external Hub with external power supply to power the dev kit?

rikaiot, 2024-08-07:
@roma-jam we tried powering the dev kit via a hub with external power supply without any difference.  

We additionally also tried directly powering the USB thumb drive with an separate 5V power supply unit but the error persists.

The interesting thing is, that the example is working when the thumb drive is already plugged in at startup.

roma-jam, 2024-08-08:
@rikaiot ,
Thanks for the reply. 

Could you please share the same log but for the case when device is present in the port while dev kit is powering up? 

Also, do you know how to use Wireshark and how to log USB traffic with it? To get the log of attaching USB Drive to the PC Host also will be helpfull.

rikaiot, 2024-08-08:
@roma-jam, 
Thanks for your help!

Here is the log of a run with the thumb drive connected while powering up the kit.
It seems that the very first run after power up also fails and triggers a reboot. After the reboot the devices is recognized successfully.
Also pressing the reset button, with the device connected works without any error.
Summarizing it up:
1. Power-Up with thumb drive connected: Device cannot be initialized
2. Reboot/Reset with thumb drive connected: Success
3. Hot-Plug of thumb drive: Device cannot be initialized

We are not familiar with capturing USB traffic with Wireshark but could try to figure it out.

But we would also be glad to send you 5 pieces of the affected thumb drives to allow you to reproduce and investigate the issue directly.

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fce2810,len:0x178c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xcb8
load:0x403cb700,len:0x2db0
entry 0x403c8914
I (33) boot: ESP-IDF v5.3-dirty 2nd stage bootloader
I (33) boot: compile time Aug  7 2024 09:59:19
I (33) boot: Multicore bootloader
I (37) boot: chip revision: v0.1
I (40) boot.esp32s3: Boot SPI Speed : 80MHz
I (45) boot.esp32s3: SPI Mode       : SLOW READ
I (50) boot.esp32s3: SPI Flash Size : 2MB
I (55) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (94) boot: End of partition table
I (98) esp_image: segment 0: paddr=00010020 vaddr=3c040020 size=0ed44h ( 60740) map
I (121) esp_image: segment 1: paddr=0001ed6c vaddr=3fc92d00 size=012ach (  4780) load
I (123) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=32a98h (207512) map
I (176) esp_image: segment 3: paddr=00052ac0 vaddr=3fc93fac size=019d8h (  6616) load
I (178) esp_image: segment 4: paddr=000544a0 vaddr=40374000 size=0ec90h ( 60560) load
I (205) boot: Loaded app from partition at offset 0x10000
I (205) boot: Disabling RNG early entropy source...
V (216) mmap: after coalescing, 1 regions are left
I (217) cpu_start: Multicore app
D (217) cpu_start: Pro cpu up
D (217) cpu_start: Starting app cpu, entry point is 0x403752fc
0x403752fc: call_start_cpu1 at C:/SysGCC/esp32/esp-idf/v5.3/components/esp_system/port/cpu_start.c:195

D (0) cpu_start: App cpu up
V CACHE_ERR: illegal error intr clr & ena mask is: 0x3f
V CACHE_ERD:(2core 1 a: eTCs error intca clr & enavamask 4s:410xf

V CACHE_ERR: illegal error intr clr & ena mask is: 0x3f
V CACHE_ERR: core 0 access error intr clr & ena mask is: 0x1f
D (249) cpu_start: calling init function: 0x4200229c on core: 0
0x4200229c: __esp_system_init_fn_init_efuse_check at C:/SysGCC/esp32/esp-idf/v5.3/components/efuse/src/esp_efuse_startup.c:31

D (253) cpu_start: calling init function: 0x42002528 on core: 0
0x42002528: __esp_system_init_fn_init_show_cpu_freq at C:/SysGCC/esp32/esp-idf/v5.3/components/esp_system/startup_funcs.c:58

I (259) cpu_start: Pro cpu start user code
I (264) cpu_start: cpu freq: 160000000 Hz
D (269) cpu_start: calling init function: 0x420021cc on core: 0
[... 2314 lines cut ...]
V (8343) USB_MSC: USB MSC handling
V (8343) USB_MSC: USB MSC handling
V (8353) USB_MSC: USB MSC handling
V (8353) USB_MSC: USB MSC handling
V (8353) USB_MSC: USB MSC handling
V (8363) USB_MSC: USB MSC handling
V (8363) USB_MSC: USB MSC handling
V (8363) USB_MSC: USB MSC handling
V (8373) USB_MSC: USB MSC handling
V (8373) USB_MSC: USB MSC handling
V (8373) USB_MSC: USB MSC handling
V (8383) USB_MSC: USB MSC handling
V (8383) USB_MSC: USB MSC handling
I (8383) example: Read speed 0.39 MiB/s
V (8393) USB_MSC: USB MSC handling
V (8393) USB_MSC: USB MSC handling
V (8403) USB_MSC: USB MSC handling
V (8403) USB_MSC: USB MSC handling
V (8403) USB_MSC: USB MSC handling
V (8413) USB_MSC: USB MSC handling
V (8413) USB_MSC: USB MSC handling
V (8423) USB_MSC: USB MSC handling
V (8423) USB_MSC: USB MSC handling
I (8423) example: Example finished, you can disconnect the USB flash drive
```

roma-jam, 2024-08-08:
@rikaiot 
Thanks for the additional log. 
The one different thing with these two cases is the timing. When device is attached to the port it has the power for a longer period of time than during the device connection. 

Let's try to increase the `Reset recovery delay` value to 500 ms in USB-OTG configuration (for detailed inf, refer to section [enumeration-configuration](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-reference/peripherals/usb_host.html#enumeration-configuration)) just to give the USB drive some time to enable flash properly. This is a delay may be found in `idf.py menuconfig`:
[image]

Please, let me know about the result. 
Thanks

tore-espressif, 2024-08-12:
Hello @rikaiot , thanks for the useful logs.
Inside the MSC driver there is another timeout that might influence your usecase. https://github.com/espressif/esp-usb/blob/master/host/class/msc/usb_host_msc/src/msc_host.c#L83

Could you try increasing this to 5000 or even 10000 ms?

(You will have to move the code from managed_components/espressif__usb_host_msc to components/espressif__usb_host_msc , as our build system does not allow modifying files in `managed_components` folder... :/ )

cc @leeebo I think you encountered similar problem in the past...

leeebo, 2024-08-13:
@tore-espressif We had issues with some specified USB disks, which were fixed by increasing the RESET_RECOVERY_MS from 20 to 500ms.

@rikaiot Which version of usb_host_msc component you using? Does the USB Disk you using work as normal on PC ?

createskyblue, 2024-08-21:
I tried two USB drives I had on hand, one being an SD card reader and the other a regular USB drive. The card reader USB encounters the issues mentioned above when used with ESP32- IDF versions 5.3, 5.2, and 4.4, whereas the regular USB drive works fine.

When I attempted to use STM32CubeMX to generate an MSC project for the STM32F103RCT6, both the card reader USB and the regular USB drive worked without any problems.

It seems that the MSC compatibility in ESP-IDF might not be very good.

tore-espressif, 2024-08-24:
@createskyblue Thanks for the test! We never claimed compatibility with SD card readers, but this is surely a feature we will add

rikaiot, 2024-08-28:
@tore-espressif  the initial issue happens with regular USB thumb drives. Is there a way we could send you some test devices to elaborate the issue?

tore-espressif, 2024-08-29:
> @tore-espressif the initial issue happens with regular USB thumb drives. Is there a way we could send you some test devices to elaborate the issue?

Hello @rikaiot please contact me at tomas.rezucha@espressif.com and we can handle the rest via email.

ColmanLai, 2024-08-30:
I have the same problem. My device is not a SD card reader, it is an USB thumb drive. As long as I know, some USB thumb drive actually is a SD card reader with a permeant SD card inside. Please acknowledge me for any progress.
Thank you.

fozzzgate, 2024-10-21:
@tore-espressif  We are also having the exact same issue with a 16GB Lexar flash drive and do not want to ship until this is resolved. Since this doesn't look like it is going to be fixed anytime soon, can you give us a hint about what function(s) need to be modified in what file(s)? Changing the timeouts has no effect on the problem.

```
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 2.00
bDeviceClass 0x0
bDeviceSubClass 0x0
bDeviceProtocol 0x0
bMaxPacketSize0 64
idVendor 0x5dc
idProduct 0xc75d
bcdDevice 1.00
iManufacturer 1
iProduct 2
iSerialNumber 3
bNumConfigurations 1
*** Configuration descriptor ***
bLength 9
bDescriptorType 2
wTotalLength 32
bNumInterfaces 1
bConfigurationValue 1
iConfiguration 0
bmAttributes 0x80
bMaxPower 100mA
        *** Interface descriptor ***
        bLength 9
        bDescriptorType 4
        bInterfaceNumber 0
        bAlternateSetting 0
        bNumEndpoints 2
        bInterfaceClass 0x8
        bInterfaceSubClass 0x6
        bInterfaceProtocol 0x50
        iInterface 0
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x81   EP 1 IN
                bmAttributes 0x2        BULK
                wMaxPacketSize 64
                bInterval 0
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x2    EP 2 OUT
                bmAttributes 0x2        BULK
                wMaxPacketSize 64
                bInterval 0
Device info:
         Capacity: 15270 MB
         Sector size: 512
         Sector count: 31273727
         PID: 0xC75D
         VID: 0x 5DC
         iProduct: USB Flash Drive
         iManufacturer: Lexar
```

roma-jam, 2024-10-21:
Hi @fozzzgate, 

thanks for reporting.
The problem you have also has the `NOT READY TO READY TRANSITION - MEDIA CHANGED` as the reply on test unit ready request? 

Do you have an option to get the wireshark log of attaching such device to a PC?

fozzzgate, 2024-10-27:
@roma-jam Yes, I also have the `NOT READY TO READY TRANSITION - MEDIA CHANGED` problem with a certain USB drive.

Attached are Wireshark captures of a good Cruzer U 16GB USB drive insertion and a bad Lexar 16GB USB drive insertion.

[Wireshark Inseertion Captures.zip](https://github.com/user-attachments/files/17532783/Wireshark.Inseertion.Captures.zip)

The first difference is Frame 45:

Good: Transfer length = 8,   CDB (12 bytes): a20000000000000000080000
Bad:  Transfer length = 252, CDB (10 bytes): 2300000000000000fc00

The `NOT READY TO READY TRANSITION - MEDIA CHANGED` is Frame 56:

Despite it saying `Current Error (0x70)` below, it mounts just fine under Windows and under ESP32-S3 if inserted before boot.

```
Frame 56: 45 bytes on wire (360 bits), 45 bytes captured (360 bits) on interface \\.\USBPcap1, id 0
USB URB
    [Source: 1.5.1]
    [Destination: host]
    USBPcap pseudoheader length: 27
    IRP ID: 0xffffb98a2bbd7aa0
    IRP USBD_STATUS: USBD_STATUS_SUCCESS (0x00000000)
    URB Function: URB_FUNCTION_BULK_OR_INTERRUPT_TRANSFER (0x0009)
    IRP information: 0x01, Direction: PDO -> FDO
    URB bus id: 1
    Device address: 5
    Endpoint: 0x81, Direction: IN
    URB transfer type: URB_BULK (0x03)
    Packet Data Length: 18
    [Request in: 55]
    [Time from request: 1.726817000 seconds]
    [bInterfaceClass: Mass Storage (0x08)]
USB Mass Storage
SCSI Payload (Request Sense Response Data)
    [LUN: 0x0000]
    [Command Set:Direct Access Device (0x00) ]
    [SBC Opcode: Request Sense (0x03)]
    [Request in: 53]
    [Response in: 58]
    .111 0000 = SNS Error Type: Current Error (0x70)
    Valid: 112
    0... .... = Filemark: False
    .0.. .... = EOM: False
    ..0. .... = ILI: False
    .... 0110 = Sense Key: Unit Attention (0x6)
    Sense Info: 0x00000000
    Additional Sense Length: 10
    Command-Specific Information: 00000000
    Additional Sense Code+Qualifier: Not Ready To Ready Change, Medium May Have Changed (0x2800)
    Field Replaceable Unit Code: 0x00
    0... .... = SKSV: False
    .000 0000 0000 0000 0000 0000 = Sense Key Specific: 0x000000
```

fozzzgate, 2024-10-27:
@roma-jam @tore-espressif Attached are Wireshark JSON packet dumps of the bad (Lexar) and good (Sandisk Cruzer U) insertions. I have removed the "noise" between the two captures (i.e. timestamps, GUIDs) to make diffs easier.

[USB Wireshark & JSON.zip](https://github.com/user-attachments/files/17533509/USB.Wireshark.JSON.zip)

The dumps are identical up to Frame 56, where the 'bad' Lexar drive returns `Not Ready To Ready Change`. However, whatever Windows did to 'fix' the problem seems to have been completed by Frame 65, when Windows successfully requests the Serial Number, which is returned in Frame 68.

For the 'good' Cruzer drive, Frame 56 returns `Invalid Command Operation Code (0x2000)` (vs. the Lexar `Not Ready To Ready Change`) and the Serial Number isn't requested until Frame 135 (and returned in Frame 138).

fozzzgate, 2024-10-27:
This Stack Overflow answer seems to hint at a solution:

[Android SCSI USB communication issues](https://stackoverflow.com/questions/65327569/android-scsi-usb-communication-issues)

fozzzgate, 2024-10-27:
Created pull request that fixes the issue: https://github.com/espressif/esp-usb/pull/81

closed 2024-10-29 by roma-jam