espressif/esp-idf #14320: ESP-BLE-MESH : RPL is full during message communication using vendor model

asgar-ayla, 2024-08-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.2

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3

### Power Supply used.

USB

### What is the expected behavior?

The vendor client model application flashed on ESP32-S3 should run without any error during communication with the vendor server model flashed on ESP32-H2.

### What is the actual behavior?

The vendor client model running on ESP32-S3 application goes unresponsive after some successful message transactions.

### Steps to reproduce.

1. Provision the vendor client model node (ESP32-S3).
2. Provision 3 vendor server model nodes (ESP32-H2).
3. Create a ESP-BLE-MESH of 4 nodes and start message passing between client and server nodes.
4. After a few successful transactions, the ESP32-S3 stops responding after printing "RPL is full".

### Debug Logs.

```plain
BLE_MESH: RPL is full!
BLE_MESH: RPL is full!
BLE_MESH: RPL is full!
BLE_MESH: RPL is full!
BLE_MESH: RPL is full!
BLE_MESH: RPL is full!
BLE_MESH: RPL is full!
BLE_MESH: RPL is full!
BLE_MESH: RPL is full!
```

### More Information.

The ESP32-H2 (Server node) uses ESP-IDF 5.1.2 while the ESP32-S3 (Client node) uses ESP-IDF 4.4.2.
Also, the value of CONFIG_BLE_MESH_CRPL is set to default value of 10 on the Client node side.

chegewara, 2024-09-30:
Hi,
im not sure if this is the same issue, just different log, so please check this one:

https://github.com/espressif/esp-idf/issues/13384

forx157, 2025-01-03:
Hi, 
"RPL is full" indicates that the size of the RPL should be increased. Ideally, the size of the RPL should be set to match the number of nodes in the current network.

closed 2025-03-31 by Alvin1Zhang