espressif/esp-idf #14322: ESP32C3 esp_wifi_stop() and esp_wifi_deinit() cannot reduce power consumption

ShengjieCh, 2024-08-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-v5.2.2

### Espressif SoC revision.

ESP32C3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

Using the disable wifi function supported by host-fg (esp_wifi_stop() and esp_wifi_deinit()), the power consumption of esp32 will be reduced.

### What is the actual behavior?

After disabling wifi, power consumption has not been significantly reduced(283mW->277mW).
enable wifi: [image]
disable wifi: [image]
In contrast, without initializing wifi, the power consumption of esp32 is only 147mW.
[image]
See details: https://github.com/espressif/esp-hosted/issues/405

### Steps to reproduce.

[image]

[image]

mantriyogesh, 2024-08-07:
@kapilkedawat can we please check with relevant team?

esp-lis, 2024-08-08:
@mantriyogesh Is bluetooth disaled after disabling wifi?

ShengjieCh, 2024-08-08:
Only disabled wifi, bluetooth still open.

esp-lis, 2024-08-08:
you can try again after disabling wifi and bluetooth.

ShengjieCh, 2024-08-08:
@esp-lis My usage scenario is to use Bluetooth and not use wifi for the time being.

mantriyogesh, 2024-08-08:
> @mantriyogesh Is bluetooth disaled after disabling wifi?

enable, disable feature is flexible at api level. features right now supported are wifi and bluetooth individually. (one feature = one request).

See host side code:
https://github.com/espressif/esp-hosted/blob/dcb21761b3bba65a843c34937aef843a712c26d2/esp_hosted_fg/host/linux/host_control/c_support/test.c#L70-L73

maps to:
Slave side mapping code:
https://github.com/espressif/esp-hosted/blob/dcb21761b3bba65a843c34937aef843a712c26d2/esp_hosted_fg/esp/esp_driver/network_adapter/main/slave_control.c#L1824-L1973

So correct, when wifi requested to disabled, wifi only is disabled. Similar is the case with bluetooth.
One disable doesn't auto disable another thing.

mantriyogesh, 2024-08-08:
@ShengjieCh , correct wifi power use would only be realised when bluetooth is shut off completely.

If you keep bluetooth on and only change wifi off-on, still the rf should be up.
There is single rf time shared in ESP chipsets.

So regarding your original query, disable both, and only run APIs to which you have power doubt. (with and without wifi de-init, keeping bluetooth all time off)
Otherwise, it would be complicated to predict the power usage in my opinion.

ShengjieCh, 2024-08-08:
@mantriyogesh Should I turn off wifi and Bluetooth after starting esp32, and then enable Bluetooth alone to achieve the expected power consumption of only disable wifi? This seems like a long process, with the extra effort of turning Bluetooth off and on again.

mantriyogesh, 2024-08-08:
Yes. Hosted currently by default enables wifi and bluetooth.
The procedure you said for now is fine.

ShengjieCh, 2024-08-08:
This special logic cannot be added to the mater of our code. I still hope you can figure out why the power consumption cannot be reduced correctly after disabling wifi. If it is confirmed that it is necessary to turn off all wifi and Bluetooth, and then turn on Bluetooth to achieve the expected function, please let me know and our team will evaluate this function, thanks.

mantriyogesh, 2024-08-08:
I am not sure what do you mean by,

> This special logic cannot be added to the mater of our code.

Code is already in place in ESP-Hosted, just need to call one of the APIs.

for example,
```c
test_disable_wifi();
test_disable_bt();
//check power
test_enable_bt(); 
// check power
```
But as you see in the slave code, ESP-Hosted does
```c
esp_wifi_stop(); and
esp_wifi_deinit();
```
upon calling `test_disable_wifi();` at host.

Your specific problem here reported is wrt Wi-Fi power consumption.
So exactly what code changes you do at slave and measure power at what place is very crucial, to even ESP-IDF team to handle this query.

As I stated in original issue, RF state is very important, unless there is defined RF state, power measurements taken would be confusing.

esp-lis, 2024-08-08:
@ShengjieCh @mantriyogesh I understand that your application uses both Wi-Fi and Bluetooth modules. Initially (step 1), since neither Wi-Fi nor Bluetooth is enabled, the power consumption appears to be low. Then (step 2), after enabling the Wi-Fi and Bluetooth modules, the power consumption increases (mainly due to the RF module). Finally (step 3), you disable the Wi-Fi module but do not disable the Bluetooth module; you expect the power consumption to be similar to step 1, but the actual measurement is similar to step 2.

On the ESP32-C3, the Wi-Fi and Bluetooth modules share a common RF module, which means that enabling either the Wi-Fi or Bluetooth module in your application will cause the RF module to be enabled.

I’m not sure whether the coexistence module can be enabled in your application (perhaps it’s not supported, or perhaps it can’t be adapted to your application scenario). The coexistence module provides management for multiple modem modules (Wi-Fi and Bluetooth, including those associated with the RF module), which might help with your problem.

For more information about the coexistence module, refer to [coexist](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/coexist.html).

ShengjieCh, 2024-08-08:
@mantriyogesh Does host-fg support this coexistence module? I can't seem to find esp_coex_status_bit_clear in the code.
@esp-lis The coexistence module seems to work only when wifi and Bluetooth are turned on at the same time. When I only enable Bluetooth, why is the power consumption as high as if wifi and Bluetooth are enabled at the same time? Could it be that when both wifi and Bluetooth are turned on, each accounts for 50%, and when wifi is turned off, Bluetooth accounts for 100%?
[image]
This is an important dependency of the product we are about to release. The impact of substandard power consumption is very serious. Thank you very much for your important and timely help.

mantriyogesh, 2024-08-08:
@ShengjieCh ,
Not yet, I am also going through the documentation.
Also even if I wish to support, it would take good time.

We use the default settings.

ShengjieCh, 2024-08-09:
@mantriyogesh @esp-lis I learned from Espressif's RF colleagues that "the power consumption is mainly radio frequency. If you turn off WiFi alone, there will be almost no change. It should be that your ble controller does not enable modemsleep, so no matter what state the WiFi is in, RF rx is still always on." After using **idf.py menuconfig** to turn on modemsleep, the power consumption is as expected after turning off wifi. Many thanks to yogesh for your long-term help!
 [image]
[image]

mantriyogesh, 2024-08-12:
Yes. Good find. I think it is better if we set this option by default, in ESP-Hosted

alexccmy, 2024-09-13:
I would like to ask how much standby current is generally on low power consumption. I use 3.3v power supply, and the current in hibernation is 900ua. Can it be reduced to 4ua?

esp-lis, 2024-10-09:
@alexccmy you can use $IDF_PATH/examples/system/deep_sleep/ to test for this, thanks.