espressif/esp-idf #14323: Wifi warning: exceed max band, 2g, ngroup:3

matoha, 2024-08-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

I have updated from IDF 4.4 to 5.3. Everything seems to work, but I keep receiving a warning on the terminal, which is printed multiple times per second. The Wi-Fi *does* work as expected, but I don't want to release the update to production if there are potential issues with the Wi-Fi.

```
W (19796) wifi:exceed max band, 2g, ngroup:3
```

I've tried searching for this warning and can't find anything. Sadly the Wi-Fi code is closed-source so I am unable to debug further.

### What is the actual behavior?

See above.

### Steps to reproduce.

I initialise Wi-Fi following the example here https://github.com/espressif/esp-idf/blob/v5.3/examples/wifi/getting_started/station/main/station_example_main.c

My Wi-Fi code is:

```
    s_wifi_event_group = xEventGroupCreate();

    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());
    intf = esp_netif_create_default_wifi_sta();
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));

    ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &event_handler, nullptr));
    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_STA_GOT_IP, &event_handler, nullptr));
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA) );
```

and 

```
    wifi_config_t wifi_config{};
    wifi_config.sta.scan_method = WIFI_ALL_CHANNEL_SCAN;
    wifi_config.sta.sort_method = WIFI_CONNECT_AP_BY_SIGNAL;
    wifi_config.sta.threshold.rssi = -80;
    wifi_config.sta.failure_retry_cnt = 10;
    wifi_config.sta.threshold.authmode = WIFI_AUTH_OPEN;
    wifi_config.sta.sae_pwe_h2e = WPA3_SAE_PWE_HUNT_AND_PECK;
    wifi_config.sta.sae_h2e_identifier[0] = 0x00;
    memcpy(wifi_config.sta.ssid, my_ssid, sizeof(wifi_config.sta.ssid));
    memcpy(wifi_config.sta.password, my_password, sizeof(wifi_config.sta.password));

    ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_STA, &wifi_config));
    ESP_ERROR_CHECK(esp_wifi_start() );
```

The warnings start after `esp_wifi_start`, not at any point before.

### Debug Logs.

```plain
I (10359) net80211: net80211 rom version: e7ae62f
I (10363) wifi:wifi driver task: 3fcc78ac, prio:23, stack:6656, core=0
I (10396) wifi:wifi firmware version: 0caa81945
I (10396) wifi:wifi certification version: v7.0
I (10397) wifi:config NVS flash: enabled
I (10399) wifi:config nano formating: disabled
I (10403) wifi:Init data frame dynamic rx buffer num: 32
I (10408) wifi:Init static rx mgmt buffer num: 5
I (10412) wifi:Init management short buffer num: 32
I (10417) wifi:Init static tx buffer num: 16
I (10421) wifi:Init tx cache buffer num: 32
I (10425) wifi:Init static tx FG buffer num: 2
I (10429) wifi:Init static rx buffer size: 1600
I (10433) wifi:Init static rx buffer num: 10
I (10437) wifi:Init dynamic rx buffer num: 32
I (10442) wifi_init: rx ba win: 6
I (10445) wifi_init: accept mbox: 6
I (10449) wifi_init: tcpip mbox: 32
I (10453) wifi_init: udp mbox: 6
I (10457) wifi_init: tcp mbox: 6
I (10461) wifi_init: tcp tx win: 5744
I (10466) wifi_init: tcp rx win: 5744
I (10470) wifi_init: tcp mss: 1440
I (10474) wifi_init: WiFi/LWIP prefer SPIRAM
I (10479) wifi_init: WiFi IRAM OP enabled
I (10484) wifi_init: WiFi RX IRAM OP enabled
I (16503) phy_init: phy_version 680,a6008b2,Jun  4 2024,16:41:10
I (16550) wifi:mode : sta (34:85:18:92:3b:08)
I (16551) wifi:enable tsf
W (16560) wifi:exceed max band, 2g, ngroup:3
W (16561) wifi:exceed max band, 2g, ngroup:3
W (16562) wifi:exceed max band, 2g, ngroup:3
W (16563) wifi:exceed max band, 2g, ngroup:3
W (16580) wifi:exceed max band, 2g, ngroup:3
I (18966) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1, snd_ch_cfg:0x0
I (18967) wifi:state: init -> auth (0xb0)
I (18974) wifi:state: auth -> assoc (0x0)
I (18996) wifi:state: assoc -> run (0x10)
I (19029) wifi:connected with PR******, aid = 5, channel 1, BW20, bssid = 80:bc:37:48:**:**
I (19030) wifi:security: WPA2-PSK, phy: bgn, rssi: -68
I (19037) wifi:pm start, type: 1
I (19038) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (19043) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 25000, mt_pti: 14, mt_time: 10000
I (19057) wifi:<ba-add>idx:0 (ifx:0, 80:bc:37:48:**:**), tid:0, ssn:0, winSize:64
I (19181) wifi:AP's beacon interval = 102400 us, DTIM period = 1
W (19183) wifi:exceed max band, 2g, ngroup:3
W (19387) wifi:exceed max band, 2g, ngroup:3
W (19489) wifi:exceed max band, 2g, ngroup:3
W (19694) wifi:exceed max band, 2g, ngroup:3
W (19796) wifi:exceed max band, 2g, ngroup:3
W (19899) wifi:exceed max band, 2g, ngroup:3
I (19902) wifi:<ba-add>idx:1 (ifx:0, 80:bc:37:48:**:**), tid:1, ssn:0, winSize:64
W (20001) wifi:exceed max band, 2g, ngroup:3
I (20053) esp_netif_handlers: sta ip: 172.24.187.189, mask: 255.255.255.0, gw: 172.24.187.1
W (20206) wifi:exceed max band, 2g, ngroup:3
W (20411) wifi:exceed max band, 2g, ngroup:3
[repeats indefinitely]
```
```

MaxwellAlan, 2024-08-09:
Hi @matoha 

We compared the differences between versions v4.4 and v5.3 regarding this line of code and found that the internal processing logic of the relevant code is the same. The difference is that in v5.3, the log level of this code was set to warning, while in v4.4, the log would not be printed. The log indicates that there is some anomaly in our internal processing of the country-related fields carried by the router's beacon/probe response, but it should not affect normal communication. If possible, please provide a wireless packet capture from the router so we can analyze the field in detail, and also provide the model and software/hardware version of the router.

Thanks.

matoha, 2024-08-09:
Hi @MaxwellAlan , thanks for your response.

I have tried my phone as a hotspot and the warning goes away, so it is indeed router-specific. The country code for both is GB though so I don't think it's the country. 

The router belongs to the office management company (it's building-wide Wi-Fi) and I unfortunately don't have direct access to it. Any suggestions how could I help debug this further?

Thanks
Martin

MaxwellAlan, 2024-08-14:
Hi @matoha 

Could you upload a wireless sniffer pkts for the router, we can check the country info field in the beacon/probe resp frame of the routers.

Thanks.

matoha, 2024-08-14:
Hi @MaxwellAlan ,

Please find the packets attached, hopefully it is in the correct format (I am able to open it in Wireshark).

Best
Martin
[wifi-sniff.pcap.zip](https://github.com/user-attachments/files/16616813/wifi-sniff.pcap.zip)

MaxwellAlan, 2024-08-19:
Hi @matoha 

Thanks for the sniffer, it's very helpful, we will try to be compatible with this format。

karamdali, 2024-09-25:
I've got the same warring message. how did you solve it?

matoha, 2024-09-26:
Hi,

It seems Espressif have made an update to fix this (thanks!) but I haven't had time to test it yet, just came back from holidays. The fix is on the master branch only at the moment.

Since from their comments above it seemed this is not an issue affecting the connectivity, I've silenced the warning with

`esp_log_level_set("wifi", ESP_LOG_ERROR);`

Certainly not a solution, but at least my terminal isn't cluttered with the warnings.

MaxwellAlan, 2024-10-08:
Hi

> It seems Espressif have made an update to fix this (thanks!) but I haven't had time to test it yet, just came back from holidays. The fix is on the master branch only at the moment.

Fix by this commit: 8daa2e308b5ac2f08d3dc506626e7d872588e362

closed 2024-10-08 by MaxwellAlan