espressif/esp-idf #14327: Little FS message at warning level.

ammaree, 2024-08-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

For some time, possibly last 12 months, we have been receiving a message at warning level.

'vfs esp_vfs_register_fd_range is successful for range <54; 64) and VFS ID 5'

Surely this message should be output at debug or verbose level?

The source is here:
https://github.com/espressif/esp-idf/blob/8e4454b285ad39881c5bf3f440d8be617a2f18a8/components/vfs/vfs.c#L161

Any chance this can be change to debug or verbose?

igrr, 2024-08-07:
I see that I changed the log level in when merging https://github.com/espressif/esp-idf/pull/7413, perhaps this was a mistake.  Thanks for bringing this up.

I'm surprised, though, that littlefs would call `esp_vfs_register_fd_range`. This function is only necessary for LwIP...

BrianPugh, 2024-08-07:
we don't call `esp_vfs_register_fd_range` in `esp_littlefs`, so maybe it's coming from another filesystem @ammaree is using? I don't have my esp-hardware handy to test out.

ammaree, 2024-08-07:
@igrr 

Thanks for the feedback, I will also check if anything else in our FW might call esp_vfs_register_fd_range.

On a similar note, another new wifi message at incorrect Error level has started coming through.
`wifi Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2`

I would guess this should also be at DEBUG or VERBOSE level?

ammaree, 2024-08-07:
@igrr 
@BrianPugh 

Have checked our code. We only call `esp_vfs_register` that calls `esp_vfs_register_common`

Nothing else that calls `esp_vfs_register_fd_range` to trigger this message, so I assume from LWIP ?

ammaree, 2024-09-27:
> I see that I changed the log level in when merging #7413, perhaps this was a mistake. Thanks for bringing this up.
> 
> I'm surprised, though, that littlefs would call `esp_vfs_register_fd_range`. This function is only necessary for LwIP...
@igrr 
Any chance we can have this fixed?

dizcza, 2024-11-30:
I have had this annoying warning for years...

ammaree, 2024-12-01:
Thanks for joining the club🤣

closed 2024-12-20 by espressif-bot

lptr, 2025-02-19:
Great to see this fixed, though I don't see the fix present on the `release/5.4` branch yet. Is that planned before 5.4.1? Thanks!

haberturdeur, 2025-02-26:
Hi, sorry for the delay, the patch was added to the `release/5.4` yesterday. It should make it to the rest of the release branches in time.

Thank you.