espressif/esp-idf #14330: Building pthread example for linux target fails with undefined reference error

cristianfunes79, 2024-08-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

# Error
Building pthread example using linux platform fails with undefined reference error
```sh
/usr/bin/ld: esp-idf/main/libmain.a(cpp_pthread.cpp.o): in function `create_config(char const*, int, int, int)':
/home/prj/pthread/main/cpp_pthread.cpp:79:(.text._Z13create_configPKciii+0x42): undefined reference to `esp_pthread_get_default_config'
/usr/bin/ld: esp-idf/main/libmain.a(cpp_pthread.cpp.o): in function `app_main':
/home/prj/pthread/main/cpp_pthread.cpp:90:(.text.app_main+0x41): undefined reference to `esp_pthread_get_default_config'
/usr/bin/ld: /home/prj/pthread/main/cpp_pthread.cpp:91:(.text.app_main+0x5b): undefined reference to `esp_pthread_set_cfg'
/usr/bin/ld: /home/prj/pthread/main/cpp_pthread.cpp:97:(.text.app_main+0xf0): undefined reference to `esp_pthread_set_cfg'
/usr/bin/ld: /home/prj/pthread/main/cpp_pthread.cpp:102:(.text.app_main+0x17d): undefined reference to `esp_pthread_set_cfg'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

# Steps to reproduce the issue:
- Copy `pthread/std::thread example`
- `idf.py --preview set-target linux`
- `idf.py build`

sudeep-mohanty, 2024-08-08:
Hi @cristianfunes79, Indeed the `pthread` examples in IDF have not been adapted to build with the `linux` target yet. We will take a look. Thanks for reporting it!

0xjakob, 2024-08-08:
@cristianfunes79 Please note that the "standard" pthread APIs are available on the Linux target, e.g. `pthread_create()`. The native Linux APIs will simply be used.

The `esp_pthread_*` APIs are indeed not available on the Linux target. This is because they are hardware-specific and some configurations are not represented on Linux. E.g., `esp_pthread_cfg_t::stack_alloc_caps` does not make any sense on the Linux target and `esp_pthread_cfg_t::prio` cannot easily be set on standard Linux systems without elevated privileges. Since you brought this up, however, we could think of adding some empty functions to allow building and running applications that use the `esp_pthread_*` APIs.

Please note, if you need to rely on some properties, such as checking if memory is in SPI-RAM, then the IDF fork of QEMU is probably a much better option here.

cristianfunes79, 2024-08-08:
Thanks! I'm working on that, I can create a PR once it is ready.

cristianfunes79, 2024-08-08:
I've already opened a PR: #14339

SoucheSouche, 2024-09-04:
Closing this issue since a follow up PR was created.

closed 2024-09-04 by SoucheSouche