espressif/esp-idf #14332: Initializing both LCD and Camera causes Interrupt wdt timeout

pkvarnfors, 2024-08-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5,3

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-S3-DevKitC-1 v1.1

### Power Supply used.

USB

### What is the expected behavior?

Initializing both LCD using esp_lcd and then Camera using esp_camera should result in working LCD and Camera without errors during initialization.

### What is the actual behavior?

After initializing the LCD using esp_lcd with 8-bit i8080 and then initializing esp_camera with an OV2640 a WDT Intrerrupt Timeout occurs. This happens when the ISR for camera VSYNC is installed in ll_cam_init_isr()

When running only the LCD it works and also when running only the camera it also works. 
When initializing the camera first and then the LCD the WDT happens during initialzing the LCD.

### Steps to reproduce.

The WDT happens during initialization of the periphials. The attached code should trigger the error during init_camera().

```
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_timer.h"
#include "esp_lcd_panel_io.h"
#include "esp_lcd_panel_vendor.h"
#include "esp_lcd_panel_ops.h"
#include "driver/gpio.h"
#include "esp_err.h"
#include "esp_log.h"

#include "esp_camera.h"

static const char *TAG = "esp32_i8080_camera";

#define LCD_PIXEL_CLOCK_HZ     (2 * 1000 * 1000)

#define LCD_BK_LIGHT_ON_LEVEL  1
#define LCD_BK_LIGHT_OFF_LEVEL !LCD_BK_LIGHT_ON_LEVEL
#define LCD_PIN_NUM_DATA0          8
[... 299 lines cut ...]
I (5680) lcd_panel.io.i80: Inter Alloc LCD  Reg  6004106c, Mask 2
...
I (2350) s3 ll_cam: Inter Alloc GDMA Reg  6003f00c, Mask 2
I (2360) s3 ll_cam: Inter Alloc CAM  Reg  6004106c, Mask 4
...

ISR 6004106c: 2-4 - Enabled
ISR 6003f00c: 0-2 - Enabled
ISR 6004106c: 2-2

ISR 6004106c: 2-4 - Enabled
ISR 6003f00c: 0-2 - Enabled
ISR 6004106c: 2-2

ISR 6004106c: 2-4 - Enabled
ISR 6003f00c: 0-2 - Enabled
ISR 6004106c: 2-2

ISR 6004106c: 2-4 - Enabled
ISR 6003f00c: 0-2 - Enabled
ISR 6004106c: 2-2

```
...
And then Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)

WangYuxin-esp, 2024-08-09:
@pkvarnfors Hi, Thank you for reporting this issue. We have already dealt with a similar issue [here](https://github.com/espressif/esp-idf/issues/12882), so could you check if the ESP32-camera you are currently using includes this [change](https://github.com/espressif/esp32-camera/pull/629)?

pkvarnfors, 2024-08-09:
Hi, 
Yes I have read that issue and it looks similar. I have tested the suggested solution in the referenced issue to run camera and lcd on different CPUs but with no success.

The changes references in your comment is included in our code which uses SDK v5.3 and esp_camera 2.0.10.

WangYuxin-esp, 2024-08-09:
The changes [here](https://github.com/espressif/esp32-camera/blob/master/Kconfig#L241) will add some modifications when `LCD_CAM_ISR_IRAM_SAFE` is enabled. Has this option been enabled in your configuration menu?

pkvarnfors, 2024-08-10:
Yes, it is. I have tested with different combinations of LCD and CAM IRAM_SAFE, and also to initialize the camera and lcd in different order. When initilizing the camera first and then the lcd, the camera init works and the WDT trigger in the lcd init, but after the panic reboot the WDT triggers in the camera init. The same if I use the hardware reset, it is just after a power cycle that I get the camera init to work.

```
I (1385) main_task: Calling app_main()
I (1395) esp32_i8080_camera: Start camera_task
I (1395) esp32_i8080_camera: Initilizing camera
D (1405) gdma: new group (0) at 0x3fca70c4
D (1405) gdma: new pair (0,0) at 0x3fca7150
D (1415) gdma: new rx channel (0,0) at 0x3fca7088
I (1415) s3 ll_cam: DMA Channel=0
I (1425) cam_hal: cam init ok
D (1425) camera: Enabling XCLK output
D (1425) camera: Initializing SCCB
I (1435) sccb: pin_sda 7 pin_scl 6
I (1435) sccb: sccb_i2c_port=0
D (1435) intr_alloc: Connected src 42 to int 2 (cpu 1)
D (1445) camera: Searching for camera address
I (1455) camera: Detected camera at address=0x30
I (1455) camera: Detected OV2640 camera
D (1455) ov2640: OV2640 Attached
I (1465) camera: Camera PID=0x26 VER=0x42 MIDL=0x7f MIDH=0xa2
D (1465) camera: Doing SW reset of sensor
I (1565) s3 ll_cam: node_size: 3840, nodes_per_line: 1, lines_per_node: 6
I (1565) s3 ll_cam: dma_half_buffer_min:  3840, dma_half_buffer: 15360, lines_per_half_buffer: 24, dma_buffer_size: 30720
I (1565) cam_hal: buffer_size: 30720, half_buffer_size: 15360, node_buffer_size: 3840, node_cnt: 8, total_cnt: 10
I (1585) cam_hal: Allocating 153600 Byte frame buffer in PSRAM
D (1585) intr_alloc: Connected src 66 to int 3 (cpu 1)
D (1595) intr_alloc: Connected src 24 to int 3 (cpu 1)
I (1595) cam_hal: cam config ok
D (1605) camera: Setting frame size to 320x240
I (1605) ov2640: Set PLL: clk_2x: 1, clk_div: 3, pclk_auto: 1, pclk_div: 8
I (1685) esp32_i8080_camera: Camera Init OK
I (1685) esp32_i8080_camera: Camera Init done
I (2895) esp32_i8080_camera: Turn off LCD backlight
I (2895) gpio: GPIO[1]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (2895) esp32_i8080_camera: Initialize Intel 8080 bus
D (2895) intr_alloc: Connected src 24 to int 8 (cpu 0)
D (2905) gdma: new tx channel (0,0) at 0x3fcaf840
D (2905) lcd_panel.io.i80: new i80 bus(0) @0x3fcaf5b8, 40 dma nodes
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 

Core  1 register dump:
PC      : 0x4037d74b  PS      : 0x00060034  A0      : 0x8037693b  A1      : 0x3fc96ef0
0x4037d74b: vPortClearInterruptMaskFromISR at C:/Apps/Espressif/frameworks/v5.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:560
 (inlined by) vPortExitCritical at C:/Apps/Espressif/frameworks/v5.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:504

A2      : 0x3fc95758  A3      : 0xffffffff  A4      : 0x82018498  A5      : 0x00060023
A6      : 0x3fc95fbc  A7      : 0x0000abab  A8      : 0x40001c38  A9      : 0x00000000
0x40001c38: __call__xtos_set_intlevel in ROM

A10     : 0x00060021  A11     : 0x00000000  A12     : 0xb33fffff  A13     : 0x3fc9d9e0
A14     : 0x00060323  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
Core  1 was running in ISR context:
EPC1    : 0x42025867  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4037d74b
0x42025867: uart_hal_write_txfifo at C:/Apps/Espressif/frameworks/v5.3/esp-idf/components/hal/uart_hal_iram.c:27
0x4037d74b: vPortClearInterruptMaskFromISR at C:/Apps/Espressif/frameworks/v5.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:560
 (inlined by) vPortExitCritical at C:/Apps/Espressif/frameworks/v5.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:504

Backtrace: 0x4037d748:0x3fc96ef0 0x40376938:0x3fc96f10 0x403779e5:0x3fc96f30 0x4037904f:0x3fc9da50 0x420039fe:0x3fc9da70 0x4037e2e9:0x3fc9da90 0x4037d3a5:0x3fc9dab0
0x4037d748: vPortClearInterruptMaskFromISR at C:/Apps/Espressif/frameworks/v5.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:560
 (inlined by) vPortExitCritical at C:/Apps/Espressif/frameworks/v5.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:504
0x40376938: shared_intr_isr at C:/Apps/Espressif/frameworks/v5.3/esp-idf/components/esp_hw_support/intr_alloc.c:454
0x403779e5: _xt_lowint1 at C:/Apps/Espressif/frameworks/v5.3/esp-idf/components/xtensa/xtensa_vectors.S:1240
0x4037904f: xt_utils_wait_for_intr at C:/Apps/Espressif/frameworks/v5.3/esp-idf/components/xtensa/include/xt_utils.h:82
 (inlined by) esp_cpu_wait_for_intr at C:/Apps/Espressif/frameworks/v5.3/esp-idf/components/esp_hw_support/cpu.c:55
0x420039fe: esp_vApplicationIdleHook at C:/Apps/Espressif/frameworks/v5.3/esp-idf/components/esp_system/freertos_hooks.c:58
0x4037e2e9: prvIdleTask at C:/Apps/Espressif/frameworks/v5.3/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4344 (discriminator 1)
0x4037d3a5: vPortTaskWrapper at C:/Apps/Espressif/frameworks/v5.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

Core  0 register dump:
PC      : 0x40379052  PS      : 0x00060834  A0      : 0x82003a01  A1      : 0x3fc9d2f0
0x40379052: esp_cpu_wait_for_intr at C:/Apps/Espressif/frameworks/v5.3/esp-idf/components/esp_hw_support/cpu.c:64

[... cut ...]
0x420039fe: esp_vApplicationIdleHook at C:/Apps/Espressif/frameworks/v5.3/esp-idf/components/esp_system/freertos_hooks.c:58
0x4037e2e9: prvIdleTask at C:/Apps/Espressif/frameworks/v5.3/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4344 (discriminator 1)
0x4037d3a5: vPortTaskWrapper at C:/Apps/Espressif/frameworks/v5.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

```

Attached is the complete VS Code project where the camera is running in its own task on CPU1.

[esp32_i8080_camera.zip](https://github.com/user-attachments/files/16568892/esp32_i8080_camera.zip)