espressif/esp-idf #14335: service port=0 in mdns_service_add causes a error

nopnop2002, 2024-08-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2-420-gc4fc74a6de

### Espressif SoC revision.

Chip is ESP32D0WDQ6 (revision 1)

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 Development Board

### Power Supply used.

USB

### What is the expected behavior?

service port=0 in mdns_service_add not causes a error.

### What is the actual behavior?

service port=0 in mdns_service_add causes a error

### Steps to reproduce.

```
    //initialize mDNS
    ESP_ERROR_CHECK( mdns_init() );
    //set mDNS hostname (required if you want to advertise services)
    ESP_ERROR_CHECK( mdns_hostname_set(CONFIG_MY_HOSTNAME));
    ESP_LOGI(TAG, "mdns hostname set to: [%s]", CONFIG_MY_HOSTNAME);

    //initialize service
    //ESP_ERROR_CHECK( mdns_service_add(NULL, "_device-info", "_tcp", 80, NULL, 0) );
    ESP_ERROR_CHECK( mdns_service_add(NULL, "_device-info", "_tcp", 0, NULL, 0) );
```

### Debug Logs.

```plain
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x400d69ae
0x400d69ae: initialise_mdns at /home/nop/rtos/esp-idf-mdns/query-host1/main/main.c:144 (discriminator 1)

file: "./main/main.c" line 144
func: initialise_mdns
expression: mdns_service_add(NULL, "_device-info", "_tcp", 0, NULL, 0)

abort() was called at PC 0x40089257 on core 0
0x40089257: _esp_error_check_failed at /home/nop/esp-idf/components/esp_system/esp_err.c:50
```
```

david-cermak, 2025-01-10:
Thanks for reporting, fixing in https://github.com/espressif/esp-protocols/pull/730

nopnop2002, 2025-01-10:
@david-cermak 

Will this fix not cause the following error?

```
ESP_ERROR_CHECK( mdns_service_add(NULL, "_device-info", "_tcp", 0, NULL, 0) );
```

david-cermak, 2025-01-10:
Seems like it:
```cpp
    ESP_ERROR_CHECK( mdns_init() );
    ESP_ERROR_CHECK( mdns_hostname_set(hostname) );
    ESP_LOGI(TAG, "mdns hostname set to: [%s]", hostname);
    ESP_ERROR_CHECK( mdns_instance_name_set(EXAMPLE_MDNS_INSTANCE) );
    ESP_ERROR_CHECK( mdns_service_add(NULL, "_device-info", "_tcp", 0, NULL, 0) );
```

```
$ avahi-browse -a -r
+ enpxxxx IPv6 ESP32 with mDNS                               _device-info._tcp    local
= enpxxxx IPv6 ESP32 with mDNS                               _device-info._tcp    local
   hostname = [esp32-mdns.local]
   address = [fe80::xxxx:xxxx:xxxx:xxxx]
   port = [0]
   txt = []
```

nopnop2002, 2025-01-10:
I use this.
https://components.espressif.com/components/espressif/mdns/versions/1.4.3

but don't work.
```
    // initialize mDNS
    ESP_ERROR_CHECK( mdns_init() );

    // set mDNS hostname
    ESP_ERROR_CHECK( mdns_hostname_set("esp32-server") );

    // set default mDNS instance name
    ESP_ERROR_CHECK( mdns_instance_name_set("ESP32 with mDNS") );

    // add service
    //ESP_ERROR_CHECK( mdns_service_add(NULL, "_device-info", "_tcp", 80, NULL, 0) );
    ESP_ERROR_CHECK( mdns_service_add(NULL, "_device-info", "_tcp", 0, NULL, 0) );
```

david-cermak, 2025-01-13:
The fix hasn't been merged yet. It's not included in the v1.4.3 (pushed only a quick release the other day to publish that "zero item subtypes" fix)

This issue will be closed once the fix lands on master.

To test it locally you can:

1) **Point to your local mdns component**

```patch
#  idf_component.yml
dependencies:
  espressif/mdns:
+  override_path: /path/to/esp_protocol_repo/components/mdns
```

2) **Fetch the referenced PR**

```bash
$ cd /path/to/esp_protocol_repo/
$ git fetch origin pull/730/head && git checkout FETCH_HEAD
```

nopnop2002, 2025-01-13:
@david-cermak 

Thank you for your comment.

david-cermak, 2025-01-24:
The fix's been merged and published in mdns `v1.5`

closed 2025-01-24 by david-cermak

nopnop2002, 2025-01-24:
We have confirmed that it works properly with mdns1.5.1.