espressif/esp-idf #14336: I2C command link in API reference

dpnebert, 2024-08-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm using an ESP32-S3 with ESP-IDF v.5.2.1.  In the API reference for i2c (https://docs.espressif.com/projects/esp-idf/en/v5.2.2/esp32s3/api-reference/peripherals/i2c.html), it states, "In order to organize the process, the driver uses a command link, that should be populated with a sequence of commands and then passed to I2C controller for execution."  But it see in, "[Migration Guides] » [Migration from 5.1 to 5.2] » Peripherals" (https://docs.espressif.com/projects/esp-idf/en/v5.2/esp32/migration-guides/release-5.x/5.2/peripherals.html), it states, "i2c_cmd_link_xxx functions have been removed, user doesn't need to use link to link commands on its own."

I've used 'i2c_master_transmit_receive' to read a value from the device, but I'm struggling with how to do a 'write-write' using the write_buffer and an array.

Could more information about what strategy replaced the command link would be great! Thanks!

nopnop2002, 2024-08-18:
A new i2c driver is now available in ESP-IDF V5.2.

I hope this is helpful to read a value from the device.

https://github.com/espressif/esp-idf/tree/master/examples/peripherals/i2c/i2c_tools

DeltaCore, 2024-09-03:
I have this issue to. I need the ability to create an i2c command with a repeated start command:
```
-> start, i2c address + write bit, 2 byte data
-> start, i2c address + write bit, 2 byte data
-> start, i2c address +  read bit, 9 byte data
-> stop
```

With the new API function `i2c_master_transmit_receive` i can't insert the repeated start operation into that transaction.

I would love to move to the new driver because of the nice way to configure the master and devices, but this is hindering me for now.

Any plans to reintroduce that style of creating i2c transactions?

dpnebert, 2024-09-05:
An hour after you posted, I stared typing up a response, but then realized I didn't know what you were truly going for.

Actually just did it again.  I found a good issue for illustration: [https://github.com/espressif/esp-idf/issues/13952](https://github.com/espressif/esp-idf/issues/13952)

I believe you'll have to split up this to three calls:
```
uint8_t data1[3] = {0x01, 0x10, 0x20};
i2c_master_transmit(handle, data1, 3, -1); // this will write 0x10 to register address 0x01,  and 0x20 to register address 0x02

uint8_t data2[3] = {0x03, 0x10, 0x20};
i2c_master_transmit(handle, data1, 3, -1); // this will write 0x10 to register address 0x03,  and 0x20 to register address 0x04

uint8_t data3[1] = [0x04};
uint8_t data4[9] = [0x10, 0x11, 0x12, 0x13, 0x14, 0x15, 0x16, 0x17, 0x18};
i2c_master_transmit(handle, data3, 1, data4, 9, -1); // this will write 0x04 to set the register address to be read from.  Then will perform 9 reads.

I believe this to be correct.  If anyone could correct me, I'd appreciate helping me get better at the "new" way of doing ESP32s.
```

DeltaCore, 2024-09-06:
This sadly won't work in my case since a `i2c_master_transmit` puts a stop command at the end. Your suggestion would end up with this i2c activity on the bus:
```
# first i2c_master_transmit
<start>, <dev-addr> | WRITE_BIT, 0x01, 0x10, 0x20, <stop>

# second i2c_master_transmit
<start>, <dev-addr> | WRITE_BIT, 0x03, 0x10, 0x20, <stop>

# i2c_master_transmit_receive
<start>, <dev-addr> | WRITE_BIT, 0x04, <start>, <data[0]>, ... , <data[8]>, <stop>
```

Unfortunatly the sensor i am working with is very strict about it's commands.

DeltaCore, 2024-09-06:
I thought about that as well, but `i2c_master_transmit` internally uses `s_i2c_synchronous_transaction` or `s_i2c_synchronous_transaction` respectively which are private functions I can't access easily.

I would much rather have an API that allows me to create such specific i2c bus transactions, much like the old API.
Don't get me wrong, the new API is nice, but the advanced use case is just missing for my requirements.
The i2c_operation_t just looks like what I want but I can't access the needed functions for that :(

Thenicolaibulow, 2024-11-27:
@DeltaCore  - Did you ever find a nice work around for this?

I'm faced with a issue very similar to yours, whilst implementing I2C comms with a ADS7142 TI ADC, which requires custom operation-mode data, when reading and writing to it. I was struggling to get the old driver correctly writing to it, whilst reading went fine. 

However, whilst trying to debug said behaviour, I decided to try and implement the new driver instead, yet, just as you describe, it lacks the functionality to 'build' the I2C frames manually.

mythbuster5, 2025-04-11:
Hi guys in this thread. We already provide `i2c_master_multi_buffer_transmit` and `i2c_master_execute_defined_operations` for flexible usage on master. We will also backport this to v5.2 and v5.3. Hope this can solve your issue. And will close this issue. Feel free to reopen if you have any questions.

mythbuster5, 2025-04-17:
Seems no more update, I will conclude this issue. Feel free to reopen if you have more updates, thanks

closed 2025-04-17 by mythbuster5