espressif/esp-idf #14337: multiple definition of `opendir'

diplfranzhoepfinger, 2024-08-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

where can this come from ? 

https://gist.github.com/diplfranzhoepfinger/1c969e31ae6fc1c1c913a9f6422840a4

diplfranzhoepfinger, 2024-08-08:
seems to be similar with: 
https://github.com/espressif/esp-idf/issues/14174

pacucha42, 2024-08-09:
Hi @diplfranzhoepfinger - yes, this is duplicate, please use the patch from https://github.com/espressif/esp-idf/issues/14174#issuecomment-2226783209 before we provide a full fix. Thank you

closed 2024-08-09 by pacucha42