espressif/esp-idf #14340: Zero-copy `esp_netif_transmit` when prepending headers

redfast00, 2024-08-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

If you want to make a custom network adaptor with ESP-NETIF, you need to implement three functions:

- [esp_netif_transmit()](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/network/esp_netif_driver.html#_CPPv418esp_netif_transmitP11esp_netif_tPv6size_t)
-  [esp_netif_free_rx_buffer()](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/network/esp_netif_driver.html#_CPPv424esp_netif_free_rx_bufferPvPv)
- [esp_netif_receive()](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/network/esp_netif.html#_CPPv417esp_netif_receiveP11esp_netif_tPv6size_tPv)

The signature of the transmit function is `esp_err_t esp_netif_transmit(esp_netif_t *esp_netif, void *data, size_t len)`. This function is called by the network stack. The implementation needs to transmit the data frame to the hardware.

A problem I'm currently facing while implementing esp_netif_transmit, is that I need to prepend some headers to the data before transmitting it via DMA. Since I don't control which buffer I'm passed, I can't prepend the headers directly to the data, so I need to allocate a new, bigger buffer, add the headers and copy the data.

Is there a way to control which buffers are passed to esp_netif_transmit? That way I could add some headroom to the buffer.

Alternatively, is there a way to chain buffers before sending them to the 802.11 MAC TX slots? That way, I could have one DMA entry with the headers which points to the next one that has the data.

david-cermak, 2024-08-13:
You can use [`esp_netif_transmit_wrap()`](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/network/esp_netif_driver.html#_CPPv423esp_netif_transmit_wrapP11esp_netif_tPv6size_tPv) and use the last argument as opaque pointer to your DMA buffer struct.
the default wlan port passes lwip's pbuf as the network stack buffer (these pbufs support prep-ending and chaining, too)

https://github.com/espressif/esp-idf/blob/8e4454b285ad39881c5bf3f440d8be617a2f18a8/components/esp_netif/lwip/netif/wlanif.c#L92

david-cermak, 2024-10-30:
@redfast00 Any update on this, can we close it?

closed 2024-10-30 by redfast00