espressif/esp-idf #14343: ESP32-S3 I2C master fails after receiving a NACK in data byte

mircopz, 2024-08-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Espressif SoC revision.

ESP32-S3 (revision v0.1)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-DevKitC-1 v1.1

### Power Supply used.

USB

### What is the expected behavior?

I2C works normally after an ACK error received on data byte.

### What is the actual behavior?

The I2C transaction following an error fails. On the bus I can see a byte relating to the previous transaction, while the address of the new transaction should be there. After this failure, everything resumes normally.

### Steps to reproduce.

My devKit is acting as I2C master. On the I2C bus an external EEPROM (M24C02) is connected with address 0x50.
This is the code to generate an I2C transaction, where I want to write its registers from address 0x00 with values 0xFF, 0xFE...

```c
uint16_t address = 0x50;
uint8_t buff[17] = { 0x00, 0xFF, 0xFE, 0xFD, 0xFC, 0xFB, 0xFA, 0xF9, 0xF8, 0xF7, 0xF6, 0xF5, 0xF4, 0xF3, 0xF2, 0xF1, 0xF0 };
size_t size = 17;

...   
    uint8_t cmd_buff[I2C_LINK_RECOMMENDED_SIZE(1)] = { 0 };
    cmd = i2c_cmd_link_create_static(cmd_buff, I2C_LINK_RECOMMENDED_SIZE(1));
    ret = i2c_master_start(cmd);
    if (ret != ESP_OK) {
        goto end;
    }
    ret = i2c_master_write_byte(cmd, (address << 1) | I2C_MASTER_WRITE, true);
    if (ret != ESP_OK) {
        goto end;
    }
    if(size){
        ret = i2c_master_write(cmd, buff, size, true);
        if (ret != ESP_OK) {
            goto end;
        }
    }
    ret = i2c_master_stop(cmd);
    if (ret != ESP_OK) {
        goto end;
    }
    ret = i2c_master_cmd_begin((i2c_port_t)i2c_num, cmd, timeOutMillis / portTICK_RATE_MS);

end:
    if(cmd != NULL){
        i2c_cmd_link_delete_static(cmd);
    }
...
```

This is a normal transaction (working)

[image]

Then I set the EEPROM in write protect mode. In this case, each data byte is followed by a NACK. As I expected, the same transaction fails with a NACK in first data byte, as you can see in the following sequence.

[image]

The problem arises now. If I try to make any another I2C transaction with any I2C device (both the EEPROM and a generic sensor), I get the following situation.

[image]

It seems that, even if tx FIFO is reset when the new transaction begins, the outgoing byte on SDA line is the byte that was queued after the previous byte not ACKed (0xFE was the next byte to 0xFF).
It seems that a byte is moved from tx FIFO to data shifter before obtaining an ACK/NACK, and its content are not reset when clearing the FIFO.

[image]

After this failure, everything goes back to normal and I2C starts working again.
The same problem does not appear on ESP32. It seems related to ESP32-S3 I2C controller.
Has anyone encountered a similar problem?

mythbuster5, 2025-01-22:
Hi, sorry for the late reply... May I know how to `set the EEPROM in write protect mode` because I want to reproduce this locally, and I need some clue to check whether this is a software issue or hardware issue,

mircopz, 2025-01-22:
> Hi, sorry for the late reply... May I know how to `set the EEPROM in write protect mode` because I want to reproduce this locally, and I need some clue to check whether this is a software issue or hardware issue,

To disable EEPROM write operations write control (WC) pin must be driven high.

[image]

AxelLin, 2025-02-24:
@mythbuster5 Can you reproduce the issue now?