espressif/esp-idf #14345: ESP32-S3 USB Host CDC Generic

antunesls, 2024-08-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

I have tried all available examples in ESP-IDF related to USB communication, including both the Host CDC and Host USB libraries, but I am unable to establish communication with my UPS device.

I can successfully communicate with other devices using known drivers like CP210x, but I am struggling with my Ragtech Easy Pro 1200 UPS, which uses a generic driver.

Here is the output of the dmesg | tail command from Linux with my UPS connected:

[ 7855.208112] usb 1-1.3: new full-speed USB device number 3 using xhci_hcd
[ 7855.320032] usb 1-1.3: New USB device found, idVendor=04d8, idProduct=000a, bcdDevice= 1.00
[ 7855.320049] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 7855.320055] usb 1-1.3: Product: USB Serial Port
[ 7855.320059] usb 1-1.3: Manufacturer: Ragtech Sistemas de Energia
[ 7855.419658] cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device
[ 7855.419897] usbcore: registered new interface driver cdc_acm
[ 7855.419902] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters

Despite this, I am unable to establish serial communication with the UPS from the ESP32-S3. 

I have been attempting to resolve this issue for a month without success. I have also tried using MicroPython and TinyUSB with no results.

Any assistance or guidance on how to resolve this issue would be greatly appreciated.

Thank you.

roma-jam, 2024-08-12:
Hi @antunesls , 

Thanks for reporting the problem. 

Could you specify the issue which prevents the Host CDC and USB Host libraries to communicate with the device?
 
If that is possible, share the debug output log with verbosity level=VERBOSE (can be changed via  `idf.py menuconfig`) for: 
1. `usb/host/cdc/*` example you are using
2. `usb/host/usb_host_lib` example for your device

Thanks.

antunesls, 2024-08-12:
For some reason the device take a few seconds more than others to connect, I just have to wait :? .

Now its connecting but shows up a buffer error.

My buffer is set to 4096 in the code and in the menuconfig. (CONFIG_USB_HOST_CONTROL_TRANSFER_MAX_SIZE=4096)

[usb_cdc_example_main.zip](https://github.com/user-attachments/files/16580855/usb_cdc_example_main.zip)

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x178c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xcb8
load:0x403cb700,len:0x2db0
entry 0x403c8914
I (27) boot: ESP-IDF v5.3 2nd stage bootloader

I (27) boot: compile time Aug 10 2024 12:23:54
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode : DIO
I (43) boot.esp32s3: SPI Flash Size : 2MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label Usage Type ST Offset Length
I (64) boot: 0 nvs WiFi data 01 02 00009000 00006000
I (71) boot: 1 phy_init RF data 01 01 0000f000 00001000
I (79) boot: 2 factory factory app 00 00 00010000 00100000
I (86) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=0c954h ( 51540) map
I (108) esp_image: segment 1: paddr=0001c97c vaddr=3fc92b00 size=02904h ( 10500) load
I (111) esp_image: segment 2: paddr=0001f288 vaddr=40374000 size=00d90h ( 3472) load
I (117) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=21c44h (138308) map
I (149) esp_image: segment 4: paddr=00041c6c vaddr=40374d90 size=0dca8h ( 56488) load
I (168) boot: Loaded app from partition at offset 0x10000
I (168) boot: Disabling RNG early entropy source...
I (180) cpu_start: Multicore app
I (189) cpu_start: Pro cpu start user code
I (189) cpu_start: cpu freq: 160000000 Hz
I (189) app_init: Application information:
I (192) app_init: Project name: cdc-host-example
I (198) app_init: App version: 1
I (202) app_init: Compile time: Aug 10 2024 14:04:58
I (208) app_init: ELF file SHA256: 4454c768a...
I (213) app_init: ESP-IDF: v5.3
I (218) efuse_init: Min chip rev: v0.0
I (223) efuse_init: Max chip rev: v0.99
I (228) efuse_init: Chip rev: v0.2
I (232) heap_init: Initializing. RAM available for dynamic allocation:
I (240) heap_init: At 3FC95C78 len 00053A98 (334 KiB): RAM
I (246) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (252) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (258) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (265) spi_flash: detected chip: winbond
I (269) spi_flash: flash io: dio
W (273) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (286) sleep: Configure to isolate all GPIO pins in sleep state
I (293) sleep: Enable automatic switching of GPIO sleep configuration
I (300) main_task: Started on CPU0
I (310) main_task: Calling app_main()
I (310) USB-CDC: Instalando USB Host
I (340) USB-CDC: Instalando driver CDC-ACM
I (340) USB-CDC: Abrindo dispositivo CDC ACM 0x04D8:0x000A...
E (440) cdc_acm: cdc_acm_host_open(731): USB device with VID: 0x04D8, PID: 0x000A not found
I (440) USB-CDC: Falha ao abrir dispositivo
I (440) USB-CDC: Abrindo dispositivo CDC ACM 0x04D8:0x000A...
E (540) cdc_acm: cdc_acm_host_open(731): USB device with VID: 0x04D8, PID: 0x000A not found
I (540) USB-CDC: Falha ao abrir dispositivo
I (540) USB-CDC: Abrindo dispositivo CDC ACM 0x04D8:0x000A...
E (640) cdc_acm: cdc_acm_host_open(731): USB device with VID: 0x04D8, PID: 0x000A not found
I (640) USB-CDC: Falha ao abrir dispositivo
I (640) USB-CDC: Abrindo dispositivo CDC ACM 0x04D8:0x000A...
E (740) cdc_acm: cdc_acm_host_open(731): USB device with VID: 0x04D8, PID: 0x000A not found
I (740) USB-CDC: Falha ao abrir dispositivo
I (740) USB-CDC: Abrindo dispositivo CDC ACM 0x04D8:0x000A...
E (750) USBH: usb_transfer_t num_bytes > data_buffer_size
E (750) cdc_acm: cdc_acm_host_send_custom_request(1257): CTRL transfer failed
E (760) cdc_acm: cdc_acm_host_line_coding_set(1196):
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x4200752a
0x4200752a: app_main at C:/Users/lukss/cdc-host-example/main/usb_cdc_example_main.c:218 (discriminator 1)

file: "./main/usb_cdc_example_main.c" line 218
func: app_main
expression: cdc_acm_host_line_coding_set(cdc_dev, &line_coding)

abort() was called at PC 0x40379bd7 on core 0
0x40379bd7: _esp_error_check_failed at C:/Users/lukss/esp/v5.3/esp-idf/components/esp_system/esp_err.c:49

Backtrace: 0x40375abe:0x3fc98010 0x40379be1:0x3fc98030 0x4038092d:0x3fc98050 0x40379bd7:0x3fc980c0 0x4200752a:0x3fc980f0 0x420213bb:0x3fc98150 0x4037a571:0x3fc98180
0x40375abe: panic_abort at C:/Users/lukss/esp/v5.3/esp-idf/components/esp_system/panic.c:463
0x40379be1: esp_system_abort at C:/Users/lukss/esp/v5.3/esp-idf/components/esp_system/port/esp_system_chip.c:92
0x4038092d: abort at C:/Users/lukss/esp/v5.3/esp-idf/components/newlib/abort.c:38
0x40379bd7: _esp_error_check_failed at C:/Users/lukss/esp/v5.3/esp-idf/components/esp_system/esp_err.c:49
0x4200752a: app_main at C:/Users/lukss/cdc-host-example/main/usb_cdc_example_main.c:218 (discriminator 1)
0x420213bb: main_task at C:/Users/lukss/esp/v5.3/esp-idf/components/freertos/app_startup.c:208
0x4037a571: vPortTaskWrapper at C:/Users/lukss/esp/v5.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

ELF file SHA256: 4454c768a

tore-espressif, 2024-08-12:
Hi @antunesls thank you for the log. 

There indeed might be a bug, because we always allocate bMaxPacketSize0 for Endpoint 0. https://github.com/espressif/esp-usb/blob/master/host/class/cdc/usb_host_cdc_acm/cdc_acm_host.c#L572 . Up until now, we have not seen this problem because USB devices usually implement 64 bytes MPS for EP0, so there isn't a problem.

Could you confirm this by running `lsusb -v` on your linux machine a copy relevant part of the output here?

your example fails on `cdc_acm_host_line_coding_set/get` API (which uses EP0). I think you might not need this at all. This API is useful only for real USB<->UART converters. Could you just remove these calls and try again?

antunesls, 2024-08-12:
Hi @tore-espressif , 

Its runs but I not having the same result as the intend when a I used plugged in a machine linux x86 , so I was trying set the boundrate and other configs to try solved it.

Follow the resquest (lsusb -v) :

Bus 001 Device 005: ID 04d8:000a Microchip Technology, Inc. CDC RS-232 Emulation Demo
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            2 Communications
  bDeviceSubClass         2 Abstract (modem)
  bDeviceProtocol         0
  bMaxPacketSize0         8
  idVendor           0x04d8 Microchip Technology, Inc.
  idProduct          0x000a CDC RS-232 Emulation Demo
  bcdDevice            1.00
  iManufacturer           1 Ragtech Sistemas de Energia
  iProduct                2 USB Serial Port
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0043
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      1 AT-commands (v.25ter)
      iInterface              0
      CDC Header:
        bcdCDC               1.10
      CDC Call Management:
        bmCapabilities       0x00
        bDataInterface          1
      CDC ACM:
        bmCapabilities       0x02
          line coding and serial state
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               2
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0

tore-espressif, 2024-08-12:
> bMaxPacketSize0 8

This confirms the problem. 

As a temporary workaround you can try to avoid calling the set/get line coding API, until it is fixed in esp-usb

antunesls, 2024-08-12:
OK, while the bug is not fixed, I'll resume my issue with the comunication, so maybe my issue could be not related with this bug.

If someone could help me.

My base is my raspberry pi 4 , running raspbery OS.

Follow my code in python (it works).

import serial
import time
import datetime

import json

# Configuração Serial
cfg_serial_port = 'COM5'
cfg_serial_interval = 5

# payloads
cmd_req1 = bytes.fromhex("FF FE 00 8E 01 8F")
cmd_req2 = bytes.fromhex("AA 04 00 80 1E 9E")
cmd_get_data = bytes.fromhex("AA 04 00 80 1E 9E")

power_current_kwh_out = 0.0
power_current_kwh_in = 0.0
power_last_update = time.time()

def logData(text, end):
    current_datetime = datetime.datetime.now()
    formatted_datetime = current_datetime.strftime("%d/%b/%Y %H:%M:%S")
    if end:
        print(f"[{formatted_datetime}] {text}", end=" ")
    else:
        print(f"[{formatted_datetime}] {text}")

def linear(val, a, b, decimals = 1):
    return round(b + val * a, decimals)

def processPower(wOut, wIn):
    global power_last_update
    global power_current_kwh_out
    global power_current_kwh_in

    curr = time.time()
    diff = curr - power_last_update # in seconds
    power_last_update = curr

    slice = (wOut / 1000) * (diff / 3600)
    power_current_kwh_out += slice
    
    slice = (wIn / 1000) * (diff / 3600)
    power_current_kwh_in += slice

def build_payload(buffer):
    iOut = linear(buffer[0x0D], 0.1152, 0, 2)

    vOut = linear(buffer[0x1E], 0.555, 0, 2)
    wOut = iOut*vOut

    vIn = linear(buffer[0x0C], 1.06, 0, 2)
    wIn = iOut*vIn # não tem "iIn"

    processPower(wOut, wIn)

    payload = {
        "Power_Out_Percent": {"value": linear(buffer[0x0E], 1, 0), "unit": "%"},
        "Current_Out": {"value": iOut, "unit": "A"},
        "Voltage_Out": {"value": round(vOut, 1), "unit": "V"},
        "Voltage_In": {"value": round(vIn, 1), "unit": "V"},
        "Power_Out": {"value": round(wOut, 1), "unit": "W"},
        "Power_In": {"value": round(wIn, 1), "unit": "W"},
        "Energy_Out": {"value": round(power_current_kwh_out, 2), "unit": "kWh"},
        "Energy_In": {"value": round(power_current_kwh_in, 2), "unit": "kWh"},
        "Temperature": {"value": linear(buffer[0x0F], 1, 0), "unit": "C"},
        "Battery_State": {"value": linear(buffer[0x08], 0.392, 0), "unit": "%"},
        "Battery_Voltage": {"value": linear(buffer[0x0B], 0.0671, 0), "unit": "V"},
        "Frequency": {"value": linear(buffer[0x18], -0.1152, 65), "unit": "Hz"},
    }

    return payload

def send_and_ignore(serial_port, command, wait_time=0.1):
    while True:
        serial_port.write(command)
        time.sleep(wait_time)
        if serial_port.in_waiting > 0:
            logData(str(serial_port.in_waiting)+" Available", False)
            serial_port.read()
            return
        logData("No response ...", False)

def mainLoop():
    # Configuração da porta serial
    logData(f"serial open: {cfg_serial_port}", False)
    serial_port = serial.Serial(cfg_serial_port)
    if serial_port.isOpen():
        logData(serial_port.name + " is open…", False)
    else:
        logData(f"serial fail: {ret}", False)
        time.sleep(10)
        return

    serial_port.write(cmd_req1)
    serial_port.write(cmd_req2)

    while True:
        time.sleep(cfg_serial_interval)

        serial_port.write(cmd_get_data)
        logData(f"Requesting Data... ", True)

        #wait = 100
        while serial_port.in_waiting < 31:
            time.sleep(0.1)
            print(serial_port.in_waiting, end=" ")
                  
        print()

        time.sleep(0.1)
        buffer = serial_port.read(serial_port.in_waiting)

        # Monta o payload
        payload = build_payload(buffer)

        strPayload = json.dumps(payload)
        logData(strPayload, False)
        
       
while True:
    mainLoop()

The output:

antunesls@valkyrie:~ $ python ups2.py
[12/Aug/2024 10:46:42] serial open: /dev/ttyACM0
[12/Aug/2024 10:46:42] /dev/ttyACM0 is open…
[12/Aug/2024 10:46:47] Requesting Data...  11 32
[12/Aug/2024 10:46:47] {"Power_Out_Percent": {"value": 3, "unit": "%"}, "Current_Out": {"value": 23.04, "unit": "A"}, "Voltage_Out": {"value": 0.0, "unit": "V"}, "Voltage_In": {"value": 209.9, "unit": "V"}, "Power_Out": {"value": 0.0, "unit": "W"}, "Power_In": {"value": 4835.6, "unit": "W"}, "Energy_Out": {"value": 0.0, "unit": "kWh"}, "Energy_In": {"value": 0.01, "unit": "kWh"}, "Temperature": {"value": 2, "unit": "C"}, "Battery_State": {"value": 53.7, "unit": "%"}, "Battery_Voltage": {"value": 16.8, "unit": "V"}, "Frequency": {"value": 47.0, "unit": "Hz"}}

But my code in the ESP32S3 attached not return the response more than 31 cacracters, like the python code.

It could be related with the boudrate ?

tore-espressif, 2024-08-12:
> It could be related with the boudrate ?

From the information you provided, it doesn't seem be related.

Looking at your `handle_rx` funciton
``` c
// Callback para processar dados recebidos e montar o payload
static bool handle_rx(const uint8_t *data, size_t data_len, void *arg)
{
    ESP_LOGI(TAG, "Dados recebidos");
    ESP_LOG_BUFFER_HEXDUMP(TAG, data, data_len, ESP_LOG_INFO);

    // Verifica se recebeu dados suficientes
    if (data_len >= 31)
    {
        char json_payload[512];
        build_payload(data, json_payload, sizeof(json_payload));
        ESP_LOGI(TAG, "Payload: %s", json_payload);
    }
    else
    {
        vTaskDelay(pdMS_TO_TICKS(100));
    }

    return true;
}
```

You should not call any blocking API (`vTaskDelay`) in the callback. Could you do something as simple as
``` c
static bool handle_rx(const uint8_t *data, size_t data_len, void *arg) {
    ESP_LOG_BUFFER_HEXDUMP(TAG, data, data_len, ESP_LOG_INFO);
    return true;
}
```
This way we can see what exactly is received

tore-espressif, 2024-08-13:
@antunesls I'm sorry but I'm afraid I can't help you any further. It seems that the CDC-ACM driver gives you the data your device has sent. 

If there is any other data post-processing (such as reaction on timeout, or concatenating messages as you do in your python code) you will have to implement this yourself

tore-espressif, 2024-08-14:
@antunesls We closed the issue because the fix has been merged. 
If you have any new findings or you require additional help, feel free to reopen

closed 2024-08-14 by tore-espressif

antunesls, 2024-08-14:
Hi @tore-espressif , thanks for all support.

I saw that this.l merge was done in the ESP-USB repository, can I use it now or should I wait for a new  Esp-Idf release ?