espressif/esp-idf #14346: esp-wifi-mesh internal_communication example console say `fail to find a network`

m986883511, 2024-08-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I test internal_communication project on my esp32-s3 and esp32-c3.

but i found it can not connect. monitor say `fail to find a network`.

i watch the readme.md, it say `configure the mesh network channel, router SSID, router password and mesh softAP settings.` I dont known to moify, so I use default setting, is ok??

follow is my log, can you help me test the example.

esp32s3 serial log
```shell
I (27) boot: ESP-IDF v5.2.1 2nd stage bootloader
I (27) boot: compile time Aug 10 2024 15:30:33
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 2MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=2b874h (178292) map
I (131) esp_image: segment 1: paddr=0003b89c vaddr=3fc96500 size=04580h ( 17792) load
I (135) esp_image: segment 2: paddr=0003fe24 vaddr=40374000 size=001f4h (   500) load
I (137) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=9d1c8h (643528) map
I (261) esp_image: segment 4: paddr=000dd1f0 vaddr=403741f4 size=12204h ( 74244) load
I (285) boot: Loaded app from partition at offset 0x10000
I (285) boot: Disabling RNG early entropy source...
I (297) cpu_start: Multicore app
I (306) cpu_start: Pro cpu start user code
I (306) cpu_start: cpu freq: 160000000 Hz
I (306) cpu_start: Application information:
I (309) cpu_start: Project name:     internal_communication
I (316) cpu_start: App version:      1
I (320) cpu_start: Compile time:     Aug 10 2024 15:29:45
I (326) cpu_start: ELF file SHA256:  90c99047f...
I (331) cpu_start: ESP-IDF:          v5.2.1
I (336) cpu_start: Min chip rev:     v0.0
I (341) cpu_start: Max chip rev:     v0.99 
I (346) cpu_start: Chip rev:         v0.1
I (351) heap_init: Initializing. RAM available for dynamic allocation:
I (358) heap_init: At 3FCA0090 len 00049680 (293 KiB): RAM
I (364) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (370) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (376) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (383) spi_flash: detected chip: gd
I (387) spi_flash: flash io: dio
W (391) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (404) sleep: Configure to isolate all GPIO pins in sleep state
I (411) sleep: Enable automatic switching of GPIO sleep configuration
I (418) main_task: Started on CPU0
I (428) main_task: Calling app_main()
I (458) pp: pp rom version: e7ae62f
I (468) net80211: net80211 rom version: e7ae62f
I (478) wifi:wifi driver task: 3fcaa554, prio:23, stack:6656, core=0
I (488) wifi:wifi firmware version: a9f5b59
I (488) wifi:wifi certification version: v7.0
I (488) wifi:config NVS flash: enabled
I (488) wifi:config nano formating: disabled
[... 139 lines cut ...]

I (5071) mesh: <MESH_NWK_LOOK_FOR_NETWORK>need_scan:0x3, need_scan_router:0x1, look_for_nwk_count:2
I (6771) mesh: [FIND][ch:0]AP:0, otherID:0, MAP:0, idle:0, candidate:0, root:0[00:00:00:00:00:00]<scan router>
I (6771) mesh: [FIND:2]fail to find a network, channel:0, cfg<channel:0, router:ROUTER_SSID, 00:00:00:00:00:00>

I (6781) mesh: <MESH_NWK_LOOK_FOR_NETWORK>need_scan:0x3, need_scan_router:0x0, look_for_nwk_count:3
I (11001) mesh: [FIND][ch:0]AP:13, otherID:0, MAP:0, idle:0, candidate:0, root:0[00:00:00:00:00:00]
I (11001) mesh: [FIND:3]fail to find a network, channel:0, cfg<channel:0, router:ROUTER_SSID, 00:00:00:00:00:00>

I (11001) mesh: <MESH_NWK_LOOK_FOR_NETWORK>need_scan:0x3, need_scan_router:0x1, look_for_nwk_count:4
I (12711) mesh: [FIND][ch:0]AP:0, otherID:0, MAP:0, idle:0, candidate:0, root:0[00:00:00:00:00:00]<scan router>
I (12711) mesh: [FIND:4]fail to find a network, channel:0, cfg<channel:0, router:ROUTER_SSID, 00:00:00:00:00:00>

I (12711) mesh: <MESH_NWK_LOOK_FOR_NETWORK>need_scan:0x3, need_scan_router:0x0, look_for_nwk_count:5
I (16931) mesh: [FIND][ch:0]AP:16, otherID:0, MAP:0, idle:0, candidate:0, root:0[00:00:00:00:00:00]
I (16931) mesh: [FIND:5]fail to find a network, channel:0, cfg<channel:0, router:ROUTER_SSID, 00:00:00:00:00:00>

I (16941) mesh: <MESH_NWK_LOOK_FOR_NETWORK>need_scan:0x3, need_scan_router:0x1, look_for_nwk_count:6
I (18641) mesh: [FIND][ch:0]AP:0, otherID:0, MAP:0, idle:0, candidate:0, root:0[00:00:00:00:00:00]<scan router>
I (18641) mesh: [FIND:6]fail to find a network, channel:0, cfg<channel:0, router:ROUTER_SSID, 00:00:00:00:00:00>

I (18651) mesh: <MESH_NWK_LOOK_FOR_NETWORK>need_scan:0x3, need_scan_router:0x0, look_for_nwk_count:7
I (22861) mesh: [FIND][ch:0]AP:13, otherID:0, MAP:0, idle:0, candidate:0, root:0[00:00:00:00:00:00]
I (22861) mesh: [FIND:7]fail to find a network, channel:0, cfg<channel:0, router:ROUTER_SSID, 00:00:00:00:00:00>
```

m986883511, 2024-08-17:
version is C:\Espressif\frameworks\esp-idf-v5.2.1

TroyMitchell911, 2024-08-18:
Hi, Is your network a mobile hotspot?  If not, please try to connect another network or try to change authmode. 

Best regard.

TroyMitchell911, 2024-08-18:
> Hi, Is your network a mobile hotspot? If not, please try to connect another network or try to change authmode.
> 
> Best regard.

If you can not resolve this problem and waiting long time, you can get some help here: https://esp32.com/viewforum.php?f=13

zhangyanjiaoesp, 2024-08-19:
@m986883511 
please set the ssid and password of a router that can be successfully connected.
[image]

m986883511, 2024-08-19:
> @m986883511 please set the ssid and password of a router that can be successfully connected. [...]

I set same config but the error same as before
[image]

m986883511, 2024-08-19:
> Hi, Is your network a mobile hotspot? If not, please try to connect another network or try to change authmode.
> 
> Best regard.

no, it connect to anotner esp32 device.

zhangyanjiaoesp, 2024-08-20:
> @m986883511 please set the ssid and password of a router that can be successfully connected. [...]

@m986883511 This is just an example, do you have a router whose ssid is test and password is 12345678 in your environment？

m986883511, 2024-08-20:
> @m986883511 This is just an example, do you have a router whose ssid is test and password is 12345678 in your environment？

https://docs.espressif.com/projects/esp-mdf/en/latest/api-guides/general-notes.html#network-configuration 
this guide say it doesnot need a external router. I want to use [Automatic root node election via the physical layer protocol] this way.

zhangyanjiaoesp, 2024-08-20:
> https://docs.espressif.com/projects/esp-mdf/en/latest/api-guides/general-notes.html#network-configuration this guide say it doesnot need a external router. I want to use [Automatic root node election via the physical layer protocol] this way.

Is that what you're saying?

> As soon as a root node is established, ESP-WIFI-MESH network configuration process becomes independent of routers.

What this means is that after root has been generated, the creation of the mesh network and the router are independent. However, when all nodes initially elect the root node, they need to elect the root node according to the rssi relative to the router. So you need to set a valid ssid and password.

m986883511, 2024-09-09:
thank you for you help, you are right.

closed 2024-09-09 by m986883511

iwonako, 2024-11-20:
> What this means is that after root has been generated, the creation of the mesh network and the router are independent. However, when all nodes initially elect the root node, they need to elect the root node according to the rssi relative to the router. So you need to set a valid ssid and password.

can esp32 iteslf be a router for mesh wifi initial ? @zhangyanjiaoesp