espressif/esp-idf #14348: I2C transaction unexpected nack detected when trying ssd1306 sample code in 5.3

blfuentes, 2024-08-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

The sample code should be able to display content in the ssd1306 display

### What is the actual behavior?

It fails throwing this exceptions:
```
E (347) i2c.master: I2C transaction unexpected nack detected
E (357) i2c.master: s_i2c_synchronous_transaction(870): I2C transaction failed
E (357) i2c.master: i2c_master_transmit(1072): I2C transaction failed
E (367) lcd_panel.io.i2c: panel_io_i2c_tx_buffer(176): i2c transaction failed
E (377) lcd_panel.ssd1306: panel_ssd1306_init(147): io tx param SSD1306_CMD_SET_MULTIPLEX failed
```

### Steps to reproduce.

1. Use the project wizzard to create a i2c_oled_example
2. Set menuconfig to set config values(height, sda an scl pins)
3. Build and flash
```
/*
 * SPDX-FileCopyrightText: 2021-2024 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: CC0-1.0
 */

#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_lcd_panel_io.h"
#include "esp_lcd_panel_ops.h"
#include "esp_err.h"
#include "esp_log.h"
#include "driver/i2c_master.h"
#include "esp_lvgl_port.h"
#include "lvgl.h"

#if CONFIG_EXAMPLE_LCD_CONTROLLER_SH1107
#include "esp_lcd_sh1107.h"
#else
#include "esp_lcd_panel_vendor.h"
#endif

static const char *TAG = "example";

#define I2C_BUS_PORT  0

[... 180 lines cut ...]
I (317) gpio: GPIO[18]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0
I (327) gpio: GPIO[17]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 
I (337) example: Install panel IO
I (337) example: Install SSD1306 panel driver
E (347) i2c.master: I2C transaction unexpected nack detected
E (357) i2c.master: s_i2c_synchronous_transaction(870): I2C transaction failed
E (357) i2c.master: i2c_master_transmit(1072): I2C transaction failed
E (367) lcd_panel.io.i2c: panel_io_i2c_tx_buffer(176): i2c transaction failed
E (377) lcd_panel.ssd1306: panel_ssd1306_init(147): io tx param SSD1306_CMD_SET_MULTIPLEX failed
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x420094ba
0x420094ba: app_main at D:/dev/test-i2c-esp32-s3/main/i2c_oled_example_main.c:102 (discriminator 1)

file: "./main/i2c_oled_example_main.c" line 102
func: app_main
expression: esp_lcd_panel_init(panel_handle)

abort() was called at PC 0x4037a213 on core 0
0x4037a213: _esp_error_check_failed at C:/Espressif/frameworks/esp-idf-v5.3/components/esp_system/esp_err.c:49
```

### More Information.

Board: ESP32-S3
SDA pin: 18
SCL pin: 17

blfuentes, 2024-08-11:
hardware issue...

closed 2024-08-11 by blfuentes

paleknight894, 2024-10-01:
> hardware issue...

@blfuentes  same question ,how did you resolve that.

laoruojin, 2025-02-27:
解决了吗

paleknight894, 2025-02-28:
解决了

paleknight894, 2025-02-28:
供电不够 需要给显示器额外供电
stm32的话不用

laoruojin, 2025-02-28:
> 供电不够 需要给显示器额外供电stm32的话不用陈妙明 ***@***.***> 于 2025年2月28日周五 下午6:20写道：
> […](#)
我这边也在做对应的开发，遇到的问题是esp32s3lightsleep模式唤醒后，ssd1306对应的i2c接口有概率会无ack响应，把lightsleep去掉，就不会有这种情况，是否有可能需要重新初始化呢，log如下：

E (56153) i2c.master: I2C hardware NACK detected
E (56153) i2c.master: I2C transaction unexpected nack detected
E (56163) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (56173) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (56183) lcd_panel.io.i2c: panel_io_i2c_tx_buffer(179): i2c transaction failed
E (56193) lcd_panel.ssd1306: panel_ssd1306_draw_bitmap(202): io tx color failed
I (56223) Application: Free internal: 145179 minimal internal: 130927
I (56253) Application: Free internal: 149967 minimal internal: 130927
I (56283) Application: Free internal: 149939 minimal internal: 130927
I (56313) Application: Free internal: 149995 minimal internal: 130927
I (56343) Application: Free internal: 152851 minimal internal: 130927
I (56373) Application: Free internal: 152823 minimal internal: 130927
E (61283) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (61283) task_wdt:  - IDLE1 (CPU 1)
E (61283) task_wdt: Tasks currently running:
E (61283) task_wdt: CPU 0: IDLE0
E (61283) task_wdt: CPU 1: taskLVGL
E (61283) task_wdt: Print CPU 1 backtrace

我找了很多有关的接口，都没有成功解决，看看大佬有没有不一样的见解

blfuentes, 2025-02-28:
> > hardware issue...
> 
> [@blfuentes](https://github.com/blfuentes) same question ,how did you resolve that.

it was just a matter of bad soldering. no software related.