espressif/esp-idf #14349: Macros in esp_check.h rely on specific variable "ret" being declared

ahemberger, 2024-08-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'd like to use some off the macros found in esp_check.h (specifically ESP_RETURN_ON_FALSE), but it seems this macro (and others in the file) rely on a variable of type esp_err_t called "ret" to be defined before the macro is called. Personally I prefer to use a different error code variable name (I like "error_status"), but this causes compilation errors. Am I missing something obvious here in how I should be using this header file? Or would it be possible to remove the variable name dependency to allow more flexibility in its use?

sudeep-mohanty, 2024-08-12:
Hi @ahemberger,
There shouldn't be a need to declare a variable `ret` when you use `ESP_RETURN_ON_FALSE` macro. You could use it something like this -

```c
include "esp_check.h"

int return_check(void)
{
    int a = 6;
    ESP_RETURN_ON_FALSE(a == 5, ESP_ERR_INVALID_ARG, "TAG", "a is not 5\n");
    return 0;
}
```

ahemberger, 2024-08-12:
Hi Sudeep, you are right, I misspoke - that particular macro does not seem to depend on ret. What tripped me up was a few lines later, using ESP_GOTO_ON_ERROR:
https://github.com/espressif/esp-idf/blob/v5.3/components/esp_common/include/esp_check.h

```
#define ESP_GOTO_ON_ERROR(x, goto_tag, log_tag, format, ...) do {                               \
        esp_err_t err_rc_ = (x);                                                                \
        if (unlikely(err_rc_ != ESP_OK)) {                                                      \
            ESP_LOGE(log_tag, "%s(%d): " format, __FUNCTION__, __LINE__, ##__VA_ARGS__);        \
            ret = err_rc_;                                                                      \
            goto goto_tag;                                                                      \
        }                                                                                       \
    } while(0)
```
This threw a compilation error when I tried doing it, and seems to depend on ret being defined before the function is called? When I simply do:

```
#include "esp_check.h"

esp_err_t error_status;
ESP_GOTO_ON_ERROR(  mcpwm_new_timer(&config, &handle), error_tag, TAG, "Create MCPWM timer failed");
```

I see this compilation error:
```

/esp-idf/esp-idf-v5.3/components/esp_common/include/esp_check.h:352:13: error: 'ret' undeclared (first use in this function)
  352 |             ret = err_rc_;                                                                      \
      |             ^~~
/sb_motor.c:109:5: note: in expansion of macro 'ESP_GOTO_ON_ERROR'
  109 |     ESP_GOTO_ON_ERROR(  mcpwm_new_timer(&config, &handle),
      |     ^~~~~~~~~~~~~~~~~
```

When I change the variable "`esp_err_t error_status;`" to "`esp_err_t ret;`" the code compiles fine.

sudeep-mohanty, 2024-08-13:
Hi @ahemberger, Indeed there is a dependency on defining a variable `ret` while using the macros `ESP_GOTO_ON_ERROR` and a few others such as `ESP_GOTO_ON_ERROR_ISR`, `ESP_GOTO_ON_FALSE` and `ESP_GOTO_ON_FALSE_ISR`. I agree that it would be better to avoid forcing users to declare a variable with fixed naming conventions. We will take a look at it. In the meantime, a quick fix would be for you to update the `esp_check.h` macro `ESP_GOTO_ON_ERROR` to use `x = err_rc_` instead of `ret = err_rc`.

sudeep-mohanty, 2024-08-14:
HI @ahemberger, After further discussions, it is concluded that we cannot update the `ESP_GOTO_ON_ERROR` macro right now since it will introduce a breaking change. You could implement a similar macro to this one where it can take any variable or a variable with the name you want.

Closing the issue for now. Please feel free to reopen the ticket if you need more assistance. Thanks!

closed 2024-08-14 by sudeep-mohanty

KaeLL, 2024-08-14:
> we cannot update the ESP_GOTO_ON_ERROR macro **right now**

by which you mean it'll go to IDF 6 backlog or it's just a overly polite way of saying it won't be changed?

sudeep-mohanty, 2024-08-14:
That's right @KaeLL, we cannot introduce a breaking change until the next major release, i.e., IDF 6.0. If the change is highly requested then there would be a higher probability of it being updated.

igrr, 2024-08-14:
Considering the number of projects which [use ESP_GOTO_ON_ERROR](https://github.com/search?q=ESP_GOTO_ON_ERROR&type=code) already, I think it wouldn't be reasonable to break this in IDF 6.0, either. Perhaps we could introduce a new macro, such as ESP_GOTO_ON_ERROR_VAR, with an additional "error variable" argument, instead. This wouldn't be a breaking change.

KaeLL, 2024-08-14:
That's where I was getting at. I though breaking changes were the point of major versions, especially small ones like this, despite the potential impact. IDF 5 did it with C IO conversion specifiers, and I personally welcomed it, just as I would this one and others for, if nothing else, the sake of taste, expressed by the hygiene and consistency with the other macros. If the alternative is adding redundant rectified versions, might as well stick to the existing ones. Just my 2¢.

forfrossen, 2024-11-02:
A little late to the party, but I'd like to add my two cents as well:

When using a framework like esp-idf, you have to adapt your coding style. You wouldn't create an issue with Meta's React team and request a feature to make it more similar to the style of Google's Angular just because you're used to how it works there?

As an alternative, you could always just create your own macros that are independent of ret being defined or change the variable name to what you like best (basically as a wrapper).