espressif/esp-idf #14351: Example of using ChaCha20-Poly1305

butaikis, 2024-08-12:
### Is your feature request related to a problem?

No

### Describe the solution you'd like.

Hello. Is it possible to get an example of using ChaCha20-Poly1305 encryption/decryption?

AdityaHPatwardhan, 2024-08-12:
Hi @butaikis,

esp-idf itself does not offer any method ChaCha20-Poly1305 encryption/decryption hence we have not added example for the same.
You can use the support from third party library https://github.com/Mbed-TLS/mbedtls
In Mbedtls I see the same example can be located at https://github.com/Mbed-TLS/mbedtls/blob/development/programs/cipher/cipher_aead_demo.c
 
Do you think an example for the same in esp-idf might be more helpful? and why?

butaikis, 2024-08-12:
@AdityaHPatwardhan It's weird, I just found mentions of her here. 
https://docs.espressif.com/projects/esp-idf/en/v4.2/esp32/api-reference/kconfig.html#config-mbedtls-chachapoly-c

In general, I need this in order to integrate into HomeKit

AdityaHPatwardhan, 2024-08-12:
Yes, the link that you have shared is a configuration option for the same third party library `https://github.com/Mbed-TLS/mbedtls`.
This configuration option only enables support of the respective cipher in mbedTLS.
Do you think the mbedtls example should be okay?

butaikis, 2024-08-12:
@AdityaHPatwardhan If it is not difficult, please make an example for esp-idf 
Thank you in advance 😊

AdityaHPatwardhan, 2024-08-12:
Hi @butaikis,

There is another third party library `libsodium` which provides the same support
Here is the link to an example test application that does the same
1) Link to example test application that `https://github.com/espressif/idf-extra-components/blob/master/libsodium/test/test_sodium.c`
2) here is another link https://github.com/espressif/esp-homekit-sdk/blob/43bab6f68569e26c738f0273960200c809ec0ec9/components/homekit/esp_hap_core/src/esp_hap_pair_verify.c#L30 that shows the usage of the same APIs in `esp-homekit-sdk`, you may refer same APIs.

I hope this should resolve your issue.
Thanks

butaikis, 2024-08-13:
@AdityaHPatwardhan Thanks, let's try it)

closed 2024-08-13 by butaikis