espressif/esp-idf #14353: esp32-c6 WiFi-6 TWT state

loraadh, 2024-08-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,
I need  to measure the current consumption of esp32-c6-DevKitC_1 V1.2 when it is in WiFi-6 TWT state (target wakeup time). I am not able to find any FW or commands to put the device in TWT. Would you please give support? Thanks.
BR

QingzhaoYin, 2025-03-04:
@loraadh  https://github.com/espressif/esp-idf/tree/master/examples/wifi/itwt  this example can help you

loraadh, 2025-03-04:
Hello @QingzhaoYin 
thanks for sharing the link. it helps indeed.
I've noticed that only individual TWT is supported. Do you have any idea why esp32-c6 does not support the TWT broadcast mode?
BR

loraadh, 2025-03-04:
interesting information @QingzhaoYin 
thanks
but what you mean by certified routers? is there a BTWT-special certification other than WiFi-6 which includes TWT?

QingzhaoYin, 2025-03-04:
@loraadh Currently, most commercially available routers do not support BTWT. As far as I know, only certain routers used for certification purposes support BTWT, and there is a dedicated test certification plan specifically for BTWT.

loraadh, 2025-03-04:
i've noticed that esp32-c6 supports itwt in station mode only. is that right? if yes, why not supporting it in AP mode as well?

QingzhaoYin, 2025-03-04:
@loraadh Currently, the ESP32C6 softAP does not support the TWT feature. After all, it’s just a softAP, which is quite different from a real router. Moreover, ESP series chips are mainly used in IoT scenarios, where low power consumption is often required. If softAP were to support TWT, it would need to stay awake constantly to manage other devices, which I believe is unnecessary. (In such cases, using an AP that supports TWT would be the most suitable option.)

loraadh, 2025-03-04:
thanks a lot

closed 2025-03-05 by Alvin1Zhang