espressif/esp-idf #14354: `merge_bin` needs to validate the input file sizes (ESPTOOL-904)

vshymanskyy, 2024-08-12:
### Is your feature request related to a problem?

If the file size exceeds the partition size, the `esptool` still finishes with success, presumably producing a broken firmware.

igrr, 2024-08-12:
@vshymanskyy Could you please clarify, what is the case when `esptool.py merge_bin` result would exceed the partition size?

We already validate that the bootloader and the app fit into the partition table, so `idf.py build` should fail in the first place. Could you please give a more detailed example illustrating the issue? Thanks.

vshymanskyy, 2024-08-12:
Yes, for example if i build a single image, and then to merge it with different partition tables (for esp modules with different flash size).
Of course, the image headers need to be adjusted, but merge_bin already does this.

igrr, 2024-08-12:
I see, thanks for explaining. In this case I'm afraid what @peterdragun mentioned is still the case:

> However esptool is not trying to understand the data during merging and does not have any information regarding the partition sizes, so I don't think this would be possible.

Basically, esptool isn't aware of the partition tables, which are an ESP-IDF-specific feature, and it's unlikely that we would add new features into esptool which would be too coupled to ESP-IDF.

The only solution I can recommend now is to run `idf.py merge-bin` (https://github.com/espressif/esp-idf/pull/13546) command for each flash size. This will result in the app being rebuilt, but if you use ccache this will happen pretty quickly. This way, partition size issues will be detected.

vshymanskyy, 2024-08-12:
I'm implementing the check in an outer infrastructure already, i just wanted to bring it to you.

closed 2024-10-09 by dobairoland