espressif/esp-idf #14355: [bug] Nimble init - ESP_ERR_NO_MEM - SDK 5.2.2 commit 876eaf80829d2501f94145066061151ae0a45a6a (HEAD -> release/v5.2, origin/release/v5.2)

filzek, 2024-08-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Nimble stop to work in this commit, error:
ret = nimble_port_init();

Return = ESP_ERR_NO_MEM 

FreeHeapSize              => 3065176 bytes
Internal Heap Size        => 143204 bytes

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40116440  PS      : 0x00060230  A0      : 0x80116761  A1      : 0x3ffd2630
0x40116440: ble_npl_os_started at C:/Espressif/frameworks/esp-idf-v5.2/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:143

A2      : 0x00000000  A3      : 0x3ffd2640  A4      : 0x00000010  A5      : 0x3ffc0b5c
A6      : 0x38695a4a  A7      : 0x6e683575  A8      : 0x00000000  A9      : 0x3ffd2570
A10     : 0x0000000a  A11     : 0x3f400a08  A12     : 0x00000000  A13     : 0x3f400b8b
A14     : 0x00000007  A15     : 0x3ffb6a04  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400911f4  LEND    : 0x40091210  LCOUNT  : 0x00000000
0x400911f4: memcpy at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:162
0x40091210: memcpy at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:197

Backtrace: 0x4011643d:0x3ffd2630 0x4011675e:0x3ffd2650 0x4011721d:0x3ffd2670 0x400fa588:0x3ffd2690
0x4011643d: ble_npl_os_started at C:/Espressif/frameworks/esp-idf-v5.2/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:143
0x4011675e: ble_hs_lock at C:/Espressif/frameworks/esp-idf-v5.2/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:218
0x4011721d: ble_att_set_preferred_mtu at C:/Espressif/frameworks/esp-idf-v5.2/components/bt/host/nimble/nimble/nimble/host/src/ble_att.c:551

rahult-github, 2024-08-12:
Hi @filzek , 

Thanks for reporting. But can you please give more information. 

1. What is the chip being used ?
2. What is the procedure being done when this issue is observed , that can be tried at our end to reproduce the behaviour. 
3. Please share your sdkconfig too. 
4. Are there any other changes done or nimble sample examples show the same behaviour too ?

filzek, 2024-08-12:
What is the chip being used ?
Esp32 REV 1.0 and Rev3 .0

What is the procedure being done when this issue is observed , that can be tried at our end to reproduce the behaviour.
Just INIT the Nimble.

Please share your sdkconfig too.
Attached
[sdkconfig-nimble-problem.txt](https://github.com/user-attachments/files/16588311/sdkconfig-nimble-problem.txt)

Are there any other changes done or nimble sample examples show the same behaviour too ?
We did not change anything at all, just update to the last commig and this is nor working anymore, unique issue prior this was with the latest issue published in the scan.

rahult-github, 2024-08-13:
Hi @filzek ,

I took your sdkconfig and used it to compile the blecent application. In blecent application, there is a option to do nimble init / deinit in a loop, which i wanted to enable and check .As seen, this works even with your sdkconfig : 

I (31) boot: ESP-IDF v5.2.2-492-g**876eaf8082**-dirty 2nd stage bootloader
I (31) boot: compile time Aug 13 2024 15:04:55
I (33) boot: Multicore bootloader
I (37) boot: chip revision: v1.0
I (41) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1658ch ( 91532) map
I (137) esp_image: segment 1: paddr=000265b4 vaddr=3ffbdb60 size=04a64h ( 19044) load
I (144) esp_image: segment 2: paddr=0002b020 vaddr=40080000 size=04ff8h ( 20472) load
I (153) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=51974h (334196) map
I (267) esp_image: segment 4: paddr=0008199c vaddr=40084ff8 size=11ca4h ( 72868) load
I (308) boot: Loaded app from partition at offset 0x10000
I (308) boot: Disabling RNG early entropy source...
I (319) cpu_start: Multicore app
I (328) cpu_start: Pro cpu start user code
I (328) cpu_start: cpu freq: 160000000 Hz
I (328) cpu_start: Application information:
I (331) cpu_start: Project name:     blecent
I (336) cpu_start: App version:      v5.2.2-492-g876eaf8082-dirty
I (343) cpu_start: Compile time:     Aug 13 2024 15:04:49
I (349) cpu_start: ELF file SHA256:  88f387c99...
I (355) cpu_start: ESP-IDF:          v5.2.2-492-g876eaf8082-dirty
I (361) cpu_start: Min chip rev:     v0.0
I (366) cpu_start: Max chip rev:     v3.99 
I (371) cpu_start: Chip rev:         v1.0
I (376) heap_init: Initializing. RAM available for dynamic allocation:
I (383) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (389) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (395) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (401) heap_init: At 3FFC5648 len 0001A9B8 (106 KiB): DRAM
I (407) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (414) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (420) heap_init: At 40096C9C len 00009364 (36 KiB): IRAM
I (428) spi_flash: detected chip: generic
I (431) spi_flash: flash io: dio
W (435) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (449) coexist: coex firmware version: 27d8387
I (454) main_task: Started on CPU0
I (464) main_task: Calling app_main()
I (474) BTDM_INIT: BT controller compile version [2b3ecee]
I (474) BTDM_INIT: Bluetooth MAC: 24:6f:28:80:2c:2a
I (484) phy_init: phy_version 4830,54550f7,Jun 20 2024,14:22:08
I (794) NimBLE_BLE_CENT: BLE Host Task Started
I (10794) NimBLE_BLE_CENT: Deinit host
I (10794) NimBLE: GAP procedure initiated: stop advertising.

I (20794) NimBLE_BLE_CENT: Init host
I (20794) BTDM_INIT: BT controller compile version [2b3ecee]
I (20794) BTDM_INIT: Bluetooth MAC: 24:6f:28:80:2c:2a
I (21034) NimBLE_BLE_CENT: BLE Host Task Started
I (21034) NimBLE_BLE_CENT: Waiting for 1 second
I (31034) NimBLE_BLE_CENT: Deinit host
I (31034) NimBLE: GAP procedure initiated: stop advertising.

I (41034) NimBLE_BLE_CENT: Init host
I (41034) BTDM_INIT: BT controller compile version [2b3ecee]
I (41034) BTDM_INIT: Bluetooth MAC: 24:6f:28:80:2c:2a
I (41274) NimBLE_BLE_CENT: BLE Host Task Started
I (41274) NimBLE_BLE_CENT: Waiting for 1 second
I (51274) NimBLE_BLE_CENT: Deinit host
I (51274) NimBLE: GAP procedure initiated: stop advertising.

So, one question for you, when you do git pull, are you also ensuring that nimble submodule is also fetched ?  >
Can you try the blecent application at your end with same setup to confirm ?

filzek, 2024-08-13:
last commig has worked great.
commit 2cd9da52fe5704328c07b90f9b26748b8833e1a6 (HEAD -> release/v5.2, origin/release/v5.2)
Merge: 726d796683 1c0a4dbfa3
Author: Island <island@espressif.com>
Date:   Mon Aug 12 20:15:05 2024 +0800

    Merge branch 'feat/print_the_ble_mac_when_init_5.2' into 'release/v5.2'

    feat(ble): Print out the BT mac when the controller is initialized

    See merge request espressif/esp-idf!32731

closed 2024-08-13 by filzek