espressif/esp-idf #14360: Destructors of C++11 thread_local object are not called after task returned

andylinpersonal, 2024-08-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2, master

### Espressif SoC revision.

ESP32-C3, ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32-c6 devkit-c

### Power Supply used.

USB

### What is the expected behavior?

TLS finalizers of C++ `thread_local` variables seems to be missing from the current toolchain.

MWE on other platform (linux,x64): https://godbolt.org/z/vfWWdh7qe

```
TestTLS
foo
~TestTLS
```

The TLS wrapper function registered the destructor of `thread_local` object to `__cxa_thread_atexit` and get called after `main` returned.
But there are no such ABI function found in ESP-IDF.

### What is the actual behavior?

Destructor of thread local variables are not called after the task died.

Logging messages excerpted from my C6
```
I (80) sleep: Configure to isolate all GPIO pins in sleep state
I (81) sleep: Enable automatic switching of GPIO sleep configuration
I (81) coexist: coex firmware version: 8da3f50af
I (82) coexist: coexist rom version 5b8dcfa
I (82) main_task: Started on CPU0
I (82) main_task: Calling app_main()
TestTLS
test
I (82) main_task: Returned from app_main()
```

### Steps to reproduce.

1. Build the code from aforementioned https://godbolt.org/z/vfWWdh7qe.
2. See the output and disassembled code.

### More Information.

Crosstool config files here:
RISC-V:
https://github.com/espressif/crosstool-NG/blob/esp-13.2.0_20240530/samples/riscv32-esp-elf/crosstool.config
ESP32-S3:
https://github.com/espressif/crosstool-NG/blob/esp-13.2.0_20240530/samples/xtensa-esp32s3-elf/crosstool.config

0xjakob, 2024-08-14:
@andylinpersonal Thanks for pointing this out! We'll take a look at this.

0xjakob, 2024-08-20:
We need to find and discuss some implementation (of `__cxa_thread_atexit_impl`, etc.), and it appears to me that we would also need to change the toolchain, so solving this problem might take a while. But we'll keep you updated here.

Lapshin, 2024-08-23:
@andylinpersonal , for now, you can workaround it with these steps:

- add `-fuse-cxa-atexit` into your CMakeLists.txt:
   `target_compile_options(${COMPONENT_LIB} PRIVATE -fuse-cxa-atexit)`
- add somewhere in your code a global variable:
  ```
  extern void *__dso_handle __attribute__ ((__visibility__ ("hidden")));
  void *__dso_handle = 0;
  ```
  
After these steps I have the output:

```
I (285) main_task: Calling app_main()
TestTLS
foo
I (285) main_task: Returned from app_main()
~TestTLS

Done
```

I also tested it with multiple tasks created with `xTaskCreate(test, "test", 2048, NULL, 10, NULL);` and seems all works fine.

@andylinpersonal , could you please test this fix and leave feedback?

andylinpersonal, 2024-08-23:
[image]
Ok, it works now!
Thanks for your suggestion @Lapshin!

closed 2024-10-28 by espressif-bot