espressif/esp-idf #14361: transport (LoadStoreAlignment). Exception was unhandled. SDK 5.2.2 [BUG]

filzek, 2024-08-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Guru Meditation Error: Core  1 panic'ed (LoadStoreAlignment). Exception was unhandled.

Core  1 register dump:
PC      : 0x401c6ae1  PS      : 0x00060030  A0      : 0x8010ba60  A1      : 0x3f94b290
0x401c6ae1: esp_transport_close at C:/Espressif/frameworks/esp-idf-v5.2/components/tcp_transport/transport.c:171 (discriminator 1)

A2      : 0x00000000  A3      : 0xffffffff  A4      : 0x3f9882e4  A5      : 0x0000010b
A6      : 0x3fff3e18  A7      : 0x3fff3644  A8      : 0x80097935  A9      : 0x3f94b270
A10     : 0x40095a96  A11     : 0x00000000  A12     : 0x00009038  A13     : 0x00000004
0x40095a96: _frxt_int_enter at C:/Espressif/frameworks/esp-idf-v5.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:140

A14     : 0x3ffc2224  A15     : 0x3ffc2228  SAR     : 0x00000020  EXCCAUSE: 0x00000009
EXCVADDR: 0x40095aae  LBEG    : 0x40090fae  LEND    : 0x40090fb9  LCOUNT  : 0x00000000
0x40095aae: _frxt_int_enter at C:/Espressif/frameworks/esp-idf-v5.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:148
0x40090fae: memset at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:150
0x40090fb9: memset at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:160

Backtrace: 0x401c6ade:0x3f94b290 |<-CORRUPTED
0x401c6ade: esp_transport_close at C:/Espressif/frameworks/esp-idf-v5.2/components/tcp_transport/transport.c:174

ELF file SHA256: 76b77a5fa

AxelLin, 2024-08-14:
@filzek 
Can you provide minimal code to reproduce this issue?

AxelLin, 2024-10-13:
@filzek
Just to confirm if you still hit this issue with v5.2.3?
Also could you share more detail about how to reproduce this issue?

closed 2025-01-17 by filzek