espressif/esp-idf #14363: ESP32s3使用通用定时器gptimer，问题1：初始化后，下降沿中断开启定时，定时中断关闭定时，运行一会后存在无法再启动定时；问题2:问题1：初始化后使用stop停止定时，更改定时时间后start启动定时，示波器看到有下降沿中断未定时

Fuzhiguo36, 2024-08-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2

### Espressif SoC revision.

ESP32-S3(ESP32-S3-WROOM-1)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

VSCODE ESPIDF

### Power Supply used.

USB

### What is the expected behavior?

ESP32-S3-WROOM-1

### What is the actual behavior?

使用需求：ESP32s3使用通用定时器gptimer，通过IO的下降触发中断检测过零信号，gptimer定时输出一个高电平。
问题1：初始化后，下降沿中断开启定时，定时中断关闭定时，运行一会后存在无法再启动定时；出现报错
E (399) gptimer: gptimer_stop(377): timer is not running
E (27500) gptimer: gptimer_start(358): timer is not enabled yet
E (54182) gptimer: gptimer_start(358): timer is not enabled yet

问题2:问题1：初始化后使用stop停止定时，更改定时时间后start启动定时，示波器看到有下降沿中断未定时输出
E (141385) gptimer: gptimer_start(358): timer is not enabled yet
E (260474) gptimer: gptimer_start(358): timer is not enabled yet
E (291013) gptimer: gptimer_start(358): timer is not enabled yet
E (387093) gptimer: gptimer_start(358): timer is not enabled yet
E (493264) gptimer: gptimer_start(358): timer is not enabled yet
E (516581) gptimer: gptimer_start(358): timer is not enabled yet
E (518403) gptimer: gptimer_start(358): timer is not enabled yet
E (553466) gptimer: gptimer_start(358): timer is not enabled yet

### Steps to reproduce.

通用部分代码：
过零检测初始化
void zreoio_init(uint8_t intr_trigger,uint8_t intr_ionum)
{
	uint8_t intr_trig,intr_ion;
	intr_trig = intr_trigger;
	intr_ion = intr_ionum;
	// 设置IO中断下降沿触发
	gpio_config_t gpio_conf = {0};
	gpio_conf.intr_type = intr_trig; // 设置1上升沿中断GPIO_PIN_INTR_POSEDGE、2下降沿GPIO_INTR_NEGEDGE
	// gpio_conf.pin_bit_mask = GPIO_INPUT_PIN_SEL;//输入指定引脚
	gpio_conf.pin_bit_mask = intr_ion; // 设置过零检测引脚READ_ZERO=17
	gpio_conf.mode = GPIO_MODE_INPUT;	// 设置为输入模式
	gpio_conf.pull_down_en = 0,			// 禁止下拉
	gpio_conf.pull_up_en = 0;		// 禁止上拉
	gpio_config(&gpio_conf);
	gpio_set_intr_type(intr_ion, intr_trig);
	gpio_install_isr_service(3);//ESP_INTR_FLAG_LEVEL6\3
	gpio_isr_handler_add(intr_ion, gpio_isr_handler, (void *)intr_ion);
}

定时器初始化
void Ptimer_init()
{
    gptimer_config_t config = {
        .clk_src = GPTIMER_CLK_SRC_DEFAULT,
        .direction = GPTIMER_COUNT_DOWN, // DOWN,
        .resolution_hz = 1000000,
        .intr_priority = 2,           //设置中断优先级
    };
    gptimer_new_timer(&config, &gptimer_handle);

    gptimer_alarm_config_t alarm_config = {
        .reload_count = 8000,               // counter will reload with 0 on alarm event
        .alarm_count = 0,                   // period = 1ms @resolution 1MHz
        .flags.auto_reload_on_alarm = true, // enable auto-reload
    };
    gptimer_set_alarm_action(gptimer_handle, &alarm_config);

    gptimer_event_callbacks_t cbs = {
        .on_alarm = example_timer_on_alarm_cb, // register user callback
    };

    ESP_ERROR_CHECK(gptimer_register_event_callbacks(gptimer_handle, &cbs, NULL));
    ESP_ERROR_CHECK(gptimer_enable(gptimer_handle));
    // ESP_ERROR_CHECK(gptimer_start(gptimer_handle));
}

过零检测中断
static void IRAM_ATTR gpio_isr_handler(void *arg)
{
	uint8_t gpio_num = (uint8_t)arg;
	if (gpio_num == READ_ZERO)
	{
			if (pid.Pid_Zero_trigger == 1)
			{
				// Ptimer_set_raw_count(9500);  // 重装定时器计数值，设置PID延时时间*US 
				GPtimer_start();
				pid.Pid_Zero_sta = 1;
				pid.C1ms++;
				zero_trigger = 1;
			}
	}
}

定时中断回调函数

static void IRAM_ATTR example_timer_on_alarm_cb(gptimer_handle_t timer, const gptimer_alarm_event_data_t *edata, void *user_ctx)
{
        pid.Pid_timerout_sta = 1;
        // gpio_set_level(LED2 , !gpio_get_level(LED2));
        PIDtime_Out();
        GPtimer_stop();         //定时器停止计数
        Ptimer_set_raw_count(8000);  // 重装定时器计数值，设置PID延时时间*US 
}

### Debug Logs.

```plain
E (399) gptimer: gptimer_stop(377): timer is not running
E (27500) gptimer: gptimer_start(358): timer is not enabled yet
E (54182) gptimer: gptimer_start(358): timer is not enabled yet

E (141385) gptimer: gptimer_start(358): timer is not enabled yet
E (260474) gptimer: gptimer_start(358): timer is not enabled yet
E (291013) gptimer: gptimer_start(358): timer is not enabled yet
E (387093) gptimer: gptimer_start(358): timer is not enabled yet
E (493264) gptimer: gptimer_start(358): timer is not enabled yet
E (516581) gptimer: gptimer_start(358): timer is not enabled yet
E (518403) gptimer: gptimer_start(358): timer is not enabled yet
E (553466) gptimer: gptimer_start(358): timer is not enabled yet
```

suda-morris, 2024-08-14:
@Fuzhiguo36  The timer interrupt and your GPIO interrupt are asynchronous, and you're not doing synchronization between these two events, which will result in gptimer_start being called again while the previous gptimer_stop is still in progress.

Fuzhiguo36, 2024-08-14:
> @Fuzhiguo36 The timer interrupt and your GPIO interrupt are asynchronous, and you're not doing synchronization between these two events, which will result in gptimer_start being called again while the previous gptimer_stop is still in progress.

您好：感谢您的回复，定时器我是在GPIO中断到的时候开启，计数器中断结束关闭，gptimer的API支持在中断内使用，我现在这样是函数使用有问题吗，请问有什么好的建议吗？谢谢您
Hello: Thank you for your reply. I turn on the timer when GPIO is interrupted, and turn off the counter when the interrupt ends. The API of gptimer supports use within the interrupt. Thank you

suda-morris, 2024-08-15:
Maybe the GPTimer interrupt and GPIO interrupt handler are installed on different CPU core, thus when CPU0 is Stopping the GPTimer, CPU1 is trying to starting the GPTimer, the state machine detects this invalid calling order. 

One suggest, you can use `esp_intr_dump` to dump the information of the interrupts, to see if they're assigned to different CPUs.

Fuzhiguo36, 2024-08-15:
> Maybe the GPTimer interrupt and GPIO interrupt handler are installed on different CPU core, thus when CPU0 is Stopping the GPTimer, CPU1 is trying to starting the GPTimer, the state machine detects [...]

您好，感谢您的帮助，我把IO中断和gptimer的任务绑定在CPU1，gptimer_start(358): timer is not enabled yet信息还在不停输出。esp_intr_dump打印信息如下，麻烦您帮忙看下，十分感谢。
CPU 0 interrupt status:
 Int  Level  Type   Status
  0     1    Level  Reserved
  1     1    Level  Reserved
  2     1    Level  Used: RTC_CORE
  3     1    Level  Used: SYSTIMER_TARGET2
  4     1    Level  Reserved
  5     1    Level  Reserved
  6     1    Level  CPU-internal
  7     1    Level  CPU-internal
  8     1    Level  Reserved
  9     1    Level  Used: FROM_CPU_INTR0
 10     1    Edge   Free (not general-use)
 11     3    Level  CPU-internal
 12     1    Level  Used: SYSTIMER_TARGET0
 13     1    Level  Used: SPI3
 14     7    Level  Reserved
 15     3    Level  CPU-internal
 16     5    Level  CPU-internal
 17     1    Level  Used: AES
 18     1    Level  Free
 19     2    Level  Free
 20     2    Level  Free
 21     2    Level  Free
 22     3    Edge   Reserved
 23     3    Level  Free
 24     4    Level  Reserved
 25     4    Level  Reserved
 26     5    Level  Free (not general-use)
 27     3    Level  Reserved
 28     4    Edge   Free (not general-use)
 29     3    Level  CPU-internal
 30     4    Edge   Reserved
 31     5    Level  Reserved
CPU 1 interrupt status:
 Int  Level  Type   Status
  0     1    Level  Reserved
  1     1    Level  Reserved
  2     1    Level  Used: FROM_CPU_INTR1
  3     1    Level  Used: SYSTIMER_TARGET1
  4     1    Level  Used: GPIO
  5     1    Level  Reserved
  6     1    Level  CPU-internal
  7     1    Level  CPU-internal
  8     1    Level  Reserved
  9     1    Level  Free
 10     1    Edge   Free (not general-use)
 11     3    Level  CPU-internal
 12     1    Level  Free
 13     1    Level  Free
 14     7    Level  Reserved
 15     3    Level  CPU-internal
 16     5    Level  CPU-internal
 17     1    Level  Free
 18     1    Level  Free
 19     2    Level  Used: TG0_T0_LEVEL
 20     2    Level  Free
 21     2    Level  Free
 22     3    Edge   Free (not general-use)
 23     3    Level  Free
 24     4    Level  Free (not general-use)
 25     4    Level  Reserved
 26     5    Level  Reserved
 27     3    Level  Reserved
 28     4    Edge   Free (not general-use)
 29     3    Level  CPU-internal
 30     4    Edge   Reserved
 31     5    Level  Reserved
Interrupts available for general use: 13
Shared interrupts: 0

Kainarx, 2024-08-16:
@Fuzhiguo36 我怀疑是你的过零检测信号有毛刺，短时间内多次进入中断，导致上一次timer还没有停的时候，又被启动了。你可以用示波器排查一下，如果是这样的话，可以使用 `GPIO glitch filter` 来过滤它。此外刚开始运行时的 `timer is not running` 错误，是因为你的定时器初始值为0，这样在刚初始化完定时器后就会立刻触发一次报警，你需要给他一个初始值（比如你的8000）后再使能它。

I suspect that your zero-crossing detection signal has glitches, causing multiple interrupts to be triggered within a short period of time. This might result in the timer being restarted before the previous one has stopped. You can use an oscilloscope to check for this issue. If this is the case, you can use a `GPIO glitch filter` to filter out the noise.

Additionally, the `timer is not running` error at the beginning is likely due to the timer's initial value being set to 0. This causes the timer to immediately trigger an alarm right after initialization. You need to set an initial value (e.g., your 8000) before enabling it.

filzek, 2024-08-21:
We have deal with Zero Crossing for a long time and are very expertise with it, the major problem in GPTIMER still with some issues in two concurrent GPTIMER being enable in the ISR the 1st will run okay, but the second sometimes will never enable or dispatch, so, there is some logic error with the GPTIMER routime.

closed 2025-03-31 by Alvin1Zhang