espressif/esp-idf #14368: Failed to build ULP Risc-V for ESP32S3

mike-2020, 2024-08-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Build successfully

### What is the actual behavior?

The build process generates the following errors:

[3/7] Building ASM object CMakeFiles/ulp_main.dir/main.ulp.S.obj
FAILED: CMakeFiles/ulp_main.dir/main.ulp.S.obj
esp32ulp-elf-as                                     --mcpu=esp32s2  -I/home/xxxx/ProjectName/main/include -I/home/xxxx/ProjectName/main/RemoteController -I/home/xxxx/esp-idf-v5.2.2/components/json -I/home/xxxx/esp-idf-v5.2.2/components/ulp/ulp_riscv/include/ulp_riscv -I/home/xxxx/esp-idf-v5.2.2/components/ulp/ulp_riscv/ulp_core/include -I/home/xxxx/ProjectName/TinyML -I/home/xxxx/ProjectName/TinyML/tflite-model -I/home/xxxx/ProjectName/TinyML/model-parameters -I/home/xxxx/ProjectName/TinyML/ingestion-sdk-c -I/home/xxxx/ProjectName/TinyML/edge-impulse-sdk -I/home/xxxx/ProjectName/build/config -I/home/xxxx/esp-idf-v5.2.2/components/newlib/platform_include -I/home/xxxx/esp-idf-v5.2.2/components/freertos/config/include -I/home/xxxx/esp-idf-v5.2.2/components/freertos/config/include/freertos -I/home/xxxx/esp-idf-v5.2.2/components/freertos/config/xtensa/include -I/home/xxxx/esp-idf-v5.2.2/components/freertos/FreeRTOS-Kernel/include -I/home/xxxx/esp-idf-v5.2.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/home/xxxx/esp-idf-v5.2.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos -I/home/xxxx/esp-idf-v5.2.2/components/freertos/esp_additions/include -I/home/xxxx/esp-idf-v5.2.2/components/esp_hw_support/include -I/home/xxxx/esp-idf-v5.2.2/components/esp_hw_support/include/soc -I/home/xxxx/esp-idf-v5.2.2/components/esp_hw_support/include/soc/esp32s3 -I/home/xxxx/esp-idf-v5.2.2/components/esp_hw_support/port/esp32s3/. -I/home/xxxx/esp-idf-v5.2.2/components/heap/include -I/home/xxxx/esp-idf-v5.2.2/components/log/include -I/home/xxxx/esp-idf-v5.2.2/components/soc/include -I/home/xxxx/esp-idf-v5.2.2/components/soc/esp32s3 -I/home/xxxx/esp-idf-v5.2.2/components/soc/esp32s3/include -I/home/xxxx/esp-idf-v5.2.2/components/hal/platform_port/include -I/home/xxxx/esp-idf-v5.2.2/components/hal/esp32s3/include -I/home/xxxx/esp-idf-v5.2.2/components/hal/include -I/home/xxxx/esp-idf-v5.2.2/components/esp_rom/include -I/home/xxxx/esp-idf-v5.2.2/components/esp_rom/include/esp32s3 -I/home/xxxx/esp-idf-v5.2.2/components/esp_rom/esp32s3 -I/home/xxxx/esp-idf-v5.2.2/components/esp_common/include -I/home/xxxx/esp-idf-v5.2.2/components/esp_system/include -I/home/xxxx/esp-idf-v5.2.2/components/esp_system/port/soc -I/home/xxxx/esp-idf-v5.2.2/components/esp_system/port/include/private -I/home/xxxx/esp-idf-v5.2.2/components/xtensa/esp32s3/include -I/home/xxxx/esp-idf-v5.2.2/components/xtensa/include -I/home/xxxx/esp-idf-v5.2.2/components/xtensa/deprecated_include -I/home/xxxx/esp-idf-v5.2.2/components/lwip/include -I/home/xxxx/esp-idf-v5.2.2/components/lwip/include/apps -I/home/xxxx/esp-idf-v5.2.2/components/lwip/include/apps/sntp -I/home/xxxx/esp-idf-v5.2.2/components/lwip/lwip/src/include -I/home/xxxx/esp-idf-v5.2.2/components/lwip/port/include -I/home/xxxx/esp-idf-v5.2.2/components/lwip/port/freertos/include -I/home/xxxx/esp-idf-v5.2.2/components/lwip/port/esp32xx/include -I/home/xxxx/esp-idf-v5.2.2/components/lwip/port/esp32xx/include/arch -I/home/xxxx/esp-idf-v5.2.2/components/lwip/port/esp32xx/include/sys -I/home/xxxx/esp-idf-v5.2.2/components/driver/include -I/home/xxxx/esp-idf-v5.2.2/components/driver/deprecated -I/home/xxxx/esp-idf-v5.2.2/components/driver/analog_comparator/include -I/home/xxxx/esp-idf-v5.2.2/components/driver/dac/include -I/home/xxxx/esp-idf-v5.2.2/components/driver/gpio/include -I/home/xxxx/esp-idf-v5.2.2/components/driver/gptimer/include -I/home/xxxx/esp-idf-v5.2.2/components/driver/i2c/include -I/home/xxxx/esp-idf-v5.2.2/components/driver/i2s/include -I/home/xxxx/esp-idf-v5.2.2/components/driver/ledc/include -I/home/xxxx/esp-idf-v5.2.2/components/driver/mcpwm/include -I/home/xxxx/esp-idf-v5.2.2/components/driver/parlio/include -I/home/xxxx/esp-idf-v5.2.2/components/driver/pcnt/include -I/home/xxxx/esp-idf-v5.2.2/components/driver/rmt/include -I/home/xxxx/esp-idf-v5.2.2/components/driver/sdio_slave/include -I/home/xxxx/esp-idf-v5.2.2/components/driver/sdmmc/include -I/home/xxxx/esp-idf-v5.2.2/components/driver/sigma_delta/include -I/home/xxxx/esp-idf-v5.2.2/components/driver/spi/include -I/home/xxxx/esp-idf-v5.2.2/components/driver/temperature_sensor/include -I/home/xxxx/esp-idf-v5.2.2/components/driver/touch_sensor/include -I/home/xxxx/esp-idf-v5.2.2/components/driver/twai/include -I/home/xxxx/esp-idf-v5.2.2/components/driver/uart/include -I/home/xxxx/esp-idf-v5.2.2/components/driver/usb_serial_jtag/include -I/home/xxxx/esp-idf-v5.2.2/components/driver/touch_sensor/esp32s3/include -I/home/xxxx/esp-idf-v5.2.2/components/esp_pm/include -I/home/xxxx/esp-idf-v5.2.2/components/esp_ringbuf/include -I/home/xxxx/esp-idf-v5.2.2/components/esp_adc/include -I/home/xxxx/esp-idf-v5.2.2/components/esp_adc/interface -I/home/xxxx/esp-idf-v5.2.2/components/esp_adc/esp32s3/include -I/home/xxxx/esp-idf-v5.2.2/components/esp_adc/deprecated/include -o CMakeFiles/ulp_main.dir/main.ulp.S.obj -c /home/xxxx/ProjectName/build/esp-idf/main/ulp_main/main.ulp.S
/home/xxxx/ProjectName/build/esp-idf/main/ulp_main/main.ulp.S: Assembler messages:
/home/xxxx/ProjectName/build/esp-idf/main/ulp_main/main.ulp.S:1: Error: syntax error. Input text was typedef.
/home/xxxx/ProjectName/build/esp-idf/main/ulp_main/main.ulp.S:1: Error:
/home/xxxx/ProjectName/build/esp-idf/main/ulp_main/main.ulp.S:2: Error: syntax error. Input text was typedef.
/home/xxxx/ProjectName/build/esp-idf/main/ulp_main/main.ulp.S:2: Error:
/home/xxxx/ProjectName/build/esp-idf/main/ulp_main/main.ulp.S:3: Error: syntax error. Input text was typedef.
/home/xxxx/ProjectName/build/esp-idf/main/ulp_main/main.ulp.S:3: Error:
/home/xxxx/ProjectName/build/esp-idf/main/ulp_main/main.ulp.S:4: Error: syntax error. Input text was typedef.
/home/xxxx/ProjectName/build/esp-idf/main/ulp_main/main.ulp.S:4: Error:
/home/xxxx/ProjectName/build/esp-idf/main/ulp_main/main.ulp.S:5: Error: syntax error. Input text was typedef.
/home/xxxx/ProjectName/build/esp-idf/main/ulp_main/main.ulp.S:5: Error:

### Steps to reproduce.

CMAKE file looks like below:

# ULP support additions to component CMakeLists.txt.
#
# 1. The ULP app name must be unique (if multiple components use ULP).
set(ulp_app_name ulp_${COMPONENT_NAME})
#
# 2. Specify all C and Assembly source files.
#    Files should be placed into a separate directory (in this case, ulp/),
#    which should not be added to COMPONENT_SRCS.
set(ulp_riscv_sources "ulp/main.c")

#
# 3. List all the component source files which include automatically
#    generated ULP export file, ${ulp_app_name}.h:
#set(ulp_exp_dep_srcs "ulp_riscv_adc_example_main.c")

#
# 4. Call function to build ULP binary and embed in project using the argument
#    values above.
ulp_embed_binary(${ulp_app_name} "${ulp_riscv_sources}" "${native_srcs}")

#native_srcs contains all C/C++ source files of the project (exlude the ones for ULP).

mike-2020, 2024-08-14:
CMake file: (native_srcs contains all C/C++ source files)

# ULP support additions to component CMakeLists.txt.
#
# 1. The ULP app name must be unique (if multiple components use ULP).
set(ulp_app_name ulp_${COMPONENT_NAME})
#
# 2. Specify all C and Assembly source files.
#    Files should be placed into a separate directory (in this case, ulp/),
#    which should not be added to COMPONENT_SRCS.
set(ulp_riscv_sources "ulp/main.c")

#
# 3. List all the component source files which include automatically
#    generated ULP export file, ${ulp_app_name}.h:
#set(ulp_exp_dep_srcs "ulp_riscv_adc_example_main.c")

#
# 4. Call function to build ULP binary and embed in project using the argument
#    values above.
ulp_embed_binary(${ulp_app_name} "${ulp_riscv_sources}" "${native_srcs}")

ESP-Marius, 2024-08-15:
Are you able to share your `ulp/main.c` file and your `sdkconfig`? Or even better share a project that can reproduce this error if possible.

mike-2020, 2024-08-15:
After many modifications, it works now. not sure which change made the magic.

Just want to confirm:
`#
# 3. List all the component source files which include automatically
#    generated ULP export file, ${ulp_app_name}.h:
#set(ulp_exp_dep_srcs "ulp_riscv_adc_example_main.c")`

Should ulp_exp_dep_srcs  contains all source files of the project? or it only needs contain the source file which call the ULP functions?

ESP-Marius, 2024-08-15:
It only needs to list the files that depends on the generated header files, i.e. the one which defines the shared variables.

closed 2024-08-16 by mike-2020

ayuusweetfish, 2024-12-10:
As a reference for future readers: I ran into a (probably) similar problem and it turned out to be due to CMake cache.

In particular, I ran the build step without setting `CONFIG_ULP_COPROC_ENABLED` in `sdkconfig`, and it yielded compilation errors (trying to compile the ULP program with the Xtensa toolchain); then I corrected `sdkconfig`, but the build still failed (reporting "*bad defsym; format is --defsym name=value*", etc.). It succeeded after an `idf.py clean` and a rebuild.

Pardon my disturbance.