espressif/esp-idf #14369: Unable to use some std::string using GNU++17  with abi:cxx11

italocjs, 2024-08-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am running espidf 4.4.7 (latest compatible with platformio) and c++17.   While all ESP32 code works file, when i try to run the same code on native and call a function that returns an std::string, it with following error:

```
Linking .pio/build/native/program
/usr/bin/ld: .pio/build/native/src/core/core.o: in function `Core::init()':
core.cpp:(.text+0x709): undefined reference to `getHalErrorString[abi:cxx11](hal_err_t)'
collect2: error: ld returned 1 exit status
*** [.pio/build/native/program] Error 1
============================== [FAILED] Took 1.67 seconds ==============================

Environment    Status    Duration
-------------  --------  ------------
native         FAILED    00:00:01.666
======================== 1 failed, 0 succeeded in 00:00:01.666 ==
````
I have tried everything i could find, but nothing seems to work, i think it may be related to the "abi:cxx11" the same code build perfectly fine on esp32dev though.  Both are supposed to be using c++ 17

Code is very simple (hal_err_t is an abstraction of esp_err_t, so i can also use it in native)

halTools.h
```
#include <string>
std::string getHalErrorString(hal_err_t error);
```

halTools.cpp
```

std::string getHalErrorString(hal_err_t error)
{
	switch (error)
	{
		case HAL_OK:
			return "HAL_OK";
		case HAL_FAIL:
			return "HAL_FAIL";
		case HAL_ERR_NO_MEM:
			return "HAL_ERR_NO_MEM";
		case HAL_ERR_INVALID_ARG:
			return "HAL_ERR_INVALID_ARG";
		case HAL_ERR_INVALID_STATE:
			return "HAL_ERR_INVALID_STATE";
		case HAL_ERR_INVALID_SIZE:
			return "HAL_ERR_INVALID_SIZE";
		case HAL_ERR_NOT_FOUND:
			return "HAL_ERR_NOT_FOUND";
		case HAL_ERR_NOT_SUPPORTED:
			return "HAL_ERR_NOT_SUPPORTED";
		case HAL_ERR_TIMEOUT:
			return "HAL_ERR_TIMEOUT";
		case HAL_ERR_INVALID_RESPONSE:
			return "HAL_ERR_INVALID_RESPONSE";
		case HAL_ERR_INVALID_CRC:
			return "HAL_ERR_INVALID_CRC";
		case HAL_ERR_INVALID_VERSION:
			return "HAL_ERR_INVALID_VERSION";
		case HAL_ERR_INVALID_MAC:
			return "HAL_ERR_INVALID_MAC";
		case HAL_ERR_NOT_FINISHED:
			return "HAL_ERR_NOT_FINISHED";
		case HAL_ERR_CUSTOM_BASE:
			return "HAL_ERR_CUSTOM_BASE";
		case HAL_ERR_CUSTOM_2:
			return "HAL_ERR_CUSTOM_2";
		case HAL_OTA_MODE:
			return "HAL_OTA_MODE";
	}
	return "Unknown";
}
```

pio.ini
```
[env:native]
lib_ldf_mode = deep+
test_framework = googletest
test_ignore = test_esp32dev
platform = native
build_unflags = -std=gnu++11
build_flags = 
    -D USE_NATIVE_ENVIRONMENT
    -std=gnu++17
lib_deps = 
    bblanchon/ArduinoJson

[env:esp32dev]
lib_ldf_mode = deep+
test_framework = googletest
test_ignore = test_native
platform = espressif32
board = simovatrack130
framework = arduino, espidf
; platform_packages =
    ; framework-arduinoespressif32 @ https://github.com/italocjs/arduino-esp32.git#idf-release/v4.4
    ; framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#idf-release/v4.4
    
board_build.partitions = partitions.csv
build_unflags = -std=gnu++11
build_flags = 
    -D USE_ESP32_ENVIRONMENT
    -std=gnu++17
upload_speed = 921600
monitor_speed = 115200
monitor_filters = 
    esp32_exception_decoder
    send_on_enter
lib_deps = 
    bblanchon/ArduinoJson

```

igrr, 2024-08-14:
> While all ESP32 code works file, when i try to run the same code on native and call a function that returns an std::string, it with following error:

> /usr/bin/ld ...

I'm sorry, your issue report contains too many things which aren't controlled by or related to ESP-IDF, the major two being the PlatformIO build system (especially the part about building the host application) and the host toolchain. As you see, the build for the ESP works fine, and that's the only part of your project which depends on ESP-IDF...

Without knowing more about how `platform = native` works in PlatformIO I'm afraid it's hard for ESP-IDF developers to help you. I would recommend seeking help on PlatformIO forum, instead.

(In case it helps, recent ESP-IDF versions contain a similar feature for [building applications for host](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/host-apps.html), and use [newer C++ standard](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/cplusplus.html#c-language-standard) by default as well.)

cristianfunes79, 2024-08-17:
Usually undefined reference means that you're not building that file and it is not available for the linker. Maybe you can add an `#error` to that file just to confirm it is being at least build for native compilation.

closed 2025-03-31 by Alvin1Zhang