espressif/esp-idf #14372: Secure Boot V2 on esp32s3 failed

aygh4266, 2024-08-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi everyone, 
so I am facing a problem running the app image after enabling sbv2 on esp32s3.
I have only enabled the secure boot v2 on esp32s3. I am using a ESP32-S3-DevKitM-1 with esp32s3-Mini-1 Chip.

I have generated the key pair using openSSL, enabled the secure boot v2 in menuconfig and entered the private key's path.

[image]

After that I have followed the steps shown in espressif documentation :https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/security/secure-boot-v2.html#how-to-enable-secure-boot-v2

I've used the ESP IDE but it didn't work so I've used the ESP-IDF CMD.

after running `idf.py monitor` 

```
Second output:
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
invalid header: 0x020150aa
invalid header: 0x020150aa
invalid header: 0x020150aa
invalid header: 0x020150aa
invalid header: 0x020150aa
```

```
#
# Security features
#
CONFIG_SECURE_SIGNED_ON_BOOT=y
CONFIG_SECURE_SIGNED_ON_UPDATE=y
CONFIG_SECURE_SIGNED_APPS=y
CONFIG_SECURE_BOOT_V2_RSA_SUPPORTED=y
CONFIG_SECURE_BOOT_V2_PREFERRED=y
CONFIG_SECURE_BOOT_V2_RSA_ENABLED=y
CONFIG_SECURE_SIGNED_APPS_RSA_SCHEME=y
CONFIG_SECURE_BOOT=y
CONFIG_SECURE_BOOT_V2_ENABLED=y
CONFIG_SECURE_BOOT_BUILD_SIGNED_BINARIES=y
CONFIG_SECURE_BOOT_SIGNING_KEY="sbv2_private_dev.pem"
# CONFIG_SECURE_BOOT_ENABLE_AGGRESSIVE_KEY_REVOKE is not set
# CONFIG_SECURE_BOOT_FLASH_BOOTLOADER_DEFAULT is not set
# CONFIG_SECURE_BOOT_INSECURE is not set
# CONFIG_SECURE_FLASH_ENC_ENABLED is not set
CONFIG_SECURE_ROM_DL_MODE_ENABLED=y
# CONFIG_SECURE_DISABLE_ROM_DL_MODE is not set
CONFIG_SECURE_ENABLE_SECURE_ROM_DL_MODE=y
# CONFIG_SECURE_INSECURE_ALLOW_DL_MODE is not set
# end of Security features
```
Output after running the command `esptool.py --chip esp32s3 --no-stub get_security_info`

```
esptool.py v4.7.0
Found 2 serial ports
Serial port COM4
Connecting....
Chip is ESP32-S3 (QFN56) (revision v0.1)
Features: WiFi, BLE, Embedded Flash 8MB (GD)
Crystal is 40MHz
MAC: f4:12:fa:87:c4:d8
Enabling default SPI flash mode...

Security Information:
=====================
Flags: 0x00000001 (0b1)
Key Purposes: (9, 0, 0, 0, 0, 0, 12)
Chip ID: 9
API Version: 0
Secure Boot: Enabled
Flash Encryption: Disabled
SPI Boot Crypt Count (SPI_BOOT_CRYPT_CNT): 0x0
Hard resetting via RTS pin...
```
Security eFuses after running the command `espefuse.py --chip esp32s3 -p COM4 summary`

```
Security fuses:
DIS_DOWNLOAD_ICACHE (BLOCK0)                       Set this bit to disable Icache in download mode (b = False R/W (0b0)
                                                   oot_mode[3:0] is 0; 1; 2; 3; 6; 7)
DIS_DOWNLOAD_DCACHE (BLOCK0)                       Set this bit to disable Dcache in download mode (  = False R/W (0b0)
                                                   boot_mode[3:0] is 0; 1; 2; 3; 6; 7)
DIS_FORCE_DOWNLOAD (BLOCK0)                        Set this bit to disable the function that forces c = False R/W (0b0)
                                                   hip into download mode
DIS_DOWNLOAD_MANUAL_ENCRYPT (BLOCK0)               Set this bit to disable flash encryption when in d = False R/W (0b0)
                                                   ownload boot modes
SPI_BOOT_CRYPT_CNT (BLOCK0)                        Enables flash encryption when 1 or 3 bits are set  = Disable R/W (0b000)
                                                   and disabled otherwise
SECURE_BOOT_KEY_REVOKE0 (BLOCK0)                   Revoke 1st secure boot key                         = False R/W (0b0)
SECURE_BOOT_KEY_REVOKE1 (BLOCK0)                   Revoke 2nd secure boot key                         = False R/W (0b0)
SECURE_BOOT_KEY_REVOKE2 (BLOCK0)                   Revoke 3rd secure boot key                         = False R/W (0b0)
KEY_PURPOSE_0 (BLOCK0)                             Purpose of Key0                                    = SECURE_BOOT_DIGEST0 R/- (0x9)
KEY_PURPOSE_1 (BLOCK0)                             Purpose of Key1                                    = USER R/W (0x0)
KEY_PURPOSE_2 (BLOCK0)                             Purpose of Key2                                    = USER R/W (0x0)
KEY_PURPOSE_3 (BLOCK0)                             Purpose of Key3                                    = USER R/W (0x0)
KEY_PURPOSE_4 (BLOCK0)                             Purpose of Key4                                    = USER R/W (0x0)
KEY_PURPOSE_5 (BLOCK0)                             Purpose of Key5                                    = USER R/W (0x0)
SECURE_BOOT_EN (BLOCK0)                            Set this bit to enable secure boot                 = True R/W (0b1)
SECURE_BOOT_AGGRESSIVE_REVOKE (BLOCK0)             Set this bit to enable revoking aggressive secure  = False R/W (0b0)
                                                   boot
DIS_DOWNLOAD_MODE (BLOCK0)                         Set this bit to disable download mode (boot_mode[3 = False R/W (0b0)
                                                   :0] = 0; 1; 2; 3; 6; 7)
ENABLE_SECURITY_DOWNLOAD (BLOCK0)                  Set this bit to enable secure UART download mode   = False R/W (0b0)
SECURE_VERSION (BLOCK0)                            Secure version (used by ESP-IDF anti-rollback feat = 0 R/W (0x0000)
                                                   ure)
BLOCK_KEY0 (BLOCK4)
  Purpose: SECURE_BOOT_DIGEST0
  Key0 or user data
   = 97 1a 8b 68 17 91 85 90 40 de a6 92 96 d7 29 0b 07 0e 96 3d da 47 94 97 12 2f b3 51 ca 75 75 ed R/-
BLOCK_KEY1 (BLOCK5)
  Purpose: USER
               Key1 or user data
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLOCK_KEY2 (BLOCK6)
  Purpose: USER
               Key2 or user data
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLOCK_KEY3 (BLOCK7)
  Purpose: USER
               Key3 or user data
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLOCK_KEY4 (BLOCK8)
  Purpose: USER
               Key4 or user data
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLOCK_KEY5 (BLOCK9)
  Purpose: USER
               Key5 or user data
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

Spi Pad fuses:
SPI_PAD_CONFIG_CLK (BLOCK1)                        SPI_PAD_configure CLK                              = 0 R/W (0b000000)
SPI_PAD_CONFIG_Q (BLOCK1)                          SPI_PAD_configure Q(D1)                            = 0 R/W (0b000000)
SPI_PAD_CONFIG_D (BLOCK1)                          SPI_PAD_configure D(D0)                            = 0 R/W (0b000000)
SPI_PAD_CONFIG_CS (BLOCK1)                         SPI_PAD_configure CS                               = 0 R/W (0b000000)
SPI_PAD_CONFIG_HD (BLOCK1)                         SPI_PAD_configure HD(D3)                           = 0 R/W (0b000000)
SPI_PAD_CONFIG_WP (BLOCK1)                         SPI_PAD_configure WP(D2)                           = 0 R/W (0b000000)
SPI_PAD_CONFIG_DQS (BLOCK1)                        SPI_PAD_configure DQS                              = 0 R/W (0b000000)
SPI_PAD_CONFIG_D4 (BLOCK1)                         SPI_PAD_configure D4                               = 0 R/W (0b000000)
SPI_PAD_CONFIG_D5 (BLOCK1)                         SPI_PAD_configure D5                               = 0 R/W (0b000000)
SPI_PAD_CONFIG_D6 (BLOCK1)                         SPI_PAD_configure D6                               = 0 R/W (0b000000)
SPI_PAD_CONFIG_D7 (BLOCK1)                         SPI_PAD_configure D7                               = 0 R/W (0b000000)

```

I also checked whether the private key and the generated signaure match by running this command : `espsecure.py verify_signature --version 2 --keyfile sbv2_private_dev.pem build/app-template.bin`

```
espsecure.py v4.7.0
Signature block 0 is valid (RSA).
Signature block 0 verification successful using the supplied key (RSA).
Signature block 1 invalid. Skipping.
Signature block 2 invalid. Skipping.
```

but it still didn't work. Any help would be appreciated
thanks

aygh4266, 2024-08-15:
trying to flash the bootloader to address 0x0 because  I use esp32s3 shows me this:

`esptool.py -b 115200 --before default_reset --after no_reset --chip esp32s3  --no-stub write_flash --flash_mode dio --flash_size`
 ```
detect --flash_freq 80m 0x0000 build/bootloader/bootloader.bin
esptool.py v4.7.0
Found 2 serial ports
Serial port COM4
Connecting....
Chip is ESP32-S3 (QFN56) (revision v0.1)
Features: WiFi, BLE, Embedded Flash 8MB (GD)
Crystal is 40MHz
MAC: f4:12:fa:87:c4:d8
Enabling default SPI flash mode...
Configuring flash size...
Auto-detected Flash size: 8MB

A fatal error occurred: Secure Boot detected, writing to flash regions < 0x8000 is disabled to protect the bootloader. Use --force to override, please use with caution, otherwise it may brick your device!
```

aygh4266, 2024-08-15:
I Solved the problem:

The signed Bootloader image must be flashed at 0x0. Using the above command with --force solved the problem.

You can keep the ticket for the other to benefit from the solution

mahavirj, 2024-08-16:
@aygh4266 

Bootloader offset comes from `BOOTLOADER_OFFSET_IN_FLASH` config option:

https://github.com/espressif/esp-idf/blob/d7ca8b94c852052e3bc33292287ef4dd62c9eeb1/components/bootloader/Kconfig.projbuild#L5-L14

Same config option gets used in IDF build system to construct different flashing commands. Probably you didn't use `idf.py flash` based workflow and hence this problem. 

Closing this issue as it should not be a problem if you follow instructions per IDF secure boot v2 guide.

closed 2024-08-16 by mahavirj

aygh4266, 2024-08-16:
> Closing this issue as it should not be a problem if you follow instructions per IDF secure boot v2 guide.
I have followed the instructions per IDF and also shown the webinar on YouTube regarding secure boot v2 but it didn't work. So I have mentioned this issue but you can close it, since it works right now

dmitrij9992905, 2025-06-29:
Hello everyone!

I have tha same issue even with self-generated Flash Encryption key with Secure boot.
FE - Flash Encryption
SB - Secure Boot
OB - onboard-generated flash encryption key
H - host-generated key
P - preencrypted images

AES128XTS:
FE OB + SB -> invalid header
FE H + SB -> invalid header
FE H P + SB -> invalid header

AES256XTS:
FE OB + SB -> invalid header
FE H + SB -> tbc
FE H P + SB -> tbc

SB -> OK

In the ESP32 case in my experience, FE H + SB -> OK, but didn't work on ESP32S3.
Looks like a maskROM bug.
I was flashing entire flash image built from necessary binaries (bootloader and app are already encrypted), tried both encrypted and plain