espressif/esp-idf #14375: ESP32-MINI-1 module The UART can output messages, but it can't burn them (only the first time).

Knockoi, 2024-08-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esptool.py v4.6

### Espressif SoC revision.

ESP32-MINI-1U module

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

I designed the development version myself, but I followed the official manual and the schematic of the ESP32-DevKitM-1 board to the letter.

### Power Supply used.

USB

### What is the expected behavior?

Programming can be done using USB to UART (using CH340X) while driving I2C, 5 ADC, 4 PWM and two RGB channels.
GPIO34, 35, 37, 38, 39 are ADCs.
GPIO22 SCL
GPIO21SDA
GPIO23, 25, 33, 18 are PWM.
GPIO19, 26 are RGB

### What is the actual behavior?

The first time I programmed it, there was no problem at all, but the second time I couldn't access the FLASH, and every time after that, no matter how I replaced the programmer or restarted the computer or used the external power supply, I couldn't use it, and I could only use the UART to output the contents of the first time I programmed it all the time.

### Steps to reproduce.

1. Step1 To connect to ESP32 using ch340x, the connection is
2. Step2 Use the ARDUINO IDE to burn Example01 of sparkfun bmi270.
3. Step3 wait...
...

### Debug Logs.

```plain
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download

esptool.py v4.6
Serial port COM8
Connecting....
Chip is ESP32-U4WDH (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: cc:7b:5c:19:5e:78
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x0005efff...
Compressed 18304 bytes to 12742...
Writing at 0x00001000... (100 %)
Wrote 18304 bytes (12742 compressed) at 0x00001000 in 1.4 seconds (effective 102.8 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 146...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.1 seconds (effective 293.5 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 462.2 kbit/s)...
Hash of data verified.
Compressed 320256 bytes to 182645...
Writing at 0x00010000... (8 %)
Writing at 0x0001944c... (16 %)
Writing at 0x00027170... (25 %)
Writing at 0x0002c68f... (33 %)
Writing at 0x00031eb0... (41 %)
Writing at 0x0003768c... (50 %)
Writing at 0x0003cd5d... (58 %)
Writing at 0x00042243... (66 %)
Writing at 0x000475bb... (75 %)
Writing at 0x0004d5ea... (83 %)
Writing at 0x000573d0... (91 %)
Traceback (most recent call last):
  File "esptool\__init__.py", line 1064, in _main
  File "esptool\__init__.py", line 859, in main
  File "esptool\cmds.py", line 595, in write_flash
  File "esptool\loader.py", line 131, in inner
  File "esptool\loader.py", line 1032, in flash_defl_block
  File "esptool\loader.py", line 435, in check_command
  File "esptool\loader.py", line 404, in command
  File "esptool\loader.py", line 337, in read
StopIteration

A fatal error occurred: The chip stopped responding.
上傳失敗: 上傳錯誤: exit status 2
```

### More Information.

I'm running and burning in arduino ide.
I also try to use the node mcu programming system.
and the official flash_download_tool_3.9.7 from Loxin.
I can't program or wipe the flash.
More pictures of the error and how to connect the circuits:https://github.com/Knockoi/E-Stick-V3/tree/main/Debug

arosyc, 2024-08-16:
Did you try download with FT232 or CP2102 USB-UART chips? @Knockoi

Knockoi, 2024-08-16:
> Did you try download with FT232 or CP2102 USB-UART chips? @Knockoi

Thanks for the reply, yes I have tested other burner ICs and also updated the drivers but no sign of improvement.

closed 2025-03-14 by dobairoland