espressif/esp-idf #14378: [Mocking] KconfigError when running idf.py --preview set-target Linux

wilmer-suarez-simplisafe, 2024-08-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm currently running ESP-IDF v5.02.
I'm working on setting up host unit testing on macOS. I've successfully done so for a couple of custom components (non esp-idf) so far, but I'm having issues with a particular component where when I run `idf.py --preview set-target linux` I get the error below. I'm setting up the directory structure and CMakeLists.txt files in the same way for all the components I've configured so far (following https://docs.espressif.com/projects/esp-idf/en/release-v5.0/esp32/api-guides/linux-host-testing.html); but I'm not getting the same behavior here. Since this failing before getting to the 'build' step, dependencies and mocks (if necessary) shouldn't be the underlying issue here.

```
 idf.py --preview set-target linux
/Users/wilmer.suarez/Desktop/Repositories/repo/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/Users/wilmer.suarez/Desktop/Repositories/repo/components/comp/host_test/test_suite_1/build' not found. Nothing to clean.
Executing action: set-target
Set Target to: linux, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory /Users/wilmer.suarez/Desktop/Repositories/repo/components/comp/host_test/test_suite_1/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=linux -DIDF_TARGET=linux -DCCACHE_ENABLE=0 /Users/wilmer.suarez/Desktop/Repositories/repo/components/comp/host_test/test_suite_1"...
-- Found Git: /opt/homebrew/bin/git (found version "2.43.0")
CMake Warning (dev) at /Users/wilmer.suarez/Desktop/Repositories/repo/esp-idf/tools/cmake/build.cmake:129 (enable_language):
  project() should be called prior to this enable_language() call.
Call Stack (most recent call first):
  /Users/wilmer.suarez/Desktop/Repositories/repo/esp-idf/tools/cmake/build.cmake:191 (__build_set_lang_version)
  /Users/wilmer.suarez/Desktop/Repositories/repo/esp-idf/tools/cmake/idf.cmake:53 (__build_init)
  /Users/wilmer.suarez/Desktop/Repositories/repo/esp-idf/tools/cmake/project.cmake:12 (include)
  CMakeLists.txt:3 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- The C compiler identification is AppleClang 15.0.0.15000309
-- The CXX compiler identification is AppleClang 15.0.0.15000309
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test ver_gnu17_supported
-- Performing Test ver_gnu17_supported - Success
-- Performing Test ver_gnu++20_supported
-- Performing Test ver_gnu++20_supported - Success
-- Component directory /Users/wilmer.suarez/Desktop/Repositories/repo/esp-idf/components/asio does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/wilmer.suarez/Desktop/Repositories/repo/esp-idf/components/aws_iot does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/wilmer.suarez/Desktop/Repositories/repo/esp-idf/components/coap does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/wilmer.suarez/Desktop/Repositories/repo/esp-idf/components/esp32 does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/wilmer.suarez/Desktop/Repositories/repo/esp-idf/components/expat does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/wilmer.suarez/Desktop/Repositories/repo/esp-idf/components/libsodium does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/wilmer.suarez/Desktop/Repositories/repo/esp-idf/components/micro-ecc does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/wilmer.suarez/Desktop/Repositories/repo/esp-idf/components/nghttp does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/wilmer.suarez/Desktop/Repositories/repo/esp-idf/components/nimble does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/wilmer.suarez/Desktop/Repositories/repo/components/ss3_fw_tools does not contain a CMakeLists.txt file. No component will be added
-- The ASM compiler identification is AppleClang
-- Found assembler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Building ESP-IDF components for target linux
-- building FREERTOS MOCKS (only task, event-groups and queue)
-- building CUSTOM_COMP_1 MOCK
-- building CUSOMT_COMP_2
-- Project sdkconfig file /Users/wilmer.suarez/Desktop/Repositories/repo/components/comp/host_test/test_suite_1/sdkconfig
Traceback (most recent call last):
  File "/Users/wilmer.suarez/Desktop/Repositories/repo/esp-idf/tools/kconfig_new/confgen.py", line 607, in <module>
    main()
  File "/Users/wilmer.suarez/Desktop/Repositories/repo/esp-idf/tools/kconfig_new/confgen.py", line 258, in main
    config = kconfiglib.Kconfig(args.kconfig)
[... 8 lines cut ...]
  File "/Users/wilmer.suarez/.espressif/python_env/idf5.0_py3.12_env/lib/python3.12/site-packages/kconfiglib.py", line 2989, in _parse_block
    prev = self._parse_block(None, parent, prev)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/wilmer.suarez/.espressif/python_env/idf5.0_py3.12_env/lib/python3.12/site-packages/kconfiglib.py", line 2989, in _parse_block
    prev = self._parse_block(None, parent, prev)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/wilmer.suarez/.espressif/python_env/idf5.0_py3.12_env/lib/python3.12/site-packages/kconfiglib.py", line 3028, in _parse_block
    self._parse_block(_T_ENDMENU, node, node)
  File "/Users/wilmer.suarez/.espressif/python_env/idf5.0_py3.12_env/lib/python3.12/site-packages/kconfiglib.py", line 3028, in _parse_block
    self._parse_block(_T_ENDMENU, node, node)
  File "/Users/wilmer.suarez/.espressif/python_env/idf5.0_py3.12_env/lib/python3.12/site-packages/kconfiglib.py", line 2977, in _parse_block
    raise KconfigError(
kconfiglib.KconfigError: /Users/wilmer.suarez/Desktop/Repositories/repo/esp-idf/components/esp_hw_support/Kconfig:24: '/Users/wilmer.suarez/Desktop/Repositories/repo/esp-idf/components/esp_hw_support/./port/linux/Kconfig.mac' not found (in 'rsource "./port/$IDF_TARGET/Kconfig.mac"'). Check that environment variables are set correctly (e.g. $srctree, which is unset or blank). Also note that unset environment variables expand to the empty string.
CMake Error at /Users/wilmer.suarez/Desktop/Repositories/repo/esp-idf/tools/cmake/kconfig.cmake:176 (message):
  Failed to run confgen.py
  (/Users/wilmer.suarez/.espressif/python_env/idf5.0_py3.12_env/bin/python;/Users/wilmer.suarez/Desktop/Repositories/repo/esp-idf/tools/kconfig_new/confgen.py;--list-separator=semicolon;--kconfig;/Users/wilmer.suarez/Desktop/Repositories/repo/esp-idf/Kconfig;--sdkconfig-rename;/Users/wilmer.suarez/Desktop/Repositories/repo/esp-idf/sdkconfig.rename;--config;/Users/wilmer.suarez/Desktop/Repositories/repo/components/comp/host_test/test_suite_1/sdkconfig;--defaults;/Users/wilmer.suarez/Desktop/Repositories/repo/components/comp/host_test/test_suite_1/sdkconfig.defaults;--env-file;/Users/wilmer.suarez/Desktop/Repositories/repo/components/comp/host_test/test_suite_1/build/config.env).
  Error 1
Call Stack (most recent call first):
[... cut ...] ig)
  /Users/wilmer.suarez/Desktop/Repositories/repo/esp-idf/tools/cmake/project.cmake:447 (idf_build_process)
  CMakeLists.txt:20 (project)

-- Configuring incomplete, errors occurred!
cmake failed with exit code 1, output of the command is in the /Users/wilmer.suarez/Desktop/Repositories/repo/components/comp/host_test/test_suite_1/build/log/idf_py_stderr_output_2007 and /Users/wilmer.suarez/Desktop/Repositories/repo/components/comp/host_test/test_suite_1/build/log/idf_py_stdout_output_2007
```

ESP-Marius, 2024-08-16:
Thanks for reporting this. We'll take a look at it and get back to you.

ESP-Marius, 2024-08-20:
@wilmer-suarez-simplisafe 

The actual error seems to be: 
 
```
kconfiglib.KconfigError: /Users/wilmer.suarez/Desktop/Repositories/repo/esp-idf/components/esp_hw_support/Kconfig:24: '/Users/wilmer.suarez/Desktop/Repositories/repo/esp-idf/components/esp_hw_support/./port/linux/Kconfig.mac' not found (in 'rsource "./port/$IDF_TARGET/Kconfig.mac"'). Check that environment variables are set correctly (e.g. $srctree, which is unset or blank). Also note that unset environment variables expand to the empty string.
```

So what seems to happen is that something in your build tries to pull in `esp-hw-support`, which in turn tries to include `port/linux/Kconfig.mac` (which does not exist).

The linux target was in very early preview mode on 5.0, so lots of components do not support being built for the linux target.

I assume you limit the components pulled in with `set(COMPONENTS main)` right? If so then I suggest you try to chase down what pulls in `esp-hw-support` into the build and stop that from happening.

If you are not restricted to using IDF 5.0 then trying this on 5.3 would probably provide better results, as a lot of these kind of issues with the linux target has been fixed by now.

wilmer-suarez-simplisafe, 2024-08-20:
Yes, I'm limiting components using `set(COMPONENTS main)`. I don't have anything pulling in `esp-hw-support`.
I did manage to get past this. I didn't provide a mock for every direct dependency for the `component-under-test`. I.e., everything under every REQUIRES command in the associated `CMakelists.txt` files. Once I did that, it finished successfully.
It would be nice if that could be made clear in the error printout. Is that the case in 5.3?

ESP-Marius, 2024-08-21:
> It would be nice if that could be made clear in the error printout. Is that the case in 5.3?

In 5.3 all components build for linux (a lot of them are just empty components though), so I dont think we should see any such issues there.

ESP-Marius, 2024-08-26:
I think we can consider this issue resolved then. Feel free to reopen if you think there is still something we need to follow up on here.

closed 2024-08-26 by ESP-Marius