espressif/esp-idf #14379: [Target: Linux] call to undeclared function 'getentropy'

darvik80, 2024-08-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

build successfull

### What is the actual behavior?

/Users/ivan.kishchenko/esp/esp-idf-v5.3/components/esp_hw_support/port/linux/esp_random.c:20:18: error: call to undeclared function 'getentropy'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
    int result = getentropy(&random_number, sizeof(random_number));
                 ^
/Users/ivan.kishchenko/esp/esp-idf-v5.3/components/esp_hw_support/port/linux/esp_random.c:37:18: error: call to undeclared function 'getentropy'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
        result = getentropy(buf + chunk_num * GETENTROPY_MAX_LEN, GETENTROPY_MAX_LEN);
                 ^
/Users/ivan.kishchenko/esp/esp-idf-v5.3/components/esp_hw_support/port/linux/esp_random.c:42:14: error: call to undeclared function 'getentropy'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
    result = getentropy(buf + FULL_CHUNKS_NUM * GETENTROPY_MAX_LEN, REST_CHUNK_SIZE);

### Steps to reproduce.

1. . $HOME/esp/esp-idf-v5.3/export.sh
2. idf.py create-project esp32-linux
2. cd esp32-linux
3.  modify: CMakefile
...
set(COMPONENTS main)
...
4. idf.py --preview set-target linux
5. idf.py build

### Build or installation Logs.

```plain
Executing action: all (aliases: build)
Running ninja in directory /Users/ivan.kishchenko/projects/cpp/esp32-linux/build
Executing "ninja all"...
[0/1] Re-running CMake...CMake Warning (dev) at /Users/ivan.kishchenko/esp/esp-idf-v5.3/tools/cmake/build.cmake:137 (enable_language):
  project() should be called prior to this enable_language() call.
Call Stack (most recent call first):
  /Users/ivan.kishchenko/esp/esp-idf-v5.3/tools/cmake/build.cmake:214 (__build_set_lang_version)
  /Users/ivan.kishchenko/esp/esp-idf-v5.3/tools/cmake/idf.cmake:52 (__build_init)
  /Users/ivan.kishchenko/esp/esp-idf-v5.3/tools/cmake/project.cmake:29 (include)
  CMakeLists.txt:7 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target linux
-- Project sdkconfig file /Users/ivan.kishchenko/projects/cpp/esp32-linux/sdkconfig
-- Components: esp_common esp_hw_support esp_rom esp_system freertos hal heap linux log main soc
-- Component paths: /Users/ivan.kishchenko/esp/esp-idf-v5.3/components/esp_common /Users/ivan.kishchenko/esp/esp-idf-v5.3/components/esp_hw_support /Users/ivan.kishchenko/esp/esp-idf-v5.3/components/esp_rom /Users/ivan.kishchenko/esp/esp-idf-v5.3/components/esp_system /Users/ivan.kishchenko/esp/esp-idf-v5.3/components/freertos /Users/ivan.kishchenko/esp/esp-idf-v5.3/components/hal /Users/ivan.kishchenko/esp/esp-idf-v5.3/components/heap /Users/ivan.kishchenko/esp/esp-idf-v5.3/components/linux /Users/ivan.kishchenko/esp/esp-idf-v5.3/components/log /Users/ivan.kishchenko/projects/cpp/esp32-linux/main /Users/ivan.kishchenko/esp/esp-idf-v5.3/components/soc
-- Configuring done (2.2s)
-- Generating done (0.0s)
-- Build files have been written to: /Users/ivan.kishchenko/projects/cpp/esp32-linux/build
[11/48] Building C object esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/port/linux/esp_random.c.o
FAILED: esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/port/linux/esp_random.c.o 
/Library/Developer/CommandLineTools/usr/bin/cc -DESP_PLATFORM -DIDF_VER=\"v5.3\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -DprojCOVERAGE_TEST=0 -I/Users/ivan.kishchenko/projects/cpp/esp32-linux/build/config -I/Users/ivan.kishchenko/esp/esp-idf-v5.3/components/esp_hw_support/include -I/Users/ivan.kishchenko/esp/esp-idf-v5.3/components/freertos/config/include -I/Users/ivan.kishchenko/esp/esp-idf-v5.3/components/freertos/config/include/freertos -I/Users/ivan.kishchenko/esp/esp-idf-v5.3/components/freertos/config/linux/include -I/Users/ivan.kishchenko/esp/esp-idf-v5.3/components/freertos/FreeRTOS-Kernel/include -I/Users/ivan.kishchenko/esp/esp-idf-v5.3/components/freertos/FreeRTOS-Kernel/portable/linux/include -I/Users/ivan.kishchenko/esp/esp-idf-v5.3/components/freertos/FreeRTOS-Kernel/portable/linux/include/freertos -I/Users/ivan.kishchenko/esp/esp-idf-v5.3/components/freertos/esp_additions/include -I/Users/ivan.kishchenko/esp/esp-idf-v5.3/components/heap/include -I/Users/ivan.kishchenko/esp/esp-idf-v5.3/components/log/include -I/Users/ivan.kishchenko/esp/esp-idf-v5.3/components/soc/include -I/Users/ivan.kishchenko/esp/esp-idf-v5.3/components/soc/linux -I/Users/ivan.kishchenko/esp/esp-idf-v5.3/components/soc/linux/include -I/Users/ivan.kishchenko/esp/esp-idf-v5.3/components/hal/platform_port/include -I/Users/ivan.kishchenko/esp/esp-idf-v5.3/components/hal/include -I/Users/ivan.kishchenko/esp/esp-idf-v5.3/components/esp_rom/include -I/Users/ivan.kishchenko/esp/esp-idf-v5.3/components/esp_rom/include/linux -I/Users/ivan.kishchenko/esp/esp-idf-v5.3/components/esp_common/include -I/Users/ivan.kishchenko/esp/esp-idf-v5.3/components/esp_system/include -I/Users/ivan.kishchenko/esp/esp-idf-v5.3/components/linux/include -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX14.4.sdk -fcolor-diagnostics -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -Og -Wno-documentation -Wno-typedef-redefinition -Wno-char-subscripts -Wno-format-security -Wno-tautological-overlap-compare -Wno-tautological-pointer-compare -Wno-pointer-bool-conversion -Wno-string-concatenation -Wno-section -Wno-unknown-attributes -Wno-atomic-alignment -Wno-unused-but-set-variable -Wno-unused-function -Wno-gnu-variable-sized-type-not-at-end -Wno-constant-logical-operand -Wno-c2x-extensions -Wno-extern-c-compat -fmacro-prefix-map=/Users/ivan.kishchenko/projects/cpp/esp32-linux=. -fmacro-prefix-map=/Users/ivan.kishchenko/esp/esp-idf-v5.3=/IDF -fno-jump-tables -fno-use-cxa-atexit -std=gnu17 -MD -MT esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/port/linux/esp_random.c.o -MF esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/port/linux/esp_random.c.o.d -o esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/port/linux/esp_random.c.o -c /Users/ivan.kishchenko/esp/esp-idf-v5.3/components/esp_hw_support/port/linux/esp_random.c
/Users/ivan.kishchenko/esp/esp-idf-v5.3/components/esp_hw_support/port/linux/esp_random.c:20:18: error: call to undeclared function 'getentropy'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
    int result = getentropy(&random_number, sizeof(random_number));
                 ^
/Users/ivan.kishchenko/esp/esp-idf-v5.3/components/esp_hw_support/port/linux/esp_random.c:37:18: error: call to undeclared function 'getentropy'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
        result = getentropy(buf + chunk_num * GETENTROPY_MAX_LEN, GETENTROPY_MAX_LEN);
                 ^
/Users/ivan.kishchenko/esp/esp-idf-v5.3/components/esp_hw_support/port/linux/esp_random.c:42:14: error: call to undeclared function 'getentropy'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
    result = getentropy(buf + FULL_CHUNKS_NUM * GETENTROPY_MAX_LEN, REST_CHUNK_SIZE);
             ^
3 errors generated.
[20/48] Building C object esp-idf/freertos/CMakeFiles/__idf_freertos.dir/FreeRTOS-Kernel/tasks.c.o
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/ivan.kishchenko/projects/cpp/esp32-linux/build/log/idf_py_stderr_output_94203 and /Users/ivan.kishchenko/projects/cpp/esp32-linux/build/log/idf_py_stdout_output_94203
```

ESP-Marius, 2024-08-16:
Hey, which version of macos are you using?

I see that  `getentropy` might be missing on some older version of macos

darvik80, 2024-08-16:
Lastest version: 14.6.1 (23G93). 
In fact fix quite simple: 
components/esp_hw_support/port/linux/esp_random.c
...
#include <sys/random.h>
...

ESP-Marius, 2024-08-16:
Ah, we have actually already fixed this on master, but I missed the fact that you are on 5.3 and the fix hasnt been backported.

I'll make sure we do that.

ESP-Marius, 2024-09-23:
Should be OK now. Closing this issue.

closed 2024-09-23 by ESP-Marius