espressif/esp-idf #14381: ***ERROR*** A stack overflow in task uart_echo_task has been detected.

franzhoepfinger, 2024-08-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

https://github.com/espressif/esp-idf/tree/master/examples/peripherals/uart/uart_echo_rs485

```
I (31) boot: ESP-IDF v5.3-dirty 2nd stage bootloader
I (31) boot: compile time Aug 16 2024 15:02:37
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v0.2
I (39) boot.esp32s3: Boot SPI Speed : 80MHz
I (43) boot.esp32s3: SPI Mode       : DIO
I (48) boot.esp32s3: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0b6b0h ( 46768) map
I (112) esp_image: segment 1: paddr=0001b6d8 vaddr=3fc93300 size=02a44h ( 10820) load
I (115) esp_image: segment 2: paddr=0001e124 vaddr=40374000 size=01ef4h (  7924) load
I (123) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=1ae1ch (110108) map
I (149) esp_image: segment 4: paddr=0003ae44 vaddr=40375ef4 size=0d38ch ( 54156) load
I (167) boot: Loaded app from partition at offset 0x10000
I (168) boot: Disabling RNG early entropy source...
I (179) cpu_start: Multicore app
I (189) cpu_start: Pro cpu start user code
I (189) cpu_start: cpu freq: 160000000 Hz
I (189) app_init: Application information:
I (192) app_init: Project name:     echo_rs485
I (197) app_init: App version:      06a43cd-dirty
I (202) app_init: Compile time:     Aug 16 2024 15:07:04
I (208) app_init: ELF file SHA256:  9ba299e7c...
Warning: checksum mismatch between flashed and built applications. Checksum of built application is 7260e850cf599ffe91c2a526516796dd869dbbd23908143290719461902054ec
I (214) app_init: ESP-IDF:          v5.3-dirty
I (219) efuse_init: Min chip rev:     v0.0
I (224) efuse_init: Max chip rev:     v0.99 
I (229) efuse_init: Chip rev:         v0.2
I (233) heap_init: Initializing. RAM available for dynamic allocation:
I (241) heap_init: At 3FC96608 len 00053108 (332 KiB): RAM
I (247) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (253) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (259) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (266) spi_flash: detected chip: generic
I (270) spi_flash: flash io: dio
W (274) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (287) sleep: Configure to isolate all GPIO pins in sleep state
I (294) sleep: Enable automatic switching of GPIO sleep configuration
I (301) main_task: Started on CPU0
I (321) main_task: Calling app_main()
I (321) RS485_ECHO_APP: Start RS485 application test and configure UART.
I (321) RS485_ECHO_APP: UART set pins, mode and install driver.
I (331) RS485_ECHO_APP: UART start recieve loop.

***ERROR*** A stack overflow in task uart_echo_task has been detected.

Backtrace: 0x40375ab6:0x3fc9a7c0 0x4037a979:0x3fc9a7e0 0x4037b78e:0x3fc9a800 0x4037c997:0x3fc9a880 0x4037b854:0x3fc9a8a0 0x4037b84a:0x3c025a69 |<-CORRUPTED
0x40375ab6: panic_abort at /home/franz/esp-idf-v5.3/components/esp_system/panic.c:463
0x4037a979: esp_system_abort at /home/franz/esp-idf-v5.3/components/esp_system/port/esp_system_chip.c:92
0x4037b78e: vApplicationStackOverflowHook at /home/franz/esp-idf-v5.3/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:553
0x4037c997: vTaskSwitchContext at /home/franz/esp-idf-v5.3/components/freertos/FreeRTOS-Kernel/tasks.c:3701 (discriminator 7)
0x4037b854: _frxt_dispatch at /home/franz/esp-idf-v5.3/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:451
0x4037b84a: _frxt_int_exit at /home/franz/esp-idf-v5.3/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:246

ELF file SHA256: 9ba299e7c
Warning: checksum mismatch between flashed and built applications. Checksum of built application is 7260e850cf599ffe91c2a526516796dd869dbbd23908143290719461902054ec

Rebooting...
���ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x29 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403759f0
0x403759f0: esp_restart_noos at /home/franz/esp-idf-v5.3/components/esp_system/port/soc/esp32s3/system_internal.c:158

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x178c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xcb8
load:0x403cb700,len:0x2db0
entry 0x403c8914
I (31) boot: ESP-IDF v5.3-dirty 2nd stage bootloader
```

franzhoepfinger, 2024-08-16:
lets try: 

`#define ECHO_TASK_STACK_SIZE    (2*2048)`

franzhoepfinger, 2024-08-16:
also the Watchdog is triggered: 

```
[ 0x0D 0x0A 0x0A 0x0A 0x7E (72636) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (72636) task_wdt:  - IDLE0 (CPU 0)
E (72636) task_wdt: Tasks currently running:
E (72636) task_wdt: CPU 0: uart_echo_task
E (72636) task_wdt: CPU 1: IDLE1
E (72636) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x4200CDC3:0x3FC94320 0x4200D1E0:0x3FC94340 0x4037725D:0x3FC94370 0x4200644A:0x3FC9AD80 0x4200731F:0x3FC9ADB0 0x4200A395:0x3FC9ADD0 0x4200F35A:0x3FC9ADF0 0x42009EE4:0x3FC9AE10 0x4200A395:0x3FC9AE30 0x4200ED9A:0x3FC9AE50 0x4200E5D5:0x3FC9AE70 0x4200E626:0x3FC9AE90 0x4200EBCA:0x3FC9AEB0 0x42012CD3:0x3FC9AEE0 0x42012679:0x3FC9AF00 0x4200EC9B:0x3FC9B220 0x4200AC77:0x3FC9B270 0x4037B39D:0x3FC9B2C0
0x4200cdc3: task_wdt_timeout_handling at /home/franz/esp-idf-v5.3/components/esp_system/task_wdt/task_wdt.c:434
0x4200d1e0: task_wdt_isr at /home/franz/esp-idf-v5.3/components/esp_system/task_wdt/task_wdt.c:507
0x4037725d: _xt_lowint1 at /home/franz/esp-idf-v5.3/components/xtensa/xtensa_vectors.S:1240
0x4200644a: uart_ll_get_txfifo_len at /home/franz/esp-idf-v5.3/components/hal/esp32s3/include/hal/uart_ll.h:390
 (inlined by) uart_tx_char at /home/franz/esp-idf-v5.3/components/esp_driver_uart/src/uart_vfs.c:170
0x4200731f: uart_write at /home/franz/esp-idf-v5.3/components/esp_driver_uart/src/uart_vfs.c:223
0x4200a395: esp_vfs_write at /home/franz/esp-idf-v5.3/components/vfs/vfs.c:502 (discriminator 4)
0x4200f35a: write at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/syscalls/syswrite.c:11 (discriminator 1)
0x42009ee4: console_write at /home/franz/esp-idf-v5.3/components/esp_vfs_console/vfs_console.c:67
0x4200a395: esp_vfs_write at /home/franz/esp-idf-v5.3/components/vfs/vfs.c:502 (discriminator 4)
0x4200ed9a: __swrite at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/stdio.c:94
0x4200e5d5: __sflush_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/fflush.c:224
0x4200e626: _fflush_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/fflush.c:278
0x4200ebca: __sfvwrite_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/fvwrite.c:232
0x42012cd3: __sprint_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vfiprintf.c:429
 (inlined by) __sprint_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vfprintf.c:399
0x42012679: _vfprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1766 (discriminator 1)
0x4200ec9b: printf at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/printf.c:56
0x4200ac77: echo_task at /home/franz/git/fli/uart_echo_rs485/uart_echo_rs485/main/rs485_example.c:113
0x4037b39d: vPortTaskWrapper at /home/franz/esp-idf-v5.3/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

```

cristianfunes79, 2024-08-17:
it says v5.3 dirty, are you on v5.3 tag? do you have local changes? did you update submodules?

nopnop2002, 2024-08-17:
>I (39) boot.esp32s3: Boot SPI Speed : 80MHz

It seems to only occur on esp32s3.

diplfranzhoepfinger, 2024-08-19:
> it says v5.3 dirty, are you on v5.3 tag? do you have local changes? did you update submodules?

yes, have installed this  https://github.com/espressif/esp-idf/issues/14337#issuecomment-2278106938

songruo, 2024-09-13:
https://github.com/espressif/esp-idf/pull/14382 PR merged.

closed 2024-09-13 by songruo