espressif/esp-idf #14385: [MAYBE BUG]sending ESP32 WiFi mesh to the root node

TroyMitchell911, 2024-08-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I sent a message to the root node in this way from the child node:
```c++
static uint8_t tx_buf[10];
for(int i = 0; i < 10; i++) {
	tx_buf = i;
}
mesh_data_t data;
data.data = tx_buf;
data.size = sizeof(tx_buf);
data.proto = MESH_PROTO_BIN;
data.tos = MESH_TOS_P2P;
esp_mesh_send(nullptr, &data, MESH_DATA_P2P, NULL, 0);
```

I follow here, where the 'to' parameter of esp_cesh_dend is displayed as NULL, so it can be sent to the root node：[https://docs.espressif.com/projects/esp ... -reference](https://docs.espressif.com/projects/esp-idf/zh_CN/v5.3/esp32/api-reference/network/esp-wifi-mesh.html#mesh-api-reference)

But I encountered error code 4008 and the data.size equals 0 when receiving at the root node, which is as follows:
```c++
err = esp_mesh_recv(&from, &data, portMAX_DELAY, &flag, nullptr, 0);
if(err != ESP_OK || data.size != sizeof(msg_t)) {
	ESP_LOGI(TAG, "fuck: err: %x size:%d", err, data.size);
	continue;
}
```

Is it a bug?

TroyMitchell911, 2024-08-17:
version: esp-idf v5.3

m986883511, 2024-08-17:
can you help me watch my problem, I can not test the example. https://github.com/espressif/esp-idf/issues/14346

zhangyanjiaoesp, 2024-08-19:
@TroyMitchell911 
Do you mean the `esp_mesh_send()` returns OK on the child node, but the `esp_mesh_recv()` returns error on the root node? 
Is the `from` argument to the `esp_mesh_recv()` function the address of the child node?

TroyMitchell911, 2024-08-22:
> @TroyMitchell911 Do you mean the `esp_mesh_send()` returns OK on the child node, but the `esp_mesh_recv()` returns error on the root node? Is the `from` argument to the `esp_mesh_recv()` function the address of the child node?

I think I forgot to set mesh_data.size before I call `esp_mesh_recv()`....
sorry for noise.

closed 2024-08-22 by TroyMitchell911