espressif/esp-idf #14386: wifi driver issue

KOEGlike, 2024-08-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.4

### Espressif SoC revision.

esp32c3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

wokwi simulator

### Power Supply used.

USB

### What is the expected behavior?

that the code works

### What is the actual behavior?

it dosnt connect to wifi, gives an error

### Steps to reproduce.

1. clone my [repo](https://github.com/KOEGlike/automaticChickenCoopDoor)
2. build with esp-idf 5.1.4
3. install wokwi emulator
4. open diagram.json in project root (this should bring up the wokwi emulator)
5. run the wokwi emulator 
6. look at the consle

### Debug Logs.

```plain
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x420
load:0x403ce000,len:0x90c
load:0x403d0000,len:0x2370
entry 0x403ce000
I (53) cpu_start: Unicore app
I (53) cpu_start: Pro cpu up.
I (27) cpu_start: Pro cpu start user code
I (27) cpu_start: cpu freq: 160000000 Hz
I (27) cpu_start: Application information:
I (27) cpu_start: Project name:     automaticChickenCoopDoor
I (27) cpu_start: App version:      bc1364a
I (27) cpu_start: Compile time:     Aug 17 2024 20:51:20
I (28) cpu_start: ELF file SHA256:  053f0fb22549c9c7...
I (28) cpu_start: ESP-IDF:          v5.1.4
I (29) cpu_start: Min chip rev:     v0.3
I (29) cpu_start: Max chip rev:     v1.99 
I (29) cpu_start: Chip rev:         v0.3
I (29) heap_init: Initializing. RAM available for dynamic allocation:
I (30) heap_init: At 3FC9C910 len 000236F0 (141 KiB): DRAM
I (30) heap_init: At 3FCC0000 len 0001C710 (113 KiB): DRAM/RETENTION
I (31) heap_init: At 3FCDC710 len 00002950 (10 KiB): DRAM/RETENTION/STACK
I (31) heap_init: At 50000020 len 00001FC8 (7 KiB): RTCRAM
I (32) spi_flash: detected chip: gd
I (32) spi_flash: flash io: dio
I (32) sleep: Configure to isolate all GPIO pins in sleep state
I (32) sleep: Enable automatic switching of GPIO sleep configuration
I (33) app_start: Starting scheduler on CPU0
I (33) main_task: SStarting...
ssid: Wokwi-GUEST
I (48) pp: pp rom version: 9387209
I (49) net80211: net80211 rom version: 9387209
W (50) wifi:wifi osi_nvs_open fail ret=4353

W (51) wifi_init: Failed to unregister Rx callbacks
E (57) wifi_init: Failed to deinit Wi-Fi driver (0x3001)
E (62) wifi_init: Failed to deinit Wi-Fi (0x3001)
[   574][E][WiFiGeneric.cpp:248] wifiLowLevelInit(): esp_wifi_init 4353
SSID: Wokwi-GUEST
I (85) pp: pp rom version: 9387209
I (86) net80211: net80211 rom version: 9387209
W (87) wifi:wifi osi_nvs_open fail ret=4353

W (89) wifi_init: Failed to unregister Rx callbacks
E (95) wifi_init: Failed to deinit Wi-Fi driver (0x3001)
E (101) wifi_init: Failed to deinit Wi-Fi (0x3001)
[   613][E][WiFiGeneric.cpp:248] wifiLowLevelInit(): esp_wifi_init 4353
[   630][E][STA.cpp:298] begin(): STA enable failed!
```

### More Information.

I had to troubleshoot to 4 hours to migrate from Arduino to IDF, might have messed up something.

FabioGazzoni, 2024-08-20:
Same problem. I tried it from two different computers with Ubuntu 22.04. Possibly some library was updated since on a third PC it works with the same files.

**Logs:**

```
I (29) boot: ESP-IDF v5.2-dirty 2nd stage bootloader
I (29) boot: compile time Aug  8 2024 21:05:52
I (29) boot: Multicore bootloader
I (34) boot: chip revision: v3.0
I (38) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 2MB
I (51) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00100000
I (90) boot: End of partition table
I (94) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=2433ch (148284) map
I (154) esp_image: segment 1: paddr=00034364 vaddr=3ffb0000 size=06208h ( 25096) load
I (163) esp_image: segment 2: paddr=0003a574 vaddr=40080000 size=05aa4h ( 23204) load
I (172) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=94400h (607232) map
I (381) esp_image: segment 4: paddr=000d4428 vaddr=40085aa4 size=11294h ( 70292) load
I (421) boot: Loaded app from partition at offset 0x10000
I (421) boot: Disabling RNG early entropy source...
I (432) cpu_start: Multicore app
I (441) cpu_start: Pro cpu start user code
I (441) cpu_start: cpu freq: 160000000 Hz
I (441) cpu_start: Application information:
I (444) cpu_start: Project name:     robot
I (449) cpu_start: App version:      d6cfe84-dirty
I (454) cpu_start: Compile time:     Aug  8 2024 21:05:44
I (460) cpu_start: ELF file SHA256:  f24b65233...
I (466) cpu_start: ESP-IDF:          v5.2-dirty
I (471) cpu_start: Min chip rev:     v0.0
I (476) cpu_start: Max chip rev:     v3.99 
I (480) cpu_start: Chip rev:         v3.0
I (486) heap_init: Initializing. RAM available for dynamic allocation:
I (493) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (498) heap_init: At 3FFBD410 len 00022BF0 (138 KiB): DRAM
I (505) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (511) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (518) heap_init: At 40096D38 len 000092C8 (36 KiB): IRAM
I (525) spi_flash: detected chip: generic
I (528) spi_flash: flash io: dio
W (532) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
W (546) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (557) main_task: Started on CPU0
I (567) main_task: Calling app_main()
I (597) wifi_station_netif: ESP_WIFI_MODE_STA
I (597) net80211: OS adapter function magic error! Magic -559038801 is expected, but it is 3f412
W (597) wifi_init: Failed to unregister Rx callbacks
E (607) wifi_init: Failed to deinit Wi-Fi driver (0x3001)
E (607) wifi_init: Failed to deinit Wi-Fi (0x3001)
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x400d7dd5
0x400d7dd5: wifi_init_sta at /home/fabio/esp/esp-idf/components/micro_ros_espidf_component/network_interfaces/uros_wlan_netif.c:70 (discriminator 1)

file: "/IDF/components/micro_ros_espidf_component/network_interfaces/uros_wlan_netif.c" line 70
func: wifi_init_sta
expression: esp_wifi_init(&cfg)

abort() was called at PC 0x40089447 on core 0
0x40089447: _esp_error_check_failed at /home/fabio/esp/esp-idf/components/esp_system/esp_err.c:50
```

FabioGazzoni, 2024-08-21:
Solution for a project with microros and ROS2 humble:

```
sudo rm -r esp
mkdir esp
cd esp
git clone https://github.com/espressif/esp-idf.git
cd esp-idf
git checkout v5.2
sudo apt-get install git wget flex bison gperf python3 python3-pip python3-venv cmake ninja-build ccache libffi-dev libssl-dev dfu-util libusb-1.0-0
./install.sh esp32
export IDF_TOOLS_PATH="$HOME/required_idf_tools_path"
./install.sh
. ./export.sh
cd components
git clone https://github.com/micro-ROS/micro_ros_espidf_component.git
cd micro_ros_espidf_component
git checkout humble
cd esp/esp-idf/components/micro_ros_espidf_component/
pip3 install catkin_pkg lark-parser colcon-common-extensions
. $IDF_PATH/export.sh
cd examples/int32_publisher
idf.py set-target esp32
idf.py menuconfig
idf.py build
idf.py flash
idf.py monitor

```

KOEGlike, 2024-08-22:
Turns out I need to turn on nvs:
> W (50) wifi:wifi osi_nvs_open fail ret=4353

closed 2024-08-22 by KOEGlike