espressif/esp-idf #14388: mbedtls_mpi_cmp_mpi() called Guru Meditation Error..

butaikis, 2024-08-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am using the library for SRP, there is a problem with `srp_verify_key()`, it causes a reboot and an error. Can someone tell me what it might be related to?

Library: https://github.com/ifullgaz/esp32-srp

Example
```
srp_init(NULL, 0);
  const char *username = "Pair-Setup";
  const char *password = "111-22-333";
  mbedtls_mpi *public_key;
  srp_server = srp_new_server(SRP_TYPE_3072, SRP_CRYPTO_HASH_ALGORITHM_SHA512);
  debug("srp_new_client");
  srp_client = srp_new_client(SRP_TYPE_3072, SRP_CRYPTO_HASH_ALGORITHM_SHA512);
  debug("Server srp_set_username");
  srp_set_username(srp_server, username);
  debug("Server srp_set_auth_pass");
  srp_set_auth_password(srp_server, (const unsigned char *)password, strlen(password));
  debug("Server srp_gen_pub");
  srp_gen_pub(srp_server);

  mbedtls_mpi *salt_mpi = srp_get_salt(srp_server);
  debug("Server srp_gen_salt");
  srp_set_params(srp_client, NULL, NULL, salt_mpi);

  srp_gen_pub(srp_client);
  srp_set_username(srp_client, username);
  srp_set_auth_password(srp_client, (const unsigned char *)password, strlen(password));
  public_key = srp_get_public_key(srp_server);
  srp_compute_key(srp_client, public_key);

  public_key = srp_get_public_key(srp_client);
  srp_compute_key(srp_server, public_key);

  mbedtls_mpi *verfy = srp_get_verify_key(srp_client);
  int res_verify = srp_verify_key(srp_server, verfy);
  if (res_verify)
  {
    debug("Server failed to validate Client");
  }
  else
    debug("Verify client OK");
```

Error
```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4018ca33  PS      : 0x00060830  A0      : 0x800f535d  A1      : 0x3ffb4200
A2      : 0x00000000  A3      : 0x3ffb8f48  A4      : 0x3ffb8da8  A5      : 0x3ffb8b70
A6      : 0x00001800  A7      : 0x00001000  A8      : 0x800f4511  A9      : 0x3ffb41b0
A10     : 0x3ffb8ec8  A11     : 0xfffffff2  A12     : 0x000000c0  A13     : 0x00000030
A14     : 0x3ffb8ec8  A15     : 0x00000030  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

Backtrace: 0x4018ca30:0x3ffb4200 0x400f535a:0x3ffb4220 0x400ddd81:0x3ffb4240 0x400eb5f6:0x3ffb4280 0x40102ee2:0x3ffb42a0
  #0  0x4018ca30 in mbedtls_mpi_cmp_mpi at D:\Develop_VS_Code\.platformio\packages\framework-espidf\components\mbedtls\mbedtls\library/bignum.c:1084
  #1  0x400f535a in srp_verify_key at lib/esp32-srp/src/srp.c:1085
  #2  0x400ddd81 in Core::init() at src/core.cpp:321
  #3  0x400eb5f6 in setup() at src/main.cpp:6
```

butaikis, 2024-08-18:
And the second question. I receive a public key and a proof from TLV8 from the client. How do I extract them correctly and check them on the server side?

AdityaHPatwardhan, 2024-08-20:
Hi @butaikis, this is a third-party library.
Can you please raise the issue at the respective library https://github.com/ifullgaz/esp32-srp since it is related to its own implementation.

To answer your questions, the cause for crash is `LoadProhibited`, it means that the function in consideration is accessing some area of memory which is not valid.
I would ask you to check the input arguments to the function ( maybe print the MPI before the function), that should give you more understanding of the issue. Additionally carefully looking at the `srp_verify_key` API should be helpful.
Furthermore, The address in `EXCVADDR: 0x00000004` is very close to `NULL`, so maybe some API has dereferenced a NULL pointer and tried to access memory from it.

esp-idf also officially supports its own implementation of srp6a, please check if it can meet your use-case.
https://github.com/espressif/esp-idf/blob/v5.3/components/protocomm/include/crypto/srp6a/esp_srp.h

> And the second question. I receive a public key and a proof from TLV8 from the client. How do I extract them correctly and check them on the server side?
This is also not related to esp-idf. please raise the issue where this is applicable.

I am closing the issue, since the issue is more applicable to the library in question. Please raise an issue at the respective library.
Thanks,
Aditya

closed 2024-08-20 by AdityaHPatwardhan